esphome/issues #5370: Unable to access device logs wirelessly using MQTT with TLS

mak-42, 2024-01-13:
### The problem

I am unable to access device logs wirelessly using MQTT **with TLS** (there is no problem without tls).

It displays something like that:

```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/test01node.yaml...
INFO Starting log output from test01node/debug
INFO Successfully reconnected to the MQTT server
INFO Successfully reconnected to the MQTT server
INFO Successfully reconnected to the MQTT server
```

MQTT broker log:
```
2024-01-13T10:33:49: New connection from xxx.xxx.xxx.xxx:5269 on port 8883.
2024-01-13T10:33:49: OpenSSL Error[0]: error:1402542E:SSL routines:ACCEPT_SR_CLNT_HELLO:tlsv1 alert protocol version
2024-01-13T10:33:49: Client <unknown> disconnected: Protocol error.
```

It seems ESPHome does not use certificate pointed in "certificate_authority" param.

config:
```
esphome:
  name:  test01node

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

logger:

ota:
  password: !secret NODE01_OTA_PSW

wifi:
  ssid: !secret WIFI_SSID
  password: !secret WIFI_PASS

web_server:
  port: 80
  auth:
    username: !secret NODE01_WEB_LOG
    password: !secret NODE01_WEB_PSW

mqtt:
  broker: !secret MQTT_MAIN_ADDR
  port: !secret MQTT_MAIN_S_PORT
  certificate_authority: !secret MQTT_MAIN_CA_CERT
  username: !secret NODE01_MQTT_LOG
  password: !secret NODE01_MQTT_PSW
  client_id: test01node
  topic_prefix: test01node

esp32_ble_tracker:

#....
```

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

SaschaRoelofs, 2024-02-20:
Have you found a solution?

mak-42, 2024-02-20:
> Have you found a solution?

Nope.  
A workaround is using cable, but it is not very convenient.

Sian-Lee-SA, 2024-05-09:
My issue shows
```
ERROR Cannot connect to MQTT broker: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:992)
```
Which I assume is the frontend not able to connect although the device connects fine. If your log_topic is assigned then you can use MQTT-Explorer as such and see the logs in that topic.

GiuseppeP96, 2025-04-21:
still having this issue

sinipelto, 2026-03-09:
I have the same issue. Getting certificate verification errors from the esphome WebUI when choosing flash/logs through wireless, but using the cable and the connection from the device itself works fine. I also have CN name check set to False.

Configuration:
mqtt:
  client_id: "test_esp32_node"
  broker: "[HOST_FQDN]"
  discover_ip: False
  discovery_prefix: null
  discovery: False
  username: !secret mqtt_user
  password: !secret mqtt_pass
  certificate_authority: !secret mqtt_cert
  skip_cert_cn_check: True
  port: <MQTTS_PORT>
  idf_send_async: True
  topic_prefix: "testnode"
  log_topic: 
    topic: "logtopic"
    level: VERBOSE
    qos: 0
    retain: False

(!secret mqtt_cert hs the PEM value of the root CA cert)

E.g. WebUI -> Device -> Clean MQTT

Result:

INFO ESPHome 2026.2.4
INFO Reading configuration /config/esp32-ruuvitag-gateway.yaml...
INFO Clearing messages from '/+/esp32-ruuvitag-gateway/#'
INFO Please close this window when no more messages appear and the MQTT topic has been cleared of retained messages.
ERROR Cannot connect to MQTT broker: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get issuer certificate (_ssl.c:1010)

But I still get messages from the device itself to the MQTT broker well.