esphome/issues #5372: Multiple instances of DSMR fails if sensors are not equal

jeroen85, 2024-01-13:
### The problem

I have two P1 meters connected to a ESP32-C3.
It appears this only works correctly if the sensor list for both instances are equal.

As soon as the defined sensors for each instance deviate from each other, the system will not compile.

It will generate a lot of warnings about a redefined sensor list
```
In file included from src/esphome/core/util.cpp:2:
src/esphome/core/defines.h:4: warning: "DSMR_SENSOR_LIST" redefined
 #define DSMR_SENSOR_LIST(F, sep) F(energy_delivered_tariff1) sep F(power_delivered)
 
src/esphome/core/defines.h:3: note: this is the location of the previous definition
 #define DSMR_SENSOR_LIST(F, sep) F(energy_delivered_tariff1) sep F(energy_delivered_tariff2)
```

to stop with an error
```
Compiling .pioenvs/meterkast-new/lib13d/WiFi/WiFiAP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:663:14: error: 'class esphome::dsmr::Dsmr' has no member named 'set_energy_delivered_tariff2'; did you mean 'set_energy_delivered_tariff1'?
   dsmr_p1_1->set_energy_delivered_tariff2(sensor_sensor_3);
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              set_energy_delivered_tariff1
Compiling .pioenvs/meterkast-new/lib13d/WiFi/WiFiClient.cpp.o
*** [.pioenvs/meterkast-new/src/main.cpp.o] Error 1
```

If both sensor lists are equal, it compiles without any issue.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Component causing the issue

dsmr

### Example YAML snippet

```yaml
uart:
#DSMR P1-meter    
- rx_pin: GPIO7
  baud_rate: 115200
  id: uart_dsmr_p1_1

- rx_pin: GPIO3
  baud_rate: 115200
  id: uart_dsmr_p1_2

dsmr:
- id: dsmr_p1_1
  uart_id: uart_dsmr_p1_1
  request_interval: 5s
  request_pin: GPIO6

- id: dsmr_p1_2
  uart_id: uart_dsmr_p1_2
  request_interval: 5s
  request_pin: GPIO2

sensor:  
#DSMR P1 meter 1
- platform: dsmr
  dsmr_id: dsmr_p1_1
  energy_delivered_tariff1:
    name: "Energy Consumed Tariff 1"
  energy_delivered_tariff2:
    name: "Energy Consumed Tariff 2"

#DSMR P1 meter 2
- platform: dsmr
  dsmr_id: dsmr_p1_2
  energy_delivered_tariff1:
    name: "Energy Consumed Tariff 1"
  power_delivered:
    name: "Power Consumed"
    accuracy_decimals: 3
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/meterkast-new/src/main.cpp.o
In file included from src/esphome/components/api/api_frame_helper.h:7,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/defines.h:4: warning: "DSMR_SENSOR_LIST" redefined
 #define DSMR_SENSOR_LIST(F, sep) F(energy_delivered_tariff1) sep F(power_delivered)
 
src/esphome/core/defines.h:3: note: this is the location of the previous definition
 #define DSMR_SENSOR_LIST(F, sep) F(energy_delivered_tariff1) sep F(energy_delivered_tariff2)
 
Archiving .pioenvs/meterkast-new/lib03f/libAsyncTCP-esphome.a
Compiling .pioenvs/meterkast-new/lib13d/WiFi/WiFi.cpp.o
Compiling .pioenvs/meterkast-new/lib13d/WiFi/WiFiAP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:663:14: error: 'class esphome::dsmr::Dsmr' has no member named 'set_energy_delivered_tariff2'; did you mean 'set_energy_delivered_tariff1'?
   dsmr_p1_1->set_energy_delivered_tariff2(sensor_sensor_3);
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
              set_energy_delivered_tariff1
Compiling .pioenvs/meterkast-new/lib13d/WiFi/WiFiClient.cpp.o
*** [.pioenvs/meterkast-new/src/main.cpp.o] Error 1
========================= [FAILED] Took 12.76 seconds =========================
```

### Additional information

Related to
https://github.com/esphome/feature-requests/issues/2062
https://github.com/esphome/esphome/pull/4299

jeroen85, 2024-01-13:
@ssieb / @SecSelS
As you both are related to the feature and pull request to allow multiple instances of DSMR running, maybe you can give me a clue how to solve this :-)

jeroen85, 2024-01-19:
As code-owners: @glmnet / @zuidwijk 
Any idea how to solve this issue?

jeroen85, 2024-02-02:
No one?
It's a valid case as I have a district heating meter and an electricity meter. Both use a different set of sensors. I have no idea how to solve this (ok... one idea would be getting an extra ESP...)

jeroen85, 2024-02-28:
@glmnet / @zuidwijk
Is the DSMR component not supported anymore?

ssieb, 2024-02-29:
I'm only involved because I enabled multiple instances.  I didn't know it was this fragile.  Unfortunately, it looks like the component is really not designed to have more than one instance.  I'm surprised the second (or first) one works at all.

kasbrot123, 2024-05-01:
I also tried to read out two smart meters via P1. I am from Austria and my smart meters are encrypted.
Since the TX from the smart meter device is an "open collector", one can use the same UART pin, you only have to be careful not setting both request pins to high. So I have one UART device definition in my yaml. 

I tried this solution with two dsmr definitions and the only issue was that always the first key from the dsmr was used to decrypt the data. The second one got an error: "Data should start with /". I think it's because of the wrong encryption key. 

The feature was mentioned in esphome/feature-requests (DSMR - Ability to use multiple keys and uart assigned sensors # 2062) but I could not comment, since it was already closed. There I think the person testing the code was using the same key (`decryption_key: !secret dsmr_key_p1_1`). This also works for me but I would like to use two different keys.

kasbrot123, 2024-05-01:
> No one?
> It's a valid case as I have a district heating meter and an electricity meter. Both use a different set of sensors. I have no idea how to solve this (ok... one idea would be getting an extra ESP...)

Yes, I agree. I have two electricity meters and I tried to build a compact solution with two P1 ports on my own, since there is not really a product available. But I guess the common solution is to put as many "ESP reader" as smart meter in your home.