esphome/issues #5374: Copy in ESPHome configuration editor crashes HA

Lateralleap, 2024-01-14:
### The problem

If I go into the configuration editor for an ESPHome device, select a portion of YAML configuration, right click and select Copy, it seems to crash HA. Completely reproducible and nothing in logs that I can see

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP32

### Board

ESP32-S3

### Component causing the issue

N/A

Lateralleap, 2024-01-18:
This is not an issue with ESPHome, it is an issue with Home Assistant when using Chrome browset.  Does not do it with Edge

Lateralleap, 2024-01-18:
Not ESPHome issue

3ative, 2024-02-25:
Can't reproduce that here. Copy/Cut/Paste works as expected.