esphome/issues #5375: Devices can't survive ~100d uptime...

Masterz69, 2024-01-15:
### The problem

As soon my devices reached ~99d10h uptime:
* they can't go over - continue reporting same uptime for an hours;
* they start to work strange - stop reporting measurements, stop react on buttons...

[image]
[image]

Unfortunately no DEBUG configured on devices side.

[image]

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP8266

### Board

sonoff 4chror3/dualr3/..., d1 mini

Masterz69, 2024-01-15:
Device `Garland Hall` reporting same uptime (99d10h 04:58) for 5+ hours, not reporting RSSI, relay not switched by button.

But relay still can be operated from HA, as well button sensor is reporting state to HA.

As soon device restarted (same Esphome version) fuinctionality recovers on 100%.

DAVe3283, 2024-01-19:
This one will be a hard one to debug because of how long it will take to verify a fix.

I had a device at 269 days before a power outage:
[image]
I am 99% sure that was running ESPHome 2023.3.0.

You can look for changes between 2023.3.0 (>269 days uptime) and 2023.9.3 (yours drop out at <100 days). But I am not entirely sure there is a problem with ESPHome causing the dropout.

Are the devices wireless or wired Ethernet? For what it is worth, I have lots of issues with devices rebooting or losing connection when connected wirelessly after some months, but the ESP32 devices with Ethernet seem to run forever. It could be possible that your wireless access point had some issue, and the reboot of the ESPs caused the AP to resolve the connection issue. I have seen that before.

Masterz69, 2024-01-19:
> This one will be a hard one to debug because of how long it will take to verify a fix.

IMHO with sufficient skills & tools is possible to tweak uptime value in memory & check it fast.

Devices all wireless - different variations from different vendors of ESP32 & ESP8285. 
All stopped to report uptime/rssi as soon next uptime value is >100d.
Same time API connection still active, devices responds on commands from HA (switch etc.), but NO on local buttons & switches.

If device no more switching relay on button press (completely local code execution) - what else can be an cause ?

DAVe3283, 2024-01-19:
That is really odd. I updated my devices to 2023.11.4 after the power outage, and will know if they can go past 100 days in another 2 months. Would be an interesting data point to see if the issue is uptime, or some component you are using on yours.

It looks like they stop reporting around 8589xxx seconds of uptime, which is nowhere near any power of 2 boundary that would explain a crash. Though if you multiply by 1000 to get milliseconds, then we get a 33-bit number (1 1111 1111 1111 1111 0111 0001 0001 0000), but it looks like it is about to roll over to a 34-bit number. But the issue should be the 32 bit rollover (first occurs at 49.7 days), unless there is a subtle logic error tracking 32-bit overflows of the 32-bit millisecond uptime.

The source code for uptime is really straightforward though (and untouched for 3 years): https://github.com/esphome/esphome/blob/dev/esphome/components/uptime/uptime_sensor.cpp

So I would think it is some other component in your YAML that might be the source of the problem.

Masterz69, 2024-01-19:
> So I would think it is some other component in your YAML that might be the source of the problem.

Will see. At moment enabled DEBUG on some of devices, hope will get some information after 100 days :)
Couple of affected devices - Sonoff S26 - really simple one.

DAVe3283, 2024-01-19:
Mind sharing your simplest YAML that had this issue (e.g. the Sonoff S26)? You can mask out sensitive information if you're not using secrets.yaml, but please don't remove any lines/settings. Any detail could help.

DAVe3283, 2024-03-26:
I have finally ticked over 100 days uptime on many of my ESPHome devices:
[image]

All are running firmware compiled from ESPHome 2023.11.4. Here is a list of components I use in the devices that just hit 100 days uptime. These aren't (at least individually, or in the combinations I use) going to prevent 100+ days of uptime.

- api (with encryption)
- binary_sensor
  - gpio
- button
  - restart
  - template
- captive_portal
- climate (platform: pid)
- dallas
- ethernet (type: LAN8720)
- i2c
- light/rgb
- logger
  - VERBOSE on 1 device
  - DEBUG (default) on most
- mdns (to disable it)
- ota (with password)
- output
  - esp8266_pwm
  - ledc
  - template
- sensor
  - absolute_humidity
  - adc
  - bme280_i2c
  - bme680
  - cse7766
  - dallas
  - pid
  - pmsx003
  - pulse_counter
  - senseair
  - template
  - total_daily_energy
  - uptime
  - wifi_signal
- status_led
- sun
- switch
  - gpio
- time
  - ds1307
  - homeassistant
- uart
- wifi + ap

And notably, here are some components I do *not* use:

- mdns (always set to disabled)
- web_server (I notice several people who have lockups use this)

@Masterz69 what firmware did you load in to test with, and how close to 100 days are you now? Maybe we can narrow down if a specific component is responsible if yours crash again with something not in the list above.

Masterz69, 2024-03-26:
@DAVe3283
> what firmware did you load in to test with, and how close to 100 days are you now?

Now running 2023.12.7, bigger uptime is 69 days.
Previosuly no DEBUG logging was enabled, so hope to get some info next time 100d reached.

Nothing special used. Actually simple devices, like Sonoff S26 plugs, was affected.
Using mDNS/ota/api, no web_server, only WiFi connectivity.

Masterz69, 2024-04-26:
Absolutely same staff happened again.
ESPHome version 2023.12.7.
[image]

Devices reached 100d uptime and it stopped to increase.
[image]

At same time devices still connected to HA, responding on log request.

They still react on commands from Home Assistant:
```
[10:03:54][C][wifi_signal.sensor:009]: WiFi Signal 'RSSI'
[10:03:54][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[10:03:54][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[10:03:54][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[10:03:54][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[10:03:56][D][switch:012]: 'Center' Turning ON.
[10:03:56][D][switch:055]: 'Center': Sending state ON
[10:03:57][D][switch:016]: 'Center' Turning OFF.
[10:03:57][D][switch:055]: 'Center': Sending state OFF
```

BUT IGNORE swicthes and buttons located on devices - NOTHING in log on pressing/switching/...
Just nothing, so can't show anything from log here as it empty on pressing device buttons/switches.

Going to deploy latest ESPHome to have devices functional.

DAVe3283, 2024-04-26:
Can you watch the log of one of the ESPHome devices that still connects for a couple minutes, and post it here? I am curious if the device is even trying to update the Uptime sensor.

Masterz69, 2024-04-26:
> Can you watch the log of one of the ESPHome devices that still connects for a couple minutes, and post it here? I am curious if the device is even trying to update the Uptime sensor.

Unfortunately after 100 days of waiting I was little bit impatient and installed 2024.4.1 to recover.

As can see attached screenshot - device stopped reporting RSSI to HA too.
[image]

Another device stopped reporting of any sensor reading, attaching one example:
[image]

While device still Connected to HA all the time except rebooting with new version:
[image]

Compiled couple of devices with no uptime sensor.
Will monitor logs on transition next time.

DAVe3283, 2024-06-24:
I had this same thing happen on one of my devices (Emporia Vue) running ESPHome 2023.12.4. It stopped reporting any new data to Home Assistant over the API, but would provide logs and respond to commands. Told it to reboot itself, and it did (including log messages that it was rebooting cleanly), then started working again.

Reflashed it to ESPHome 2024.6.1 and will see what happens. This is extremely weird.

Masterz69, 2024-07-09:
Probably related issue. From 40+ ESPHome devices only 2 of them, 

- which already have 49d+ uptime (so one rollover of micro happened) 
- couple times in a day become unavailable in HA and reconnects again very soon. 

No device with uptime less than 49d demonstrate same behaviour in same network etc.

See nothing in device log (by network), just loosed network connection:
```
[15:27:12][D][sensor:093]: 'RSSI': Sending state -56.00000 dBm with 0 decimals of accuracy
[15:27:56][D][sensor:093]: 'Uptime': Sending state 4558389.50000 s with 0 decimals of accuracy
WARNING sns-ups-in @ 192.168.1.184: Connection error occurred: [WinError 64] The specified network name is no longer available
INFO Processing unexpected disconnect from ESPHome API for sns-ups-in @ 192.168.1.184
WARNING Disconnected from API
INFO Successfully connected to sns-ups-in @ 192.168.1.184 in 0.003s
INFO Successful handshake with sns-ups-in @ 192.168.1.184 in 2.675s
[15:28:42][D][sensor:093]: 'RSSI': Sending state -59.00000 dBm with 0 decimals of accuracy
```

Masterz69, 2024-08-05:
No more data sent to HA after second uptime roll-over: Device - uptime - Last Seen.
[image]

Log after it happens: no API connection in log, no data sending, just config...
[100d_log.txt](https://github.com/user-attachments/files/16492810/100d_log.txt)

And nothing more from 6+ sensors.

Masterz69, 2024-09-04:
```
[06:09:09][D][scheduler:328]: Incrementing scheduler major
```
This is a last line in log when second `uptime` roll-up happens.
Nothing more in log, as well no data sent to HA anymore. 
Till device restart.

<details>

<summary>Log before uptime second roll-up</summary>

```txt
INFO ESPHome 2024.4.1
INFO Reading configuration H:\u8-esphome\sns-Gate.yaml...
INFO Updating https://github.com/nrandell/dallasng.git@None
INFO Starting log output from 192.168.1.164 using esphome API
INFO Successfully connected to sns-gate @ 192.168.1.164 in 0.003s
INFO Successful handshake with sns-gate @ 192.168.1.164 in 2.446s
[05:57:39][I][app:100]: ESPHome version 2024.4.1 compiled on May 27 2024, 19:14:06
[05:57:39][I][app:102]: Project Wemos.D1_mini_v4 version ESP8266EX
[05:57:39][C][wifi:580]: WiFi:
[05:57:39][C][wifi:408]:   Local MAC: C8:C9:A3:2F:77:FB
[05:57:39][C][wifi:413]:   SSID: [redacted]
[05:57:39][C][wifi:416]:   IP Address: 192.168.1.164
[05:57:39][C][wifi:419]:   BSSID: [redacted]
[05:57:39][C][wifi:421]:   Hostname: 'sns-gate'
[05:57:39][C][wifi:423]:   Signal strength: -73 dB ▂▄▆█
[05:57:39][C][wifi:427]:   Channel: 6
[05:57:39][C][wifi:428]:   Subnet: 255.255.255.0
[05:57:39][C][wifi:429]:   Gateway: 192.168.1.1
[05:57:39][C][wifi:430]:   DNS1: 192.168.1.1
[05:57:39][C][wifi:431]:   DNS2: 0.0.0.0
[05:57:39][C][logger:166]: Logger:
[05:57:39][C][logger:167]:   Level: DEBUG
[05:57:39][C][logger:169]:   Log Baud Rate: 115200
[05:57:39][C][logger:170]:   Hardware UART: UART0
[05:57:39][C][logger:174]:   Level for 'status_led': INFO
[05:57:39][C][uptime.sensor:031]: Uptime Sensor 'Uptime'
[05:57:39][C][uptime.sensor:031]:   Device Class: 'duration'
[05:57:39][C][uptime.sensor:031]:   State Class: 'total_increasing'
[05:57:39][C][uptime.sensor:031]:   Unit of Measurement: 's'
[05:57:39][C][uptime.sensor:031]:   Accuracy Decimals: 0
[05:57:39][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[05:57:39][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Is Opened'
[05:57:39][C][gpio.binary_sensor:015]:   Device Class: 'opening'
[05:57:39][C][gpio.binary_sensor:016]:   Pin: GPIO12
[05:57:39][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Is Moving'
[05:57:39][C][gpio.binary_sensor:015]:   Device Class: 'moving'
[05:57:39][C][gpio.binary_sensor:016]:   Pin: GPIO14
[05:57:40][C][gpio.output:010]: GPIO Binary Output:
[05:57:40][C][gpio.output:011]:   Pin: GPIO13
[05:57:40][C][gpio.output:010]: GPIO Binary Output:
[05:57:40][C][gpio.output:011]:   Pin: GPIO15
[05:57:40][C][wiegand:121]: Wiegand reader:
[05:57:40][C][wiegand:122]:   D0 pin: GPIO2
[05:57:40][C][wiegand:123]:   D1 pin: GPIO0
[05:57:40][C][template.text_sensor:020]: Template Sensor 'Tag Scanned'
[05:57:40][C][output.switch:068]: Output Switch 'Walker Keep'
[05:57:40][C][output.switch:070]:   Icon: 'mdi:walk'
[05:57:40][C][output.switch:090]:   Restore Mode: always OFF
[05:57:40][C][output.switch:068]: Output Switch 'Full Keep'
[05:57:40][C][output.switch:070]:   Icon: 'mdi:gate-open'
[05:57:40][C][output.switch:090]:   Restore Mode: always OFF
[05:57:40][C][restart.button:017]: Restart Button 'Restart'
[05:57:40][C][safe_mode.button:022]: Safe Mode Button 'Safe Mode'
[05:57:40][C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[05:57:40][C][status:034]: Status Binary Sensor 'Status'
[05:57:40][C][status:034]:   Device Class: 'connectivity'
[05:57:40][C][dallasng.sensor:050]: DallasComponent:
[05:57:40][C][dallasng.sensor:051]:   Pin: GPIO16
[... 90 lines cut ...]
[06:06:05][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:06:26][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:06:26][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:06:39][D][sensor:093]: 'RSSI': Sending state -76.00000 dBm with 0 decimals of accuracy
[06:06:42][D][sensor:093]: 'Uptime': Sending state 8589788.00000 s with 0 decimals of accuracy
[06:06:47][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:06:47][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:07:08][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:07:08][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:07:29][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:07:29][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:07:50][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:07:50][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:08:09][D][sensor:093]: 'RSSI': Sending state -75.00000 dBm with 0 decimals of accuracy
[06:08:11][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:08:11][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:08:22][D][sensor:093]: 'Uptime': Sending state 8589888.00000 s with 0 decimals of accuracy
[06:08:32][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:08:32][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:08:53][D][dallasng.sensor:119]: 'Temperature' got temperature as 25312 (25.312000)
[06:08:53][D][sensor:093]: 'Temperature': Sending state 25.31200 °C with 1 decimals of accuracy
[06:09:09][D][scheduler:328]: Incrementing scheduler major
```

</details>

Masterz69, 2024-09-05:
Hello everyone taking care core components !
`Scheduler` have a fatal bug causing it stop/hung/break/ after `major` increment happens.
`Incrementing scheduler major` is a posthumous inscription - no more activities performed by device after that.

Masterz69, 2024-09-05:
Greetings, @OttoWinter !
KIndly asking to look into issue. 
`Scheduler` stops/hungs/process no more after `Incrementing scheduler major` on ~99d 10h of uptime...

DAVe3283, 2024-10-27:
I had an ESP32-C6 device due to hit the 2nd rollover tomorrow, but I noticed it had rebooted at 8330620 seconds (96.4 days). And it kept rebooting randomly since then:
[image]
I reflashed it to the latest ESPHome version, will have to see what happens next time.

Something very strange is definitely happening to some (but not all?) of these devices when they get close to 100 days uptime. I have another device coming up on 100 days next month that has experienced this issue before (an Emporia Vue). Hopefully I will get some useful logs this time. I will probably start logging 5 days in advance based on what happened here.

clydebarrow, 2024-10-29:
> `Incrementing scheduler major` is a posthumous inscription - no more activities performed by device after that

The flaw in that theory is that `major_millis_` will be incremented every 24.86 days, so that will have happened 3 times already without any apparent ill effects.

Masterz69, 2024-10-29:
> The flaw in that theory is that `millis_major_` will be incremented every 24.86 days, so that will have happened 3 times already without any apparent ill effects.

Unfortunately or not - this is not a theory, but reality.
On `Incrementing scheduler major` event which happens on ~99days 16 hours devices just stops to do any scheduled activity.

clydebarrow, 2024-10-29:
I've simulated running a fast clock and the scheduler major happily rolls over multiple times without issue; First roll over at 49.7 days
```
[21:31:05][D][sensor:094]: 'Uptime': Sending state 4223141.50000 s with 0 decimals of accuracy
[21:31:07][D][scheduler:328]: Incrementing scheduler major
[21:31:07][D][uptime.sensor:018]: Detected roll-over 🦄
[21:31:07][D][sensor:094]: 'Uptime': Sending state 4309536.00000 s with 0 decimals of accuracy
```

Next at 99 days and 10 hours
```
[21:32:32][D][sensor:094]: 'Uptime': Sending state 8543142.00000 s with 0 decimals of accuracy
[21:32:32][D][scheduler:328]: Incrementing scheduler major
[21:32:33][D][uptime.sensor:018]: Detected roll-over 🦄
[21:32:33][D][sensor:094]: 'Uptime': Sending state 8629536.00000 s with 0 decimals of accuracy
```

And it's still running. So while it appears the hang does happen at 2 to the 33 ms after boot it's not the major rollover that causes the problem, it just happens at the same time. And it's not happening on my test. I'll try with some different components.

Masterz69, 2024-10-31:
> And it's not happening on my test. I'll try with some different components.

In my case starting 2024 January each and every device reaching 99+days - stops to proceed any internal activity: no data from any sensor, no reaction on GPIO activity, nothing in log.
After a long time waiting me catched when it happens - exactly on that (last) row in log:
```
[21:04:23][D][scheduler:328]: Incrementing scheduler major
```

Masterz69, 2025-01-31:
2024.10.1 - same stuff. As far I understood - device stop any internal activity: no sensor reading, even uptime not updated anymore.

clydebarrow, 2025-02-02:
I'm happy to look into this further, but will need a *complete* yaml config for a device that exhibits the problem, because it is definitely not happening with all configs.

Masterz69, 2025-02-02:
> I'm happy to look into this further, but will need a _complete_ yaml config for a device that exhibits the problem, because it is definitely not happening with all configs.

Here is a complete set of YAML (excl. secrets.yaml) for simplest of my devices - Sonoff S26.
In my case it happens with each & every device, regardless of components used - switches, outputs, sensors, etc.
As well it happens with both ESP8266 & ESP32 platforms (using only Arduino framework).
That is so far (from January 2024) my observations.

[EH5375.zip](https://github.com/user-attachments/files/18633912/EH5375.zip)

Hope included in zip file all required packages. Let me know if something missing.

Masterz69, 2025-02-02:
> I'm happy to look into this further, but will need a _complete_ yaml config for a device that exhibits the problem, because it is definitely not happening with all configs.

Another example, recently reached the deadline - oustide environment sensor on D1 mini.
I2C, two sensors, status LED, no switches/outputs.

[EH5375s.zip](https://github.com/user-attachments/files/18634054/EH5375s.zip)

As usual - no more data received from device in HA, while still shown as connected.
Reachable from HA, so it was possible to press button to Restart.

Adding also history (exported from HA) for uptime/temperature/RSSI sensors: 

[EH5375s_history.csv](https://github.com/user-attachments/files/18634071/EH5375s_history.csv)

clydebarrow, 2025-02-02:
Sorry, that's not complete.

```
Error reading file common/package/../binary_sensor/api_connected.yaml: [Errno 2] No such file or directory: 'common/package/../binary_sensor/api_connected.yaml'
```

clydebarrow, 2025-02-04:
So I was able to compile and run that. Simulating advanced clock time, it seems to work fine:
```
[12:55:00][D][sensor:093]: 'RSSI': Sending state -55.00000 dBm with 0 decimals of accuracy
[12:55:04][D][scheduler:328]: Incrementing scheduler major
[12:55:18][D][uptime.sensor:018]: Detected roll-over 🦄
[12:55:18][V][sensor:043]: 'Uptime': Received new state 4294980.500000
[12:55:18][D][sensor:093]: 'Uptime': Sending state 4294980.50000 s with 0 decimals of accuracy
[12:56:30][V][sensor:043]: 'RSSI': Received new state -52.000000
[12:56:30][D][sensor:093]: 'RSSI': Sending state -52.00000 dBm with 0 decimals of accuracy
[12:56:58][V][sensor:043]: 'Uptime': Received new state 4295080.500000
[12:56:58][D][sensor:093]: 'Uptime': Sending state 4295080.50000 s with 0 decimals of accuracy
[12:58:00][V][sensor:043]: 'RSSI': Received new state -52.000000
[12:58:00][D][sensor:093]: 'RSSI': Sending state -52.00000 dBm with 0 decimals of accuracy
[12:58:38][V][sensor:043]: 'Uptime': Received new state 4295180.500000
[12:58:38][D][sensor:093]: 'Uptime': Sending state 4295180.50000 s with 0 decimals of accuracy
```

clydebarrow, 2025-02-04:
I think I have reproduced the problem, it's not the scheduler that's failing, but it does seem that sensors stop updating.

clydebarrow, 2025-02-04:
So there is a problem with the scheduler, but the triggering factor is this:
```
preferences:
  flash_write_interval: 0s
```

This results in the `preferences` component's syncer function being called continuously. 0s does *not* disable the flash_write interval, it literally sets it to 0.

With that removed (or a value like 60s set) the problem goes away.

I will also be raising a PR to fix the scheduler.

Masterz69, 2025-02-04:
> 0s does _not_ disable the flash_write interval, it literally sets it to 0.

There was no idea to disable, but write changes immediately.
Most of my devices change f.e. output states once/twice a day - decided to write state into flash immediately, why not.
Device with sensors have only switch with `restore_mode: RESTORE_DEFAULT_ON`, so actually sync function must do nothing, but...

Really appreciate @clydebarrow effort in digging deep and fix the issue. 
Waiting for PR is in release.

ssieb, 2025-02-04:
I think you misunderstand that restore mode.  That does mean to save the state and restore it at boot.  The default is just what is used if the state isn't saved.

clydebarrow, 2025-02-04:
> There was no idea to disable, but write changes immediately.

Fair enough, you couldn't have known how it was implemented. For now you could set it to 1s for example. I have another PR for that anyway which will make the behaviour more reasonable.

DAVe3283, 2025-02-05:
Interestingly, my Emporia Vue stops sending updates at ~100 days as well, but uses 48 hours for the interval:

```yaml
preferences:
  flash_write_interval: 48h
```

I suspect my other problem devices are using 0, because I also wanted to save the changes immediately, and [the docs said to use 0 for that](https://esphome.io/components/esphome#adjusting-flash-writes). But interesting the Emporia Vue has the same problem with a 48h interval.

closed 2025-03-27 by kbx81