esphome/issues #5380: Random panic (ld2410 + bh1750)

fornellas, 2024-01-15:
### The problem

The device seems to "randomly" panic, every ~1h or so.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.12.2

### What platform are you using?

ESP32

### Board

ESP32-C3-MINI-1

### Component causing the issue

ld2410 + bh1750

### Example YAML snippet

```yaml
esphome:
  name: "presence"
  friendly_name: "Presence"
  name_add_mac_suffix: true
  on_boot:
    - priority: 600.0
      then:
        - lambda: |-
            // Default: Connecting
            id(wifi_led_pwm).set_level(0.5);
            id(wifi_led_pwm).set_period(1000);
            id(wifi_led_pwm).turn_on();
        - script.execute: fast_reboot_factory_reset
        - switch.turn_off: ld2410_bluetooth
        - button.press: ld2410_query_params
  on_loop:
    then:
      - lambda: |-
          // Captive portal
          if(id(wifi_component).has_ap() && !id(wifi_component).has_sta()) {
            id(wifi_led_pwm).set_level(0.5);
            id(wifi_led_pwm).set_period(200);
          } else {
            id(wifi_led_pwm).set_period(1000);
            // Connected
            if(id(wifi_component).is_connected())
              id(wifi_led_pwm).set_level(1);
            // Connecting
            else
              id(wifi_led_pwm).set_level(0.5);
          }

esp32:
  board: esp32-c3-devkitm-1

globals:
  - id: fast_reboot
    type: int
    restore_value: yes
[... 412 lines cut ...]
3fca3750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001
3fca3790: 0x84000000 0xbaad5678 0x00000018 0xabba1234 0x0000000c 0x3fca3640 0x00000000 0x00000000
3fca37b0: 0xbaad5678 0x00000050 0xabba1234 0x00000044 0x3c0d42a8 0x00000002 0x7fc00000 0x3c0d42e0
3fca37d0: 0x0001c200 0x3fca3814 0x0000007a 0x00000200 0x00000000 0x3fc92aec 0x00000000 0x00000000
3fca37f0: 0x00000000 0x3fcb2720 0x3fcb2740 0x3fcb2740 0x766dd200 0xbaad5678 0x00000210 0xabba1234
3fca3810: 0x00000201 0x3b305b1b 0x5b6d3633 0x735b5d44 0x6f736e65 0x39303a72 0x203a5d34 0x2d694827
3fca3830: 0x6b6e694c 0x4b4c4820 0x32444c2d 0x3a303134 0x65725020 0x636e6573 0x69642065 0x6e617473
3fca3850: 0x3a276563 0x6e655320 0x676e6964 0x61747320 0x35206574 0x30302e36 0x20303030 0x77206d63

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038252a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
[I][logger:355]: Log initialized
```

fornellas, 2024-01-15:
Bug seems to be here:

```
0x4200d10a: esphome::ld2410::LD2410Component::handle_periodic_data_(unsigned char*, int) at toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_tree.h:2476
 (inlined by) esphome::ld2410::LD2410Component::handle_periodic_data_(unsigned char*, int) at presence/src/esphome/components/ld2410/ld2410.cpp:146

0x4200f58a: esphome::ld2410::LD2410Component::readline_(int, unsigned char*, int) at toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_tree.h:2476
 (inlined by) esphome::ld2410::LD2410Component::readline_(int, unsigned char*, int) at presence/src/esphome/components/ld2410/ld2410.cpp:478
 
0x4200f604: esphome::ld2410::LD2410Component::loop() at toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_tree.h:2476
```

milandzuris, 2025-02-05:
up