esphome/issues #5382: Transport encryption key is invalid

knaak, 2024-01-15:
### The problem

Configuring a newly created ESPHome in Integrations | Configure fails due to Encryption Key being invalid despite the key matching what is in the YAML.

This ESP32d has been previously configured under a different name/yaml in a previous 'life' of the device.  However, I freshly added it as a new device with an entirely new yaml.  I also deleted the previous ESPHome device that it initially was recognized as being before reseting and re-installing again.

When HA detects the new device, and I hit the "Configure" and enter the key from the YAML, it fails with an error:
The transport encryption key is invalid. Please ensure it matches what you have in your configuration

Here is the warning in HA Core Log:

Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: 1:40:28 PM (2 occurrences)
Last logged: 1:40:33 PM

192.168.1.63: Connection error occurred: 192.168.1.63: Invalid encryption key: received_name=motion-1

### Which version of ESPHome has the issue?

2023.12.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP32

### Board

espressif

### Component causing the issue

api key

### Example YAML snippet

```yaml
esphome:
  name: motion-1
  friendly_name: motion-1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "13Yky2f6/bRVM5W5ZH1TJus5AmjlHujhVqY/h1CZq1o="

ota:
  password: "5863482ba8c4eea4dccf949c4c02b870"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Motion-1 Fallback Hotspot"
    password: "mIv39wQr7FAD"

captive_portal:

uart:
  tx_pin: GPIO16
  rx_pin: GPIO17
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

globals:
  - id: cpu_speed
    type: int
    restore_value: no
    initial_value: "0"

ld2410:
  id: ld2410_radar
  

#Occupancy sensor
binary_sensor:
  - platform: ld2410
    has_target:
      name: Radar Target
      id: radar_has_target
    has_moving_target:
      name: Radar Moving Target
    has_still_target:
      name: Radar Still Target
#Exposes control buttons
button:
  - platform: restart
    icon: mdi:power-cycle
    name: "ESP Reboot"

  - platform: factory_reset
    disabled_by_default: True
    name: "Factory Reset ESP"
    id: factory_reset_all

  - platform: ld2410
    factory_reset:
      name: "Factory Reset Radar"
    restart:
      name: "Restart Radar"
    query_params:
      name: query params

#Exposes control switches
switch:
  - platform: ld2410
    bluetooth:
      name: "Radar Control Bluetooth"
      id: radar_bluetooth
    engineering_mode:
      name: "Radar Engineering Mode"
time:
  - platform: sntp
    id: time_sntp

#Radar firmware version
text_sensor:
  - platform: ld2410
    version:
      name: "Radar Firmware Version"
#Set distance resolution - 0.75m or 0.2m
select:
  - platform: ld2410
    distance_resolution:
      name: "Radar Distance Resolution"
```

### Anything in the logs that might be useful for us?

```txt
Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: 1:40:28 PM (2 occurrences)
Last logged: 1:40:33 PM

192.168.1.63: Connection error occurred: 192.168.1.63: Invalid encryption key: received_name=motion-1
```

### Additional information

The ESPHome update (2023.12.6) just arrived while I was wiring up this device and it is the first device that I've tried to add using this version.

Also, as mentioned, this esp32d *has been used* on this HA/ESPHome in the past.   It had left over entities which I manually deleted.  Is it possible that the key is from that old version?  That .yaml is long gone so I don't know.  I have installed it fresh/new with the key listed above.

knaak, 2024-01-15:
To resolve this problem, rather than add a new device within home assistant's esphome dashboard, I flashed it as a generic esphome device from the esphome web site and then adopted it in HA.  That worked for me.  I'll leave this open in case its related to the new release in some way.

clydebarrow, 2024-01-15:
> Please ensure it matches what you have in your configuration

When you copy and paste, make sure the '=' on the end is included. That has caught me out because double-clicking the key selects everything except the '='.

knaak, 2024-01-16:
I am sure that I pasted the key correctly.  I will close this since it appears to be unique to me and I have solved it by re-flashing as described above.  I suspect that HA was expecting the previous encryption key for some reason.

closed 2024-01-16 by knaak