esphome/issues #5384: Readings from SDS011 not reaching HASS

milkywade, 2024-01-15:
### The problem

Hi everyone,

I have an ESP32 board that has a MH-Z19 sensor and a SDS011.

For the last two esphome versions I've had an issue where the SDS011 readings do not reach HASS but MH-Z19 seem to work fine.

Today I connected to the ESP32 using the esphome dashboard and I see the below text block cycling in the logs.

Can I do anything to fix this ?

Thank you in advance.

Versions:
1. HASS: 2024.1.3 (core)
2. esphome: 2023.12.6 (running on windows 11 with anaconda3 and python 3.12.0 - see YAML used at bottom)

```
ERROR Exception in callback _ProactorReadPipeTransport._loop_reading(<_OverlappedF...hed result=77>)
handle: <Handle _ProactorReadPipeTransport._loop_reading(<_OverlappedF...hed result=77>)>
Traceback (most recent call last):
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\asyncio\events.py", line 84, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\asyncio\proactor_events.py", line 325, in _loop_reading
    self._data_received(data, length)
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\asyncio\proactor_events.py", line 274, in _data_received
    self._protocol.data_received(data)
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\site-packages\aioesphomeapi\_frame_helper\noise.py", line 163, in data_received
    self._handle_frame(frame)
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\site-packages\aioesphomeapi\_frame_helper\noise.py", line 319, in _handle_frame
    msg = self._decrypt(frame)
          ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\site-packages\noise\state.py", line 74, in decrypt_with_ad
    plaintext = self.cipher.decrypt(self.k, self.n, ad, ciphertext)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\uhuh\anaconda3\envs\esphome\Lib\site-packages\noise\backends\default\ciphers.py", line 13, in decrypt
    return self.cipher.decrypt(nonce=self.format_nonce(n), data=ciphertext, associated_data=ad)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src\\chacha20poly1305_reuseable\\__init__.py", line 136, in chacha20poly1305_reuseable.ChaCha20Poly1305Reusable.decrypt
  File "src\\chacha20poly1305_reuseable\\__init__.py", line 156, in chacha20poly1305_reuseable.ChaCha20Poly1305Reusable.decrypt
  File "src\\chacha20poly1305_reuseable\\__init__.py", line 272, in chacha20poly1305_reuseable._decrypt_with_fixed_nonce_len
  File "src\\chacha20poly1305_reuseable\\__init__.py", line 282, in chacha20poly1305_reuseable._decrypt_data
cryptography.exceptions.InvalidTag

```
Contents of bedroom1.yaml (passwords & API keys have been anonymized)

```
esphome:
  name: bedroom1
  friendly_name: Bedroom1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "kjdfbglkdjfgbksdjfbgkjdfbgkjdfbgkjdfbgjkdfb="

ota:
  password: "sdflksndflksdnfkndslkndfzetrtrt4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom1 Fallback Hotspot"
    password: "lksdnflksndf"

captive_portal:
    
uart:
  - id: uart_2
    rx_pin: GPIO3
    tx_pin: GPIO1
    baud_rate: 9600
  - id: uart_1
    rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 9600

sensor:
  - platform: mhz19
    uart_id: uart_1
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false
  - platform: sds011
    uart_id: uart_2
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"
    update_interval: 5min
```

### Which version of ESPHome has the issue?

2023.12.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
see "the problem"
```

### Anything in the logs that might be useful for us?

```txt
see "the problem"
```

### Additional information

see "the problem"

milkywade, 2024-01-17:
ok, found why lovelace complains, esp home seems to somehow have changed the entity names as x below:

[image]

ganjakyp-hub, 2024-02-09:
I have exactly the same error

ireun, 2024-03-23:
Yes, same here. I'm able to run this command though to see the logs wirelessly:

```
esphome logs .\esphome-web-16b446.yaml
INFO ESPHome 2024.3.0
INFO Reading configuration .\esphome-web-16b446.yaml...
INFO Starting log output from esphome-web-16b446.local using esphome API
INFO Successfully connected to esphome-web-16b446 @ 192.168.1.123 in 0.088s
INFO Successful handshake with esphome-web-16b446 @ 192.168.1.123 in 29.222s
[15:33:44][I][app:102]: ESPHome version 2024.3.0 compiled on Mar 23 2024, 14:50:07
```

milkywade, 2024-03-23:
yup, thanks! that command works for me too

milkywade, 2024-03-30:
changed issue title for clarity

aka13-404, 2024-05-03:
I have the same issue. I don't use HA with this device.

Masterz69, 2024-05-04:
This one helped to me when same issue occured.
[image]

aka13-404, 2024-05-04:
Amazing, I would have never guessed it's a locale problem. Setting this solved it for me as well. Thanks a lot!

honnielsen, 2024-05-08:
unbelievable, worked for me too ...

THANKS

milkywade, 2024-05-23:
Thank you, the above hack works for me as well!

No, since we are talking about a web app, to me the logs should work out of the box, as they were before, I have never had to tweak OS settings to be able to connect to a web page.

That and what is the setting change really doing to my computer and what are the risks (since it’s marked as beta) ?

P.S. I must say I had ended up liking more the command line version of the log viewer, the colored visuals are an improvement over the web version.

Ralms, 2024-08-09:
> This one helped to me when same issue occured. [...]

This has worked for me. 
Thank you! 

PS: Using Windows 10 22H2, Python 3.10.11, local installation.

iondinescu, 2025-01-19:
This regionale seettings also helped me, win11 pro, python 3.11 [image]

enebybrygg, 2026-01-28:
That same regional UTC-8 setting helped me as well using a standalone ESPHome with web dashboard, running on a windows 11.