esphome/issues #5387: TMP1075: device ID checks prevent use with TMP1075N

ducky64, 2024-01-16:
### The problem

Per the [TMP1075 datasheet](https://www.ti.com/lit/ds/symlink/tmp1075.pdf) Figure 9-2 (page 16), the device ID register is not available on the TMP1075N devices. The current code checks the device ID and marks the component failed if a valid device ID is not read out, which means it fails when used with the TMP1075N.

I did a quick'n'dirty modification of the TMP1075 component code to remove the device ID checks and was able to read temperature from two TMP1075Ns on the bus. There's probably a more elegant solution than just completely removing the device ID checks, perhaps a configuration entry to specify the device variant, or to directly skip the device ID check.

I can implement / PR one of the above if desired.

### Which version of ESPHome has the issue?

2023.10.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

custom ESP32-S3

### Component causing the issue

tmp1075

antsundq, 2025-02-24:
I run into this problem and figured out the same solution as described in this issue. As mentioned above, the device ID query is not implemented for the TMP1075N. I fixed this by modifying the component:

```
  if (die_id != EXPECT_DIEID) {
    ESP_LOGW(TAG, "'%s' - unexpected ID 0x%x found, expected 0x%x", this->name_.c_str(), die_id, EXPECT_DIEID);
    //this->mark_failed();
    return;
  }```

I think this should be fixed and I can help validate if a PR is submitted.

ssieb, 2025-02-25:
Please test the PR.

antsundq, 2025-03-02:
Done, it worked well on my setup (TMP1075N using an ESP32).

closed 2025-03-02 by ssieb