esphome/issues #5389: S3 box 3 - idf_4.4._freertos.patch not applied when compiled with cli

BigBobbas, 2024-01-16:
### The problem

After compiling and uploading firmware I get an error in the logs 
`[09:55:55]Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_4.4._freertos.patch" first`
(more log info included)
The device runs and boots, however there appears to be no audio support. If I compile using the addon then it's all fine and works as expected.

### Which version of ESPHome has the issue?

2023.12.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32-IDF

### Board

ESP32 S3 Box 3

### Component causing the issue

esp-idf, i2s

### Example YAML snippet

```yaml
using the firmware from projects page
```

### Anything in the logs that might be useful for us?

```txt
[09:55:55][D][esp-idf:000]: I (15387) AUDIO_ELEMENT: [i2s-0x3d036ffc] Element task created
[09:55:55]
[09:55:55][D][esp-idf:000]: I (15389) AUDIO_THREAD: The filter task allocate stack on external memory
[09:55:55]
[09:55:55][D][esp-idf:000]: E (15391) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
[09:55:55]Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_4.4._freertos.patch" first
[09:55:55]
[09:55:55]
[09:55:55][D][esp-idf:000]: E (15395) AUDIO_THREAD: Error creating RestrictedPinnedToCore filter
[09:55:55]
[09:55:55][D][esp-idf:000]: E (15397) AUDIO_ELEMENT: [filter] audio_thread_create failed
[09:55:55]
[09:55:55][D][esp-idf:000]: I (15399) AUDIO_ELEMENT: [raw-0x3d0372a0] Element task created
[09:55:55]
[09:55:55][D][esp-idf:000]: I (15401) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:16628551 Bytes, Inter:90052 Bytes, Dram:90052 Bytes
[09:55:55]
[09:55:55]
[09:55:55][D][esp-idf:000]: I (15403) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
[09:55:55]
[09:55:55][D][esp-idf:000]: W (15405) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_RESUME
[09:55:55]
[09:55:55][D][esp-idf:000]: E (15407) AUDIO_PIPELINE: audio_pipeline_resume failed
[09:55:55]
[09:55:55][D][esp-idf:000]: W (15442) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE
```

Windyo, 2024-05-13:
I also have this issue. This results in the BOX-3 not having any audio.

Results in same issue as https://github.com/home-assistant/core/issues/108330 but for different cause.

Note that compiling on server in my case just isn't possible - I don't have the firepower on HA.

Windyo, 2024-05-13:
Switching to linux fixed it.
Dunno why.
TL;DR "just use a linux VM to compile".

KTibow, 2024-07-14:
Had this problem on a real Linux machine. Fixed it with `~/.platformio/packages/framework-espidf; git apply ~/Documents/Apps/esphome/.esphome/build/esp32-s3-box-e81168/esp_adf_patches/idf_v4.4_freertos.patch` (command will differ for you).