esphome/issues #5390: GPIO light toggles during boot

g-kiss, 2024-01-16:
### The problem

The "early_pin_init: false" does not work with light component.

In this case the relay toggles "off" for a short time at boot (when relay was "on" at boot):

```
esp8266:    
  board: esp01_1m
  early_pin_init: false

output:    
  - platform: gpio
    id: shelly_relay
    pin: GPIO4

light:
  - platform: binary
    id: shelly_light
    name: ${light_entity_id}
    output: shelly_relay
    restore_mode: "RESTORE_DEFAULT_OFF"

```

But in this case works well: the relay stays "on" state during boot:

```
esp8266:    
  board: esp01_1m
  early_pin_init: false

switch:
  - platform: gpio
    id: shelly_switch
    name: ${device_name}_switch
    pin: GPIO4
    restore_mode: "RESTORE_DEFAULT_OFF"

```

### Which version of ESPHome has the issue?

2023.12.06

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

shelly

### Component causing the issue

light

kuzabi, 2024-09-22:
Hi,
Same here on Sonoffs, when using light:, "early_pin_init" doesn't fulfills its role.
@rainero84 would you be able to look at the code?

pedropaislopes, 2024-11-21:
hi! I can confirm the same behavior with Sonoff Mini: "early_pin_init" works as expected only for switches, not for lights.