esphome/issues #5391: Since update to version 2023.12.6 get error xtensa-lx106-elf-g++ not found

ailiady, 2024-01-17:
### The problem

Since update to version 2023.12.6 from 2023.12.5 I get error "sh 1 xtensa-lx106-elf-g++ not found" when trying to update devices.
Update to 2023.12.7 didn't fix it.

### Which version of ESPHome has the issue?

Add-on version: 2023.12.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant Core: 2024.1.3

### What platform are you using?

ESP8266

### Board

Raspberrypi 4B

### Anything in the logs that might be useful for us?

```txt
Add-on version: 2023.12.5
 There is an update available for this add-on!
 Latest add-on version: 2023.12.7
 Please consider upgrading as soon as possible.
 System: Debian GNU/Linux 12 (bookworm)  (armv7 / raspberrypi4)
 Home Assistant Core: 2024.1.3
 Home Assistant Supervisor: 2024.01.0.dev0201
```

SaschaMuehl, 2024-02-13:
Same here. it seems to be a raspberry pi 64bit Problem:
https://github.com/esphome/issues/issues/1386#issuecomment-1073117862

ailiady, 2024-02-14:
Looks like it's a problem on Home Assistant Supervised.
I had to move to Home Assistant OS on my Raspberry Pi4B, and there I don't have errors with ESPHome.