esphome/issues #5392: 2023.12.7 - ESP32 s3 Box 3 - no longer detects wakeword

BigBobbas, 2024-01-17:
### The problem

updated to 12.7 and s3 box now starts VAD and detects on first occaision of speech, but doesn't proceed to detect wakeword and then just stops. i've tried re-installing from projects page but no change. Have tested with M5 Echo and that's working ok.
If i roll back to 2023.12.6 i then can't compile as it states that microphone: requires 12.7

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32-IDF

### Board

ESP32 S3 Box 3

### Component causing the issue

esp_adf

### Example YAML snippet

```yaml
firmware from projects page
```

### Anything in the logs that might be useful for us?

```txt
no errors
```

### Additional information

unable to rollback to 12.6 as produces an error on compile stating microphone component requires v2023.12.7

piamay, 2024-01-18:
I have the same problem. The log of my esp32 S3 Box 3 only says that the detection of wake_word in context of voice_assistant failed.

Pat0856, 2024-01-18:
Meme probleme pour moi avec cette version 2023.12.7. Wake_word n'est plus détecté et pas de retour possible vers la version 2023.12.6

closed 2024-01-19 by jesserockz