esphome/issues #5395: DHT22 reporting wrong temperature when below 32F

xlogic11, 2024-01-17:
### The problem

When the temperature goes below 32F the sensor reports the wrong temperature. If the Temp goes above everything is normal

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP32

### Board

KeeYees ESP32S ESP32

### Component causing the issue

DHT22

### Anything in the logs that might be useful for us?

```txt
[10:42:23][C][dht:018]:   Pin: GPIO33
[10:42:23][C][dht:020]:   Auto-detected model: DHT22
[10:42:23][C][dht:027]:   Update Interval: 60.0s
[10:42:23][C][dht:029]:   Temperature 'Temperature'
[10:42:23][C][dht:029]:     Device Class: 'temperature'
[10:42:23][C][dht:029]:     State Class: 'measurement'
[10:42:23][C][dht:029]:     Unit of Measurement: '°C'
[10:42:23][C][dht:029]:     Accuracy Decimals: 1
[10:42:23][C][dht:030]:   Humidity 'Humidity'
[10:42:23][C][dht:030]:     Device Class: 'humidity'
[10:42:23][C][dht:030]:     State Class: 'measurement'
[10:42:23][C][dht:030]:     Unit of Measurement: '%'
[10:42:23][C][dht:030]:     Accuracy Decimals: 0
[10:42:23][C][captive_portal:088]: Captive Portal:
[10:42:23][C][mdns:115]: mDNS:
[10:42:23][C][mdns:116]:   Hostname: garage
[10:42:23][C][ota:097]: Over-The-Air Updates:
[10:42:23][C][ota:098]:   Address: 192.168.0.72:3232
[10:42:23][C][ota:101]:   Using Password.
[10:42:23][C][api:139]: API Server:
[10:42:23][C][api:140]:   Address: 192.168.0.72:6053
[10:42:23][C][api:142]:   Using noise encryption: YES
[10:42:30][D][dht:048]: Got Temperature=-3276.4°C Humidity=44.9%
[10:42:30][D][sensor:094]: 'Temperature': Sending state -3276.40015 °C with 1 decimals of accuracy
[10:42:30][D][sensor:094]: 'Humidity': Sending state 44.90000 % with 0 decimals of accuracy
[10:43:30][D][dht:048]: Got Temperature=-3276.5°C Humidity=45.0%
[10:43:30][D][sensor:094]: 'Temperature': Sending state -3276.50000 °C with 1 decimals of accuracy
[10:43:30][D][sensor:094]: 'Humidity': Sending state 45.00000 % with 0 decimals of accuracy
```

ssieb, 2024-01-17:
Most likely related to https://github.com/esphome/esphome/pull/6074.  Can you enable very_verbose logging and paste the dht logs?

AwokenByMyFears, 2024-01-18:
I am having the same issue.

ssieb, 2024-01-18:
So same request to you too.  :-)

AwokenByMyFears, 2024-01-18:
This is probably more than you want. But I don't know what you're looking for.

[22:35:44][C][logger:440]:   Level: VERBOSE
[22:35:44][C][logger:441]:   Log Baud Rate: 115200
[22:35:44][C][logger:443]:   Hardware UART: UART0
[22:35:44][C][i2c.arduino:053]: I2C Bus:
[22:35:44][C][i2c.arduino:054]:   SDA Pin: GPIO4
[22:35:44][C][i2c.arduino:055]:   SCL Pin: GPIO5
[22:35:44][C][i2c.arduino:056]:   Frequency: 10000 Hz
[22:35:44][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[22:35:44][I][i2c.arduino:069]: Results from i2c bus scan:
[22:35:44][I][i2c.arduino:075]: Found i2c device at address 0x69
[22:35:45][C][sen5x:238]: sen5x:
[22:35:45][C][sen5x:239]:   Address: 0x69
[22:35:45][C][sen5x:262]:   Productname: SEN55
[22:35:45][C][sen5x:263]:   Firmware version: 2
[22:35:45][C][sen5x:264]:   Serial number 01.56.51
[22:35:45][C][sen5x:266]:   Auto cleaning interval 604800 seconds
[22:35:45][C][sen5x:271]:   Low RH/T acceleration mode
[22:35:45][C][sen5x:281]:   Update Interval: 10.0s
[22:35:45][C][sen5x:282]:   PM  1.0 'Outside PM1 Sensor'
[22:35:45][C][sen5x:282]:     Device Class: 'pm1'
[22:35:45][C][sen5x:282]:     State Class: 'measurement'
[22:35:45][C][sen5x:282]:     Unit of Measurement: 'µg/m³'
[22:35:45][C][sen5x:282]:     Accuracy Decimals: 2
[22:35:45][C][sen5x:282]:     Icon: 'mdi:chemical-weapon'
[22:35:45][C][sen5x:283]:   PM  2.5 'Outside PM2.5 Sensor'
[22:35:45][C][sen5x:283]:     Device Class: 'pm25'
[22:35:45][C][sen5x:283]:     State Class: 'measurement'
[22:35:45][C][sen5x:283]:     Unit of Measurement: 'µg/m³'
[22:35:45][C][sen5x:283]:     Accuracy Decimals: 2
[22:35:45][C][sen5x:283]:     Icon: 'mdi:chemical-weapon'
[22:35:45][C][sen5x:284]:   PM  4.0 'Outside PM4 Sensor'
[22:35:45][C][sen5x:284]:     State Class: 'measurement'
[22:35:45][C][sen5x:284]:     Unit of Measurement: 'µg/m³'
[22:35:45][C][sen5x:284]:     Accuracy Decimals: 2
[22:35:45][C][sen5x:284]:     Icon: 'mdi:chemical-weapon'
[22:35:45][C][sen5x:285]:   PM 10.0 'Outside PM10 Sensor'
[22:35:45][C][sen5x:285]:     Device Class: 'pm10'
[22:35:45][C][sen5x:285]:     State Class: 'measurement'
[22:35:45][C][sen5x:285]:     Unit of Measurement: 'µg/m³'
[22:35:45][C][sen5x:285]:     Accuracy Decimals: 2
[22:35:45][C][sen5x:285]:     Icon: 'mdi:chemical-weapon'
[22:35:45][C][sen5x:286]:   Temperature 'Outside Temperature'
[22:35:45][C][sen5x:286]:     Device Class: 'temperature'
[22:35:45][C][sen5x:286]:     State Class: 'measurement'
[22:35:45][C][sen5x:286]:     Unit of Measurement: '°C'
[22:35:45][C][sen5x:286]:     Accuracy Decimals: 2
[22:35:45][C][sen5x:286]:     Icon: 'mdi:thermometer'
[22:35:45][C][sen5x:287]:   Humidity 'Outside Humidity'
[22:35:45][C][sen5x:287]:     Device Class: 'humidity'
[22:35:45][C][sen5x:287]:     State Class: 'measurement'
[22:35:45][C][sen5x:287]:     Unit of Measurement: '%'
[22:35:45][C][sen5x:287]:     Accuracy Decimals: 2
[22:35:45][C][sen5x:287]:     Icon: 'mdi:water-percent'
[22:35:45][C][sen5x:288]:   VOC 'Outside VOC Index'
[22:35:45][C][sen5x:288]:     Device Class: 'volatile_organic_compounds'
[22:35:45][C][sen5x:288]:     State Class: 'measurement'
[22:35:45][C][sen5x:288]:     Unit of Measurement: ''
[22:35:45][C][sen5x:288]:     Accuracy Decimals: 0
[22:35:45][C][sen5x:288]:     Icon: 'mdi:radiator'
[22:35:45][C][sen5x:289]:   NOx 'Outside NOx Index'
[22:35:45][C][sen5x:289]:     Device Class: 'nitrous_oxide'
[22:35:45][C][sen5x:289]:     State Class: 'measurement'
[22:35:45][C][sen5x:289]:     Unit of Measurement: ''
[22:35:45][C][sen5x:289]:     Accuracy Decimals: 0
[22:35:45][C][sen5x:289]:     Icon: 'mdi:radiator'
[22:35:45][C][captive_portal:088]: Captive Portal:
[22:35:45][C][mdns:115]: mDNS:
[22:35:45][C][mdns:116]:   Hostname: outside-multi-sensor
[... 242 lines cut ...]
[22:37:50][D][sensor:093]: 'Outside Humidity': Sending state 55.27000 % with 2 decimals of accuracy
[22:37:50][V][sensor:043]: 'Outside VOC Index': Received new state 431.000000
[22:37:50][D][sensor:093]: 'Outside VOC Index': Sending state 431.00000  with 0 decimals of accuracy
[22:37:50][V][sensor:043]: 'Outside NOx Index': Received new state 1.000000
[22:37:50][D][sensor:093]: 'Outside NOx Index': Sending state 1.00000  with 0 decimals of accuracy
[22:37:50][W][component:214]: Component sen5x.sensor took a long time for an operation (0.16 s).
[22:37:50][W][component:215]: Components should block for at most 20-30ms.
[22:38:00][V][sensor:043]: 'Outside PM1 Sensor': Received new state 8.700000
[22:38:00][D][sensor:093]: 'Outside PM1 Sensor': Sending state 8.70000 µg/m³ with 2 decimals of accuracy
[22:38:00][V][sensor:043]: 'Outside PM2.5 Sensor': Received new state 9.200000
[22:38:00][D][sensor:093]: 'Outside PM2.5 Sensor': Sending state 9.20000 µg/m³ with 2 decimals of accuracy
[22:38:00][V][sensor:043]: 'Outside PM4 Sensor': Received new state 9.300000
[22:38:00][D][sensor:093]: 'Outside PM4 Sensor': Sending state 9.30000 µg/m³ with 2 decimals of accuracy
[22:38:00][V][sensor:043]: 'Outside PM10 Sensor': Received new state 9.400000
[22:38:00][D][sensor:093]: 'Outside PM10 Sensor': Sending state 9.40000 µg/m³ with 2 decimals of accuracy
[22:38:00][V][sensor:043]: 'Outside Temperature': Received new state 326.149994
[22:38:00][D][sensor:093]: 'Outside Temperature': Sending state 326.14999 °C with 2 decimals of accuracy
[22:38:00][V][sensor:043]: 'Outside Humidity': Received new state 55.299999
[... cut ...]
[22:38:00][V][sensor:043]: 'Outside VOC Index': Received new state 431.000000
[22:38:00][D][sensor:093]: 'Outside VOC Index': Sending state 431.00000  with 0 decimals of accuracy
[22:38:00][V][sensor:043]: 'Outside NOx Index': Received new state 1.000000
[22:38:00][D][sensor:093]: 'Outside NOx Index': Sending state 1.00000  with 0 decimals of accuracy
[22:38:00][W][component:214]: Component sen5x.sensor took a long time for an operation (0.16 s).
[22:38:00][W][component:215]: Components should block for at most 20-30ms.

ssieb, 2024-01-18:
This is not the same issue.  It's not even the same sensor.  That issue is https://github.com/esphome/issues/issues/5365 which is fixed in dev with https://github.com/esphome/esphome/pull/6082.  You can use that PR as an external component until the next release.

AwokenByMyFears, 2024-01-18:
Sorry about that. I was told it was a DHT22 inside the SEN55 but it turns out it's an SHT40. Thanks for the link.