esphome/issues #5400: Bootloop when adding remote_transmitter on esp s3 mini

RagingCub, 2024-01-17:
### The problem

Compilation is successful, but the board never successfully boot and gets stuck in a bootloop.
When I remove the remote_transmitter section then the board boots no problem.

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

lolin_s3_mini

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
esphome:
  name: ir-remote-minisplit
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git
    board: lolin_s3_mini
    framework: espidf
    board_build.extra_flags:
      - "-include \"src/fix.h\""
  includes:
    - fix.h

esp32:
  variant: esp32s3
  board: lolin_s3_mini
  framework:
    type: esp-idf
    version: 5.1.2

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  output_power: 10

remote_transmitter:
  pin: 16
  carrier_duty_percent: 50%
```
```

### Anything in the logs that might be useful for us?

```txt
...
I (318) main_task: Calling app_main()
I (463) main_task: Returned from app_main()
[I][logger:351]: Log initialized
[C][ota:473]: There have been 9 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...

assert failed: rmt_ll_tx_set_carrier_high_low_ticks /IDF/components/hal/esp32s3/include/hal/rmt_ll.h:382 (high_ticks >= 1 && high_ticks <= 65536 && low_ticks >= 1 && low_ticks <= 65536 && "out of ran

Backtrace: 0x40376eaa:0x3fca7d00 0x4037c879:0x3fca7d20 0x40382c15:0x3fca7d40 0x42017ce7:0x3fca7e60 0x42018532:0x3fca7e90 0x4200965f:0x3fca7f00 0x420096ba:0x3fca7f50 0x4207eb19:0x3fca7f70 0x4207eb97:0x3fca7f90 0x4200d3ed:0x3fca7fb0 0x4200e931:0x3fca7ff0 0x42006f67:0x3fca80c0

ELF file SHA256: 74f288080b563837

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403768a7
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
...
```
```

### Additional information

the file `fix.h` is a fix taken from this issue #3564

Takst3in, 2024-08-01:
Did you find a solution here?

iannisimo, 2024-08-06:
> Did you find a solution here?

You can try out esphome/esphome#7176 , it should solve the issue

closed 2024-08-06 by kbx81