esphome/issues #5401: increase BME280 temperature precision

jxl77, 2024-01-18:
### The problem

In ESPhome Bme280 library, the temperature is limited to 0.01°C precision.
This is beyond BME280 accuracy, but not beyond BME280 precision. In the BME datasheet, the mean RMS error at 16x is 
given at 0.002°C. 

I like  to record sensor values at a precision that is beyond the RMS error. In this case, it mean having 0.001°C precision. 

This can be obtain at practically no cost. 
in
`float BME280Component::read_temperature_(const uint8_t *data, int32_t *t_fine) {
`
Change the lines 
 ` float const temperature = (*t_fine * 5 + 128) >> 8;
  return temperature / 100.0f;
`
to 
 ` return ((float) (*t_fine * 5 + 128))/ 25600;
`

The problem is only on temperature. Pressure and humidity are given with all decimals.
The user can thus  choose afterwards in ESPHOME its desired precision
`[12:15:05][D][sensor:094]: 'BME280 Temperature': Sending state 12.20000 °C with 3 decimals of accuracy
[12:15:05][D][sensor:094]: 'BME280 Pressure': Sending state 996.22375 hPa with 2 decimals of accuracy
[12:15:05][D][sensor:094]: 'BME280 Humidity': Sending state 53.23535 % with 2 decimals of accuracy
`

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Lolin32

### Component causing the issue

BME280

yousaf465, 2024-01-23:
fix the platform name in your PR too. 
https://discord.com/channels/429907082951524364/1199408893600735262/1199413620212252673

DAVe3283, 2024-01-26:
Looks like @jxl77 beat me to it. This was addressed by esphome/esphome#6124, IMO can be closed now.,

jxl77, 2024-01-27:
bmp280 and bme280 libraries  have been updated to allow this increase in precision.

closed 2024-01-27 by jxl77