esphome/issues #5402: ESPHome 2023.12.7 - Error compiling not found *.cpp

ss78768, 2024-01-18:
### The problem

Hi All,

After the last update, the compilation mechanism for ESPHome broke down.

Log:
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2_service.cpp.o] Error 127

I rely on your help.

Regards,
Sergey

### Which version of ESPHome has the issue?

2023.12.7 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP32

### Board

ESPHome

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.7
INFO Reading configuration /config/esphome/remote-bluetooth.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing remote-bluetooth (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/remote-bluetooth/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================= [FAILED] Took 10.29 seconds =========================
```

rp3tya, 2024-01-19:
Confirming this on a NanoPi R6S.

cmchauvin, 2024-01-30:
Same issue for me. Seeing "xtensa-esp32-elf-g++: not found" when I try to update. ESPHome 2023.12.9 add-on running on Home Assistant 2024.1.5 using RPI 4

Unfocused, 2024-02-01:
Is this a duplicate of #3092 ?