esphome/issues #5403: esphome color issues with ili9342 Cheap Yellow Display CYD

Grey-Lancaster, 2024-01-18:
### The problem

The problem is images only show 3 colors maybe/  Not at all like the ili9341

Color test BGR which displays text properly
[image]

Color test RGB
[image]

Lolinn ili9341 color test
[image]

Lolin ili 9341 image
[image]

CYD ILI9342 image
[image]

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

CYD ESP32-2432S028R

### Component causing the issue

ili9xxx

### Example YAML snippet

```yaml
esphome:
  name: cyd1
  friendly_name: CYD1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "hkm95e5mVzQKWzzEmzmOLeM5IjAioXf7oxvEfKYkoQw="

ota:
  password: "d570ab1a6e51f564abf18152ff316734"

wifi:
  networks:

  - ssid: "thegreyfox"
    password: ""
  - ssid: "shop2"
    password: "mine0313"
  - ssid: "wled"  
    password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Cyd1 Fallback Hotspot"
    password: "EY5ubMujtVvU"

captive_portal:

    
time:
  - platform: homeassistant
    id: esptime
sensor:
  - platform: homeassistant
    id: outside_temperature
    entity_id: sensor.office_outside_temp_2
    internal: true

  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.big_room_temperature
    internal: true

  - platform: homeassistant
    id: nannie_out_temperature
    entity_id: sensor.outdoor_temperature
    internal: true

  - platform: homeassistant
    id: nannie_in_temperature
    entity_id: sensor.indoor_temperature
    internal: true

  - platform: homeassistant
    id: office_temperature
    entity_id: sensor.office_temp
    internal: true

color:
  - id: my_light_red
    hex: FFFF00  #yello
  - id: my_black
    hex: '000000'  # Hex code for black
  - id: white
    hex: ffffff
  - id: blue
    hex: 0000FF    

graph:
  - id: officegraph
    sensor: outside_temperature
    duration: 3h
    width: 220
    height: 180
    x_grid: 30min
    y_grid: 5.0
    max_value: 55.0
    min_value: 15.0
    color: blue
  - id: officegraph2
    sensor: inside_temperature
    duration: 1h
    width: 220
    height: 180
    x_grid: 10min
    y_grid: 5.0
    max_value: 70.0
    min_value: 15.0
    color: blue
font:
  # - file: 'Roboto.ttf'
  #    id: font1
  #    size: 26

  - file:
      type: gfonts
      family: Roboto
      weight: 900
    id: font1
    size: 26    
  - file: 'Arimo-Regular.ttf'
    id: arimo48
    size: 48
    glyphs: " .,°0123456789CF"
  - file: 'Arimo-Regular.ttf'
    id: arimo14
    size: 14
    glyphs: "0123456789"     
image:
  - file: http://192.168.1.50:8123/local/dad2.PNG
    id: my_image
    resize: 120x120
    type: RGB24

# ============================================================ 
# Home Assistant related setup
#
light:
  - platform: monochromatic
    output: backlight_pwm
    name: Display Backlight
    id: backlight
    restore_mode: ALWAYS_ON

# ============================================================ 
# Hardware related setup
#
# Setup SPI for the display. The ESP32-2432S028R uses separate SPI buses for display and touch
spi:
  - id: tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

# Setup a pin to control the backlight
output:
  - platform: ledc
    pin: GPIO21
    id: backlight_pwm

# Setup the ili9xxx platform
#
# Display is used as 240x320 by default so we rotate it to 90°
display:
  - platform: ili9xxx
    model: ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 90
    color_order: RGB
    id: "my_display"    
    pages:
      - id: page1    
        lambda: |-
          it.fill(id(my_black));   
          it.print(20, 0, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Hello WIZARD!");

          it.print(20, 30, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Time:");
          char buffer[15];
          auto time = id(esptime).now();
          int hour = time.hour % 12; // convert to 12-hour format
          hour = hour ? hour : 12; // handle midnight and noon
          sprintf(buffer, "%d:%02d", hour, time.minute);
          it.print(100, 30, id(font1), id(my_light_red), TextAlign::TOP_LEFT,  buffer);

          it.printf(25, 55, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Inside:");
          it.printf(25, 80, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Outside:");  
          it.printf(2, 105, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Nannie Out:");      
          it.printf(2, 130, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Nannie In:");
          it.printf(2, 155, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Office:");

          if (id(inside_temperature).has_state()) {      
          it.printf(120, 55, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "%.1f°", id(inside_temperature).state);
          }      
          if (id(outside_temperature).has_state()) {      
          it.printf(130, 80, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "%.1f°", id(outside_temperature).state);
          } 
          if (id(nannie_out_temperature).has_state()) {      
          it.printf(145, 105, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "%.1f°", id(nannie_out_temperature).state);
          } 
          if (id(nannie_in_temperature).has_state()) {      
          it.printf(145, 130, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "%.1f°", id(nannie_in_temperature).state);
          } 
          if (id(office_temperature).has_state()) {      
          it.printf(145, 155, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "%.1f°", id(office_temperature).state);
          } 
          it.image(60, 200, id(my_image)); 
      - id: page2
        lambda: |-
          it.fill(id(Color::BLACK));
          it.print(0, 60, id(arimo14), TextAlign::TOP_LEFT, "75");
          it.print(0, 105, id(arimo14), TextAlign::CENTER_LEFT, "65");
          it.print(0, 150, id(arimo14), TextAlign::CENTER_LEFT, "55");
          it.print(0, 195, id(arimo14), TextAlign::CENTER_LEFT, "45");
          it.print(0, 240, id(arimo14), TextAlign::BOTTOM_LEFT, "35");
          if (id(outside_temperature).has_state()) {
          it.printf(160, 0, id(arimo48), id(blue),TextAlign::TOP_CENTER, "%.1f °C", id(inside_temperature).state);
          }
          it.graph(20, 60, id(officegraph2));

      - id: page3
        lambda: |-
          it.fill(id(Color::BLACK));
          it.print(0, 60, id(arimo14), TextAlign::TOP_LEFT, "55");
          it.print(0, 105, id(arimo14), TextAlign::CENTER_LEFT, "45");
          it.print(0, 150, id(arimo14), TextAlign::CENTER_LEFT, "35");
          it.print(0, 195, id(arimo14), TextAlign::CENTER_LEFT, "25");
          it.print(0, 240, id(arimo14), TextAlign::BOTTOM_LEFT, "15");
          if (id(outside_temperature).has_state()) {
          it.printf(160, 0, id(arimo48), id(blue),TextAlign::TOP_CENTER, "%.1f °F", id(outside_temperature).state);
          }
          it.graph(20, 60, id(officegraph));
          

                    
                

interval:
  - interval: 5s
    then:
      - display.page.show_next: my_display
      - component.update: my_display
```

nielsnl68, 2024-01-18:
Could you tell which images i see with the test patron?
From what i get you tested with a ili9341 and ili9342 is it not?

nielsnl68, 2024-01-18:
Could you tell which image belongs with which display ili9341 or ili9342?

Grey-Lancaster, 2024-01-19:
Sorry, I do not understand your question.
the first 2 pics are with
display:
  - platform: ili9xxx
    model: ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 90
    color_order: RGB
    id: "my_display" 

and I chaged to bgr for one
Grey

Grey-Lancaster, 2024-01-19:
I am assuming the ili9341 test pattern is correct because all works fine,
Neither of the ili9342 test patterns match the 9341 is why I think I have issues

nielsnl68, 2024-01-19:
can you show me again the test patronen van de ili9342 display's?
It is me not clear which image belongs to those displays,

Grey-Lancaster, 2024-01-19:
display:
  - platform: ili9xxx
    model: ili9342
    color_order: RGB

[image]

display:
  - platform: ili9xxx
    model: ili9342
    color_order: BGR

[image]

nielsnl68, 2024-01-20:
Are you sure it is a ili9342 and not a ili9341?  Anyway on TFT_eSPI there is an other config file for this kind of displays.
I will create a PR for this one in a moment so you can see of it will work or not.

nielsnl68, 2024-01-20:
Hmm just wondering, did you try to add `invert_colors: true` to the display?

Grey-Lancaster, 2024-01-20:
It is a 9342, 9341 does not fill  the screen, is rotated and scrambled.
   invert_colors: true  does that exactly.  My black becomes white

The problem is my image.  It is like the 9342 only has 4bytes of color, not 256

this is with invert=true

   
[image]

nielsnl68, 2024-01-20:
Okey, that was not the solution then :D 

lets try the init code i found on TFT_eSPI: 

```yaml
external_components:
  - source: github://pr#6125
    refresh: 1s
    components: [ ili9xxx ]

display:
  - platform: ili9xxx
    model: cyd_ili9342
    ...
```
let me know how it go's.

Grey-Lancaster, 2024-01-20:
Not sure if I am doing this right?  I also have an error about

```
esphome:
  name: cyd1
  friendly_name: CYD1

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://pr#6125
    refresh: 1s
    components: [ ili9xxx ] 
```  

```
INFO ESPHome 2023.12.8
INFO Reading configuration /config/esphome/cyd1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/6125/head
ERROR Unable to import component ili9xxx.display:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/58ba9074/esphome/components/ili9xxx/display.py", line 7, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_MIRROR_X' from 'esphome.const' (/esphome/esphome/const.py)
INFO Detected timezone 'America/New_York
```

nielsnl68, 2024-01-20:
okay, i think that has todo with the esphome branch, could you setup the DEV version of esphome?

Grey-Lancaster, 2024-01-20:
let me google :-)

nielsnl68, 2024-01-20:
Do you not run Home Assistant OS? when you you can install the esphome DEV add-on

Grey-Lancaster, 2024-01-20:
No  download errors

```
esp32:
  board: esp32dev
  framework:
    type: arduino
external_components:
  - source: github://pr#6125
    refresh: 1s
    components: [ ili9xxx ]
```

```
display:
  - platform: ili9xxx
    model: cyd_ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 90
    color_order: RGB
    id: "my_display"    
    pages:
```

```
INFO ESPHome 2024.1.0-dev
INFO Reading configuration /config/esphome/cyd1.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/6125/head
INFO Detected timezone 'America/New_York'
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing cyd1 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- SPI @ 2.0.0
Compiling .pioenvs/cyd1/src/main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:1574:20: error: expected type-specifier
   my_display = new ili9xxx::ILI9XXXCYD_ILI9342();
                    ^~~~~~~
*** [.pioenvs/cyd1/src/main.cpp.o] Error 1
========================= [FAILED] Took 11.13 seconds =========================
```

nielsnl68, 2024-01-20:
Oww my error i renamed something that i forgot to update somewhere else.

Just try again please.

Grey-Lancaster, 2024-01-20:
Glad the error meant something to you :-)
Colors ar still off and my text is reversed (mirror image)
Let me load up the test pattern for you
```

display:
  - platform: ili9xxx
    model: cyd_ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 0
    color_order: RGB
    id: "my_display"  
```

nielsnl68, 2024-01-20:
you should not need to use `color_order: RGB`

Is there any color difference between the we did have before?

nielsnl68, 2024-01-20:
Oww could you add `invert_colors: false` as well.

Grey-Lancaster, 2024-01-20:
the image was still bland
invert colors false or true?

nielsnl68, 2024-01-20:
false now.

Grey-Lancaster, 2024-01-20:
aw i messed up the color test
coming next
```
display:
  - platform: ili9xxx
    model: cyd_ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 0
    invert_colors: false
    id: "my_display" 
```
[image]

nielsnl68, 2024-01-20:
that does not look good :(

Grey-Lancaster, 2024-01-20:
```
display:
  - platform: ili9xxx
    model: cyd_ili9342
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    rotation: 0
    invert_colors: false
    id: "my_display"  
``` 

[image]

nielsnl68, 2024-01-20:
okay i made some small changes maybe this fixed it.

Grey-Lancaster, 2024-01-20:
Seems like others around are tweaking the ili9341 [LCD wiki](http://www.lcdwiki.com/Main_Page#SPI_Display) has no mention of an ili9342  I am just using 9342 as it all works except for the color depth.  I have no documentation that says that is what it is.
Can't find it now but saw somewhere v2 was with the usb c.
This page as a vendor config file that makes no sense to me but perhaps it codes to you
https://github.com/rzeldent/platformio-espressif32-sunton/blob/main/esp32-2432S028Rv2.json

Thanks for your help,
Grey

Grey-Lancaster, 2024-01-20:
Did you save your changes? :-)
I cannot tell any difference.
invert_colors true has the right colors but the text is still backwards

nielsnl68, 2024-01-20:
let  me look at what that link gives as init string it. it will take a while

nielsnl68, 2024-01-20:
Okay, let us see of this init code will work.

Grey-Lancaster, 2024-01-20:
all I have to do is install again correct?  The  refresh: 1s is supposed to pull the new code?
I cannot see any changes on the display

nielsnl68, 2024-01-20:
that is correct.

When this is still not working, could you ask the vendor from who you ordered this display for example code?

Grey-Lancaster, 2024-01-20:
I will have to try their code. It was displaying something when I first plugged it in
I have one of the files from this link but the password did not work for another.  I tried chatting with them but got nowhere.
Will check back tomorrow and let you know
Thanks again,
Grey

ESP32-2432S028R download link： http://www.jczn1688.com/zlxz
Download password: jczn1688

Grey-Lancaster, 2024-01-20:
The examples were not for an esp32 is why I did not make much progress

```
// IMPORTANT: LCDWIKI_SPI LIBRARY MUST BE SPECIFICALLY
// CONFIGURED FOR EITHER THE TFT SHIELD OR THE BREAKOUT BOARD.

//This program is a demo of how to display picture and 
//how to use rotate function to display string.

//when using the BREAKOUT BOARD only and using these hardware spi lines to the LCD,
//the SDA pin and SCK pin is defined by the system and can't be modified.
//if you don't need to control the LED pin,you can set it to 3.3V and set the pin definition to -1.
//other pins can be defined by youself,for example
//pin usage as follow:
//             CS  DC/RS  RESET  SDI/MOSI  SDO/MISO  SCK  LED    VCC     GND    
//Arduino Uno  A5   A3     A4      11        12      13   A0   5V/3.3V   GND

//Remember to set the pins to suit your display module!

/***********************************************************************************
* @attention
*
* THE PRESENT FIRMWARE WHICH IS FOR GUIDANCE ONLY AIMS AT PROVIDING CUSTOMERS
* WITH CODING INFORMATION REGARDING THEIR PRODUCTS IN ORDER FOR THEM TO SAVE
* TIME. AS A RESULT, QD electronic SHALL NOT BE HELD LIABLE FOR ANY
* DIRECT, INDIRECT OR CONSEQUENTIAL DAMAGES WITH RESPECT TO ANY CLAIMS ARISING
* FROM THE CONTENT OF SUCH FIRMWARE AND/OR THE USE MADE BY CUSTOMERS OF THE 
* CODING INFORMATION CONTAINED HEREIN IN CONNECTION WITH THEIR PRODUCTS.
**********************************************************************************/
#include <LCDWIKI_GUI.h> //Core graphics library
#include <LCDWIKI_SPI.h> //Hardware-specific library

//paramters define
#define MODEL ILI9341
#define CS   A5    
#define CD   A3
#define RST  A4
```
#define LED  A0   //if you don't need to control the LED pin,you should set it to -1 and set it to 3.3V

nielsnl68, 2024-01-20:
is that password correct? i cant get that file being downloaded with it.

Grey-Lancaster, 2024-01-20:
I just realized the file I downloaded on page 2 cannot be right if it is talking about uno and this is an esp32.

Yes, that was the problem I had with the password.  I was chatting with the vendor for 15 mins and stepped away to come back and said I was gone too long and it closed.  I will try them again

Grey-Lancaster, 2024-01-20:
I have to leave but I will try this
http://pan.jczn1688.com/1/ESP32%20module

Grey-Lancaster, 2024-01-22:
The files I found are esp-idf and I am not getting anywhere.  From that link above.
Is this helpful?

Step 5: Create a settings file for LVGL
LVGL needs a configuration file; lv_conf.h. This file contains information about the fonts, color depths, default background, styles, etc... The default LVGL template can be found in the LVGL library at the location: lvgl/lv_conf_template.h. This file must be copied to the include directory and renamed to lvgl_conf.h. Also the #if 0 must be removed to enable the file to be included. More information about setting up a project with LVGL can be obtained at [LVGL get-started/quick-overview](https://docs.lvgl.io/master/get-started/quick-overview.html#add-lvgl-into-your-project). I suggest to put the lv_conf.h file in the include directory and set the build flags to specify the location of the file, see below.

Tip

LVGL can also be downloaded manually from: https://github.com/lvgl/lvgl/archive/master.zip to extract the template. The template can also be copied from the demo application.

Important settings are:

Only the R5G6B5 format is supported on these panels.

/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH 16
Because of the SPI interface, the bytes are sent in big endian format so this must be corrected. The RGB panel interface takes care of this by swapping the GPIO lines but for the SPI controllers this is not optimal. More information about this can be found [below](https://github.com/rzeldent/esp32-smartdisplay/tree/main?tab=readme-ov-file#lv_color_16_swap).

/*Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP 1
To have a time reference use the milliseconds reference for the Arduino

/*Use a custom tick source that tells the elapsed time in milliseconds and enables this code.*/
/*It removes the need to manually update the tick with `lv_tick_inc()`)*/
#define LV_TICK_CUSTOM 1
For debugging, enable CPU usage, FPS (Frames per Second) and memory defragmentation

#define LV_USE_PERF_MONITOR 1
#define LV_USE_MEM_MONITOR 1
(Optionally) Include additional fonts.

...
#define LV_FONT_MONTSERRAT_22 1
...
Optionally, only enable widgets that are used to save on code

#define LV_USE_ARC        1
#define LV_USE_BAR        1
#define LV_USE_BTN        1
#define LV_USE_BTNMATRIX  1
...

nielsnl68, 2024-01-22:
okay, i will guide you:
Do you have TFT_eSPI installed? then replace it with files in  `1-Demo\Demo_Arduino\libraries\TFT_eSPI` folder to your `libraries` folder. 
Open `User_Setup_Select.h` in that folder and add a // for `#include <User_Setup.h>` this allows us to define our own in the demo later. Restart the arduino IDE. And do not update the package. not sure of they made other changes.

Now copy the file `1-Demo\Demo_Arduino\3_4-1_TFT_Rainbow\TFT_eSPI bottom layer replacement file` to '1-Demo\Demo_Arduino\3_4-1_TFT_Rainbow\3_4_TFT_Rainbow'

and `#include "User_Setup.h"`  at the top of the `3_4_TFT_Rainbow.ino` and run it.
That should work (i hoop).

when done try the

Grey-Lancaster, 2024-01-22:
Thank you for your patience and help
I was trying to do the idf demo not arduino

I did have to replace the tft_eSPI library as all I got was a blank screen.  It does ~work
the colors are inverted  Black is white, red is blue

[image]

nielsnl68, 2024-01-22:
Could you humor me and show the backside of it.

Grey-Lancaster, 2024-01-22:
https://www.aliexpress.us/item/3256805974330395.html
funny, haha colors do not change? neither makes a difference.  In esphome I need BGR

```
// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red
```

nielsnl68, 2024-01-23:
where do you change those settings? 

Is that an USB-C or micro USB connector on the rightside?

Grey-Lancaster, 2024-01-23:
USB-c in the middle is what I am using micro next to it
user setup.h

[image]

nielsnl68, 2024-01-24:
Can we try the ili9341 one more time with the following display settings:

```yaml
display:
  - platform: ili9xxx
    model: ili9341
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    id: "my_display"   
    dimensions: 320x240
    transform:
       mirror_y: false
       mirror_x: false
       swap_xy: true
```
You can play with the "transform setting" and dimension until your screen looks okay.

nielsnl68, 2024-01-24:
Are those two from the "same" YAML script?

Grey-Lancaster, 2024-01-24:
Basicly just diff pins and boards

Lolin

```
image:
  - file: http://192.168.1.50:8123/local/dad2.PNG
    id: my_image
    resize: 120x120
    type: RGBA

spi:
 clk_pin: 18
 mosi_pin: 23
 miso_pin: 19

°
display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: 14
    dc_pin: 27 
    reset_pin: 33
    data_rate: 20Mhz   
    spi_mode: mode0
    id: "my_display"      
    pages:
```
Suntron CYD with usb c

```
image:
  - file: http://192.168.1.50:8123/local/dad2.PNG
    id: my_image
    resize: 120x120
    type: RGBA

# ============================================================ 
# Home Assistant related setup
#
light:
  - platform: monochromatic
    output: backlight_pwm
    name: Display Backlight
    id: backlight
    restore_mode: ALWAYS_ON

# ============================================================ 
# Hardware related setup
#
# Setup SPI for the display. The ESP32-2432S028R uses separate SPI buses for display and touch
spi:
  - id: tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

# Setup a pin to control the backlight
output:
  - platform: ledc
    pin: GPIO21
    id: backlight_pwm

# Setup the ili9xxx platform
#
# Display is used as 240x320 by default so we rotate it to 90°
display:
  - platform: ili9xxx
    model: ili9341
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    id: "my_display"   
    dimensions: 240X320
    color_order: RGB
    transform:
       mirror_y: false
       mirror_x: false
       swap_xy: true
    pages:
```

Grey-Lancaster, 2024-01-24:
[image]

nielsnl68, 2024-01-25:
Hmm did i never asked for the startup log? 
The only thing i expect that it does not have PSRAM onboard and esphome switch to 8bit colors on this device.

Grey-Lancaster, 2024-01-25:
I have been messing with this for a couple of days and just got it to work.  The image is washed out on it. also.  The developer said his did not look like his.  Suggested reducing/changing the contrast which I have no clue how

Well it does say no psram.  An esp32 wroom does not have it??

[15:31:39][C][light:103]: Light 'Display Backlight'
[15:31:39][C][light:105]:   Default Transition Length: 1.0s
[15:31:39][C][light:106]:   Gamma Correct: 2.80
[15:31:39][C][homeassistant.time:010]: Home Assistant Time:
[15:31:39][C][homeassistant.time:011]:   Timezone: 'EST5EDT,M3.2.0,M11.1.0'
[15:31:39][C][psram:020]: PSRAM:
[15:31:39][C][psram:021]:   Available: NO
[15:31:39][C][graph:389]: Graph for sensor outside_temperature

Grey-Lancaster, 2024-01-25:
Thanks, I have a "normal" CYD due in soon.  Will see how it does.
So I was googling and it says the esp32 wrooms have psram and then I looked and questioned if I had the right board selected in esphome.  My esp32 wrover works well So I dunno, I guess it is just the board because I have tried esphome and https://github.com/rzeldent/esp32-smartdisplay-demo both have lousy images.

Grey

Grey-Lancaster, 2024-01-29:
Still waiting on my new board
Found a slideshow .ino that works fine
does this help

```
  // Fix image quality for CYD2USB
#ifdef TFT_INVERSION_ON
  tft.writecommand(ILI9341_GAMMASET); //Gamma curve selected
  tft.writedata(2);
  delay(120);
  tft.writecommand(ILI9341_GAMMASET); //Gamma curve selected
  tft.writedata(1);
#endif
```

Grey-Lancaster, 2024-01-30:
Got my "normal" CYD today.  This code works (dev mode) but the image is still lousy.  I know the board can display images because the slide show example is beautiful.

Anyway to try that TFT_inversion?

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/Showcases/1-Slideshow

```
esphome:
  name: cyd-v1
  friendly_name: CYD_V1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "t5/ChNs1LJ1f8aB9xrfku8Vpixfw8GenT1VGZtwW6Lk="

ota:
  password: "3366167611974d3b468d70eae3b5c099"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Cyd-V1 Fallback Hotspot"
    password: "C3r4clzj8U07"

captive_portal:
    

    
time:
  - platform: homeassistant
    id: esptime
sensor:
  - platform: homeassistant
    id: outside_temperature
    entity_id: sensor.office_outside_temp_2
    internal: true

  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.big_room_temperature
    internal: true

  - platform: homeassistant
    id: nannie_out_temperature
    entity_id: sensor.outdoor_temperature
    internal: true

  - platform: homeassistant
    id: nannie_in_temperature
    entity_id: sensor.indoor_temperature
    internal: true

  - platform: homeassistant
    id: office_temperature
    entity_id: sensor.office_temp
    internal: true

color:
[... 156 lines cut ...]
          it.printf(20, 160, id(font1), id(my_light_red), TextAlign::TOP_LEFT, "Outside:");        
          }             
      - id: page3
        lambda: |-
          it.fill(id(Color::BLACK));
          it.print(0, 60, id(arimo14), TextAlign::TOP_LEFT, "55");
          it.print(0, 105, id(arimo14), TextAlign::CENTER_LEFT, "45");
          it.print(0, 150, id(arimo14), TextAlign::CENTER_LEFT, "35");
          it.print(0, 195, id(arimo14), TextAlign::CENTER_LEFT, "25");
          it.print(0, 240, id(arimo14), TextAlign::BOTTOM_LEFT, "15");
          if (id(outside_temperature).has_state()) {
          it.printf(160, 0, id(arimo48), id(blue),TextAlign::TOP_CENTER, "%.1f °F", id(outside_temperature).state);
          }
          it.graph(20, 60, id(officegraph));
          

                    
                

interval:
  - interval: 3s
    then:
      - display.page.show_next: my_display
      - component.update: my_display     
```

nielsnl68, 2024-01-30:
It looks like none of the CYD's has PSRAM onboard sadly. I have also some 3.5" CYD's and those have also no PSRAM on board.
so it only works on 8Bit mode using the native Display functions.
As soon the LVGL component is ready it will works in 16bit mode.

closed 2024-08-22 by clydebarrow