esphome/issues #5404: Platformio 6.5.0 Flash Size Warning

DAVe3283, 2024-01-19:
### The problem

tl;dr Platformio 6.5.0 [added a warning if there is a flash size mismatch with IDF config](https://github.com/platformio/platform-espressif32/commit/275f1170dd3e28ccdd2de36bf31e5dacd24a15bb) and ESPHome is not setting IDF flash size, only the board flash size.

Background: I am using ESP32-C6 which is not supported by the recommended Platformio version or ESP-IDF version, and have to run the latest to get it to work. There is no board file for my hardware, so I have to add some custom options:

```yaml
esphome:
  name: esp32c6
  friendly_name: esp32c6
  platformio_options:
    board_upload.maximum_size: 4194304

# OLIMEX ESP32-C6-EVB actually has 512K RAM, 4MB flash
# platformio esp32-c6-devkitm-1 claims 320K RAM which appears wrong per the ESP32-C6 datasheet
# platformio esp32-c6-devkitc-1 has RAM right, but has 8MB flash, we actually have 4MB
# Use devkitc and fix flash

esp32:
  board: esp32-c6-devkitc-1 # at least until https://github.com/platformio/platform-espressif32/issues/1152 is added
  variant: esp32c6          # ditto ↑
  flash_size: 4MB
  framework:
    type: esp-idf
    version: latest # Need at least 5.1 for ESP32-C6
    platform_version: 6.5.0 # 6.4.0 added ESP32-C6 but failing on checksum, 6.5.0 works though!
```

Setting `flash_size: 4MB` doesn't appear to set the appropriate `CONFIG_ESPTOOLPY_FLASHSIZE_xMB=y` variable.
In the build directory, `sdkconfig.esp32c6` appears to have been configured for 2MB (no idea where that came from, the board file would be 8MB, and I specified 4MB):
```
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_32MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_64MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_128MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
# CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE is not set
```

This causes the compile to start off with the new warning:
```
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
```

You can work around the problem by manually specifying the appropriate CONFIG_ESPTOOLPY_FLASHSIZE_xMB value:

```diff
 esp32:
   board: esp32-c6-devkitc-1 # at least until https://github.com/platformio/platform-espressif32/issues/1152 is added
   variant: esp32c6          # ditto ↑
   flash_size: 4MB
   framework:
     type: esp-idf
     version: latest # Need at least 5.1 for ESP32-C6
     platform_version: 6.5.0 # 6.4.0 added ESP32-C6 but failing on checksum, 6.5.0 works though!
+    sdkconfig_options:
+      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

This fixes `sdkconfig.esp32c6` and silences the error:
```
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_2MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_32MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_64MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_128MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
# CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE is not set
```

I imagine that in addition to basing the partitions on the user-selected flash_size, ESPHome needs to set the appropriate CONFIG_ESPTOOLPY_FLASHSIZE_xMB=y flag.

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

Olimex ESP32-C6-EVB

### Component causing the issue

esp32

### Anything in the logs that might be useful for us?

```txt
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
```

luar123, 2024-04-05:
Same for ESP32-H2

felipejfc, 2024-04-18:
Thanks for this. I managed to fix a SHA checksum error I would get during linking with this option

+    sdkconfig_options:
+      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

QuickJack, 2024-06-08:
I have created https://github.com/platformio/platform-espressif32/pull/1396 which should fix RAM size for esp32-c6-devkitm-1.

jschneekloth, 2024-06-08:
Similar issue with esp32-c3-devkitm-1

QuickJack, 2024-06-10:
Well. not exactly. The templates for esp32-c3-devkitc-02 and esp32-c3-devkitm-1 both specify 320k RAM whereas the ESP32-C3 has 400k RAM. So these templates are wrong as well.

kbx81, 2024-12-19:
https://github.com/esphome/esphome/pull/7983 should fix this -- anyone care to test? Try it with:
```yaml
external_components:
  - source: github://pr#7983
    components: [ esp32 ]
```

DAVe3283, 2024-12-19:
I will give this a try tonight, but it sure looks like it should fix the issue, thanks!

closed 2024-12-20 by kbx81