esphome/issues #5405: PZEM-004T - detect electrysyty failure

LevKos, 2024-01-19:
### The problem

The problem
If there is no power to the high-voltage part of the PZEM-004T, it does not output anything, but the platform has processed that if there is nothing, the values do not change. Electrysyty failure cannot be tracked.
I suggest making the following changes to pzemac.cpp:

void MYPZEMAC::on_modbus_data(const std::vector<uint8_t> &data) {
    if (data.size() < 20) {
        ESP_LOGW(TAG, "Invalid size for PZEM AC!");
// start add
        this->voltage_sensor_->publish_state(0);
        this->current_sensor_->publish_state(0);
        this->power_sensor_->publish_state(0);
        this->frequency_sensor_->publish_state(0);
        this->power_factor_sensor_->publish_state(0);
//stop add
        return;
}
...

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

PZEM-004T v3

DAVe3283, 2024-01-19:
I like that idea, but it should publish `NAN` (which displays as unknown) and mark the sensor as failed too. I don't like the idea of publishing 0 because we don't _know_ that power is out, just that communication failed (cable disconnected, opto-isolator failed, etc. all would cause the same behavior).

I will try to put up a PR for this when I have time to test with my PZEM-004T to verify it works as expected.

LevKos, 2024-01-20:
The “not available” state is logical when turned on, but if the measurement results have already been transmitted, then their absence, in most cases, will indicate a lack of power on the high-voltage part.

snoopsbne, 2024-01-31:
I've done something similar except I use a timeout in the publish part of the code. I have several PZEM-004t's on the same modbus/uart and occasionally run into issues with CRC errors etc, so having it report 0 (or NAN) on the first "failure" is a bad idea (for me at least). I had a deeper look at a better way to do it, but my coding skills are not good enough to deal with modifying Modbus and the PZEM-004ts code. Ideally a timeout should be defined/optional on the Modbus side of things and if no value is reported then return 0 or NAN.

LevKos, 2024-01-31:
Thank you.
The presence of errors on the line is most likely caused by the circuitry.
I implemented a circuit with 10к resistor то 3.3V and 1N4448 diodes on TX lines, 3 in parallel, polling every 10 seconds -checksums do not fall out.
I think it is not advisable to programmatically handle physical problems.
[image]