esphome/issues #5406: Key Collector component bugged when timeout is enabled

dave0111, 2024-01-19:
### The problem

I'm using one of those cheap membrane matrix keypads that you can find on amazon, with the following config:

```
matrix_keypad:
  id: keypad1
  rows:
    - pin: 27
    - pin: 26
    - pin: 25
    - pin: 33
  columns:
    - pin: 32
    - pin: 35
    - pin: 34
  keys: "123456789*0#"
  has_diodes: false

key_collector:
  - id: pincode_reader
    source_id: keypad1
    min_length: 6
    max_length: 6
    end_keys: "#"
    end_key_required: true
    clear_keys: "*"
    allowed_keys: "0123456789"
    timeout: 60s
    on_progress:
      - logger.log:
          format: "input progress: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
    on_result:
      - logger.log:
          format: "input result: '%s', started by '%c', ended by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)", "(end == 0 ? '~' : end)" ]
    on_timeout:
      - logger.log:
          format: "input timeout: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
```

Which is almost the same exact config found on the documentation.

The issue is with the key collector, for some reason, whenever I enable a timeout, I get a stream of phantom keystrokes:

```
[14:42:24][D][matrix_keypad:081]: key @ row 1, col 1 pressed
[14:42:24][D][matrix_keypad:086]: key '5' pressed
[14:42:24][D][main:613]: input progress: '555555', started by '~'
[14:42:25][D][matrix_keypad:058]: key @ row 1, col 1 released
[14:42:25][D][matrix_keypad:063]: key '5' released
```

The phantom keystrokes only come from Row 1, Col 1, which is number 5.

Removing the timeout in the Key Collector configuration completely solves the issue.
I also tried with several different timeout values, but the problem seems to not be related with the duration, but rather with the presence of any timeout at all.

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.1.3, Supervisor 2023.12.1

### What platform are you using?

ESP32

### Board

ESP-WROOM-32

### Component causing the issue

key_collector

### Example YAML snippet

```yaml
matrix_keypad:
  id: keypad1
  rows:
    - pin: 27
    - pin: 26
    - pin: 25
    - pin: 33
  columns:
    - pin: 32
    - pin: 35
    - pin: 34
  keys: "123456789*0#"
  has_diodes: false

key_collector:
  - id: pincode_reader
    source_id: keypad1
    min_length: 6
    max_length: 6
    end_keys: "#"
    end_key_required: true
    clear_keys: "*"
    allowed_keys: "0123456789"
    timeout: 60s
    on_progress:
      - logger.log:
          format: "input progress: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
    on_result:
      - logger.log:
          format: "input result: '%s', started by '%c', ended by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)", "(end == 0 ? '~' : end)" ]
    on_timeout:
      - logger.log:
          format: "input timeout: '%s', started by '%c'"
          args: [ 'x.c_str()', "(start == 0 ? '~' : start)" ]
```

### Anything in the logs that might be useful for us?

```txt
[14:46:41][D][text_sensor:064]: 'NGB2 Online': Sending state 'online'
[14:46:41][D][matrix_keypad:081]: key @ row 1, col 1 pressed
[14:46:41][D][matrix_keypad:086]: key '5' pressed
[14:46:41][D][main:613]: input progress: '555555', started by '~'
[14:46:41][D][matrix_keypad:058]: key @ row 1, col 1 released
[14:46:41][D][matrix_keypad:063]: key '5' released
[14:46:43][D][matrix_keypad:081]: key @ row 1, col 1 pressed
[14:46:43][D][matrix_keypad:086]: key '5' pressed
[14:46:43][D][main:613]: input progress: '555555', started by '~'
[14:46:44][D][matrix_keypad:058]: key @ row 1, col 1 released
[14:46:44][D][matrix_keypad:063]: key '5' released
[14:46:45][D][matrix_keypad:081]: key @ row 1, col 1 pressed
[14:46:45][D][matrix_keypad:086]: key '5' pressed
[14:46:45][D][main:613]: input progress: '555555', started by '~'
[14:46:46][D][matrix_keypad:058]: key @ row 1, col 1 released
[14:46:46][D][matrix_keypad:063]: key '5' released
[14:47:09][D][matrix_keypad:081]: key @ row 1, col 1 pressed
[14:47:09][D][matrix_keypad:086]: key '5' pressed
[14:47:09][D][main:613]: input progress: '555555', started by '~'
[14:47:09][D][matrix_keypad:058]: key @ row 1, col 1 released
[14:47:09][D][matrix_keypad:063]: key '5' released
```

ssieb, 2024-01-19:
But it's the keypad itself that's indicating the key is pressed.  The key collector isn't making it up.

LuboSta, 2024-06-17:
i have same issue with nunber 1 when pressed it reports two or somethis three times nunber 1