esphome/issues #5408: Tuya BK7231N Thermostat BECA BAC-002 - heat/cool mode does not work - fan_only abset

shakin89, 2024-01-19:
### The problem

I have flashed with tuyacloudcutter a bunch of Beca BAC-002 thermostat for which i created a tuya cloudcutter profile. 
After configuring climate section following official instructions, i cannot get heat / cool mode function properly and fan_only mode is missing too.
As you can see in the logs, when i change operation mode from the thermostats, datapoint 102, esphome does not send back the correct value to HA interface, and when i click on the HVAC modes of HA web interface, no changes is detected on the thermostats.
The only one functioning is value off
Am i missing some configuration or is it a malfunctioning/limitation of the new platform libretiny?
Thanks in advance for any help.

p.s. here is an image that explains all, when in cool mode, the thermostat is heating... 😄 
[image]

### Which version of ESPHome has the issue?

2023.12.7

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

BK72XX

### Board

BECA BAC-002-ALW

### Component causing the issue

climate

### Example YAML snippet

```yaml
substitutions:
  nome: termostato-ragazze
  nome_esteso: "Termostato Ragazze - BAC-002-ALW"
  uptime_interval: 60s
  wifi_interval: 60s
  update_interval: 60s
  baud_rate_tuyamcu: "9600"

esphome:
  name: "${nome}"

bk72xx:
  board: generic-bk7231n-qfn32-tuya

web_server:

captive_portal:

mdns:

wifi:
  ssid: !secret wifiSSID
  password: !secret wifiPW
  power_save_mode: none
  
  manual_ip:  
    static_ip: !secret termostato_ragazze_ip
    gateway: !secret gateway
    subnet: !secret subnet_mask
  domain: !secret domain
  reboot_timeout: 5min
   
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${nome} Hotspot"
[... 271 lines cut ...]
[19:18:21][D][climate:419]:   Current Temperature: 21.00°C
[19:18:21][D][climate:425]:   Target Temperature: 24.00°C
[19:18:33][D][tuya:328]: Datapoint 102 update to 2
[19:18:33][D][select:015]: 'Operation Mode': Sending state fan_only (index 2)
[19:18:33][D][climate:396]: 'Termostato Ragazze - BAC-002-ALW' - Sending state:
[19:18:33][D][climate:399]:   Mode: HEAT_COOL
[19:18:33][D][climate:401]:   Action: IDLE
[19:18:33][D][climate:410]:   Preset: NONE
[19:18:33][D][climate:419]:   Current Temperature: 21.00°C
[19:18:33][D][climate:425]:   Target Temperature: 24.00°C
[19:18:34][D][tuya:316]: Datapoint 3 update to 42
[19:18:34][D][sensor:093]: 'termostato-ragazze temperatura attuale': Sending state 21.00000  with 1 decimals of accuracy
[19:18:34][D][climate:396]: 'Termostato Ragazze - BAC-002-ALW' - Sending state:
[19:18:34][D][climate:399]:   Mode: HEAT_COOL
[19:18:34][D][climate:401]:   Action: IDLE
[19:18:34][D][climate:410]:   Preset: NONE
[19:18:34][D][climate:419]:   Current Temperature: 21.00°C
[19:18:34][D][climate:425]:   Target Temperature: 24.00°C
[19:18:34][W][component:214]: Component tuya took a long time for an operation (0.05 s).
[19:18:34][W][component:215]: Components should block for at most 20-30ms.
[19:18:35][D][sensor:093]: 'Uptime': Sending state 357.05600 s with 0 decimals of accuracy
[19:23:57][D][tuya:597]: Setting datapoint 1 to 1
[19:23:58][D][climate:011]: 'Termostato Ragazze - BAC-002-ALW' - Setting
[19:23:58][D][climate:015]:   Mode: OFF
```

shakin89, 2024-01-19:
Well, i think i found the problem. These features are not implmented yet, and are on the way.
https://github.com/esphome/esphome/pull/5159

moriahmorgan, 2024-01-19:
Thanks for the reminder to work on this @shakin89, I hope to see this in the feb release. If you need help using this as a locally implemented code, I can help you set it up until it is in the main branch

shakin89, 2024-01-19:
Thank you for your great work!

> Thanks for the reminder to work on this @shakin89, I hope to see this in the feb release. If you need help using this as a locally implemented code, I can help you set it up until it is in the main branch

Wow, that would be awesome, so i can test it meanwhile. Thanks again

moriahmorgan, 2024-01-19:
You can fetch the locally available files from [this drive link:](https://drive.google.com/file/d/13WoYl2-o087fPspiFGO0bRESNKoqKW4O/view?usp=drive_link)

You will want to put them in the following location (under the "custom_components" folder. If this folder does not exist, create it)
[image]

additionally, any existing tuya components won't work with this setup until it is merged, since the custom component I am providing to you only allows usage of what's included in the zip, so if you're doing switches or any other thing that is not a climate state, it won't work. I see you're doing tuya_time, and I'm not sure if that will work either.

Lastly, this is a basic template for what you should follow for the climate device, note the `external_components` section. That is quite important, as well as calling `tuya_new` instead of `tuya`

```
external_components:
  source: 
    type: local
    path: custom_components

tuya_new:

climate:
  - platform: tuya_new
    id: "tuya_climate"
    name: None
    switch_datapoint: 1
    target_temperature_datapoint: 2
    current_temperature_datapoint: 3
    supports_heat: True
    supports_cool: True
    active_state:
      datapoint: 4
      cooling_value: 0
      heating_value: 1
      drying_value: 2
      fanonly_value: 3
    preset:
      eco:
        datapoint: 8
        temperature: 25
      sleep:
        datapoint: 101
    swing_mode:
      vertical_datapoint: 106
      horizontal_datapoint: 107
    fan_mode:
      datapoint: 5
      auto_value: 0
      low_value: 2
      medium_value: 3
      middle_value: 4
      high_value: 1
    visual:
      min_temperature: 16
      max_temperature: 32
      temperature_step:
        current_temperature: 0.1
```

moriahmorgan, 2024-01-19:
If you need the documentation that will "go live" when this PR gets approved, [you can preview it here](https://deploy-preview-3096--esphome.netlify.app/components/climate/tuya), this might help you pick out the values that you need to use for your specific scenario

shakin89, 2024-01-20:
@moriahmorgan many thanks to you. With your instructions it works like a charm. 

I'll wait PR to be merged to activate the remaining switches of the thermostat managed by tuya platform.

shakin89, 2024-01-20:
I think this issue can be closed

closed 2024-01-20 by shakin89

moriahmorgan, 2024-01-20:
@shakin89 I added the switch component to the tuya_new platform since I had a few minutes of down time. It should just be another folder you put under the tuya_new custom_component.

Let me know if this allows you to use switches. And don't forget to change the platform of your switches.

shakin89, 2024-01-21:
Great job!!! Many thanks to you! i'm going try it next days!

PsychoRS, 2024-01-22:
> @shakin89 I added the switch component to the tuya_new platform since I had a few minutes of down time. It should just be another folder you put under the tuya_new custom_component. You can find it in the same link.
> 
> Let me know if this allows you to use switches. And don't forget to change the platform of your switches.
> 
> EDIT: Added Sensor and Select Component as well

Tested right now in a Beca 005-ALW (same as @shakin89 , only cosmetic differences) with a CB3S module, flashed to ESPHome-Libretiny with cloudcutter.  Finally it works like a charm, really thanks.

shakin89, 2024-01-24:
@moriahmorgan thank you, all works fine. 
just one note, if someone else will download your file from google_drive, in __init__.py of switch component, correct 
dependencies = ["tuya"] to tuya_new.
Thanks again for your work!

shakin89, 2024-01-24:
No worry, not a big problem.
Same in tuya_switch.cpp and .h
namespace tuya -> namespace tuya_new

moriahmorgan, 2024-01-24:
I must have copied the wrong set of files because I swore I changed those.

Moriah Morgan

moriahmorgan, 2024-03-22:
@shakin89 @PsychoRS my PR got merged. it should be in the next large release of ESPHome (Maybe mid April?)

shakin89, 2024-03-22:
@moriahmorgan Thank you again for your work, i received the merge notification from github.
Awaiting for the next release.

PsychoRS, 2024-04-19:
> @shakin89 @PsychoRS my PR got merged. it should be in the next large release of ESPHome (Maybe mid April?)

Finally we got it merged!

Really thanks @moriahmorgan for your work.

PsychoRS, 2024-04-19:
Can't compile new release (2024.4.0) with the updated tuya climate component:

```
/config/esphome/termostato-salon.yaml:263:41: error: invalid conversion from 'const char*' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]
In file included from src/esphome.h:58,
                 from src/main.cpp:3:
src/esphome/components/tuya/climate/tuya_climate.h:18:36: note:   initializing argument 1 of 'void esphome::tuya::TuyaClimate::set_active_state_id(uint8_t)'
   18 |   void set_active_state_id(uint8_t state_id) { this->active_state_id_ = state_id; }
      |                            ~~~~~~~~^~~~~~~~
/config/esphome/termostato-salon.yaml:271:38: error: invalid conversion from 'const char*' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]
In file included from src/esphome.h:58,
                 from src/main.cpp:3:
src/esphome/components/tuya/climate/tuya_climate.h:27:33: note:   initializing argument 1 of 'void esphome::tuya::TuyaClimate::set_fan_speed_id(uint8_t)'
   27 |   void set_fan_speed_id(uint8_t fan_speed_id) { this->fan_speed_id_ = fan_speed_id; }
      |                         ~~~~~~~~^~~~~~~~~~~~
```