esphome/issues #5409: ESP 32 C 3 boards failing to boot after 2023.12.8 upgrade - M5stack C3 Stamp

dat12s, 2024-01-20:
### The problem

No issues installing and updating ESP Home configurations to multiple ESP 32 C3 boards up to and including ESP Home 2023.12.7
Carried out ESP Home Upgrade to 2023.12.8 and then carried out "Upgrade All" for the devices (wirelessly) and noted that the first C3 to upgrade was not responding, Stopped the remaining upgrades and carried out checks on the failed unit.

Connected to the C3 via usb for the logs and noted that it was failing to boot, carried out a further installation starting with clean build files and again the C3 failed to boot after installation.
All Ic2 modules removed from board and installation repeated with the same result - C3 not booting.
Carried out the same installation (via usb) on a brand new C3 (M5stack C3 Stamp) and this unit also fails to boot.
I have cautiously carried out upgrades to 2023.12.8 on my remaining devices which are various ES32 devices and Rpi Pico with no issues being apparent currently.

I have restored the 2023.12.7 backup for ESP Home taken when ESP Home was upgrade to 2023.12.8  with no effect, no installation to any C3 will now boot, for the sake of completeness I the re upgraded ESP Home to 2023.12.8 with no change to the results – C3 not booting.

### Which version of ESPHome has the issue?

2023.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

HA Core 2024.1.4 / Supervisor  2023.12.1 / OS 11.4 / Frontend 20240104.0

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1 (M5Stack C3 Stamp)

### Component causing the issue

C3 Failing to Boot after 2023.12.8 update (download by USB directly connected to HA PC)

### Example YAML snippet

```yaml
esphome:
  name: enviromental
  friendly_name: Enviromental

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "JjMojvHBxwl8oKnoPedAI+ht0NpAg0WNRRLNiQkZCvg="

ota:
  password: "d5554dec615889138ec9ba8d4f147f24"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.0.166
    gateway: 192.168.0.1
    subnet: 255.255.255.0  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
[... 358 lines cut ...]
Hash of data verified.
Compressed 12944 bytes to 9299...
Wrote 12944 bytes (9299 compressed) at 0x00000000 in 0.5 seconds (effective 229.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.1 seconds (effective 430.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 576.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM1 with baud rate 115200
[00:20:19]ESP-ROM:esp32c3-api1-20210207
[00:20:19]Build:Feb  7 2021
[00:20:19]rst:0xf (BROWNOUT_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[00:20:19]SPIWP:0xee
[00:20:19]mode:DIO, clock div:1
[00:20:19]load:0x3fcd6100,len:0x438
[00:20:19]load:0x403ce000,len:0x918
[00:20:19]load:0x403d0000,len:0x24e4
[00:20:19]entry 0x403ce000
```

kk3844, 2024-01-20:
Can confirm that I'm having a similar issue on 2023.12.8.  Boards flashed with the previous version work fine.  Any board that has been upgraded no longer connects to WiFi after wireless upgrade.  I flashed a new board with 12.8 and the board works as long as it is connected via USB.  If you power the board off, then power it on to connect over WiFi, it never comes up.  Bringing it back to serial shows garbage on the log.

kk3844, 2024-01-20:
Adding further information -- After downgrading HA/ESPHome back to 2023.12.7, the previously online unit came back online almost as if it wasn't updated.  The newly installed unit still shows offline.  These are ESP-WROOM-32 units.

clydebarrow, 2024-01-24:
The yaml given in in the first comment here builds and runs fine on a C3 for me (I did remove the static IP address, otherwise changed nothing.) The sensors fail of course since there is no hardware for them.

I tested this against the current dev branch and 2023.12.9 from the command line - same results.

vecijazz, 2024-01-24:
Same problem here. I installed an older version (pip install esphome==2023.11.3) and it's work.

clydebarrow, 2024-01-24:
Are you using Bluetooth?

vecijazz, 2024-01-24:
Not for the moment.

dat12s, 2024-01-25:
No

haldi4803, 2024-03-03:
@dat12s 
try this! Worked for me.

```
esphome:
  name: $device_name
  friendly_name: $friendly_name
  platformio_options:
    board_build.flash_mode: dio
    board_build.f_cpu: 80000000L

esp32:
  variant: ESP32C3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
```

aries1998, 2024-07-14:
> @dat12s try this! Worked for me.
> 
> ```
> esphome:
>   name: $device_name
>   friendly_name: $friendly_name
>   platformio_options:
>     board_build.flash_mode: dio
>     board_build.f_cpu: 80000000L
> 
> esp32:
>   variant: ESP32C3
>   board: esp32-c3-devkitm-1
>   framework:
>     type: esp-idf
> ```
Tried this , not work.

> Same problem here. I installed an older version (pip install esphome==2023.11.3) and it's work.

Confirm downgrade esphome version do work.