esphome/issues #5411: The transport encryption key is invalid. Please ensure it matches what you have in your configuration

sgtwilko, 2024-01-20:
### The problem

I've been using ESPHome for about 18 months.

Every couple of months I'd get a notification in Home Assistant that something needed to be reconfigured.
Normally I'd just click through the configure button and whichever device had a problem would come back online.

This last week I've been unable to get a device back online, as it's been requesting the API encryption key.
I've tried multiple times, and multiple methods, to copy the key from ESPHome into Home Assistant, but I just keep getting the error:

```
The transport encryption key is invalid. Please ensure it matches what you have in your configuration```

I've finally had a chance to delve into the logs and I've found that it's trying to talk to the wrong board!

I'm trying to get `shower-power` back online and it's connecting to `house-power`

I presume that DHCP leases were renewed at some point and the ESPHome Home Assistant plugin is caching the old IP address.

The Home Assistant instance has been upgraded and rebooted during the period I've been having the issue (I'd hoped it would fix it!)

### Which version of ESPHome has the issue?

2023.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: 18 January 2024 at 01:46:56 (6907 occurrences)
Last logged: 10:36:48

shower-power @ 192.168.1.16: Connection error occurred: shower-power @ 192.168.1.16: Connection lost
house-power @ 192.168.1.16: Connection error occurred: Ping response not received after 90.0 seconds
shower-power @ 192.168.1.16: Connection error occurred: shower-power @ 192.168.1.16: Invalid encryption key: received_name=house-power
house-power @ 192.168.1.16: Connection error occurred: house-power @ 192.168.1.16: Connection lost
192.168.1.16: Connection error occurred: 192.168.1.16: Invalid encryption key: received_name=house-power
```

ssieb, 2024-01-20:
Are both of those devices online?

sgtwilko, 2024-01-20:
Yes.
They both show as online in the ESPHome dashboard in Home assistant.

They've both been upgraded too since this started, so will have had a reboot.

I'll go and give them both a power cycle.

sgtwilko, 2024-01-20:
Yeah, that didn't help.

I have an LED which flashes when there's an issue.
It stops flashing when I view the logs, or when I updated the software.

However as soon as I leave the ESPHome dashboard the light starts flashing, and it still won't connect from home assistant.

I've just tried upgrading the the operating system to force the IP address to be looked up again, but it's still not working:

```
Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: 17:48:53 (10 occurrences)
Last logged: 17:52:59

shower-power @ 192.168.1.16: Connection error occurred: shower-power @ 192.168.1.16: Invalid encryption key: received_name=house-power
192.168.1.16: Connection error occurred: 192.168.1.16: Connection requires encryption
192.168.1.16: Connection error occurred: 192.168.1.16: Invalid encryption key: received_name=house-power
```

ssieb, 2024-01-21:
Did you add them by name or IP address?  Remove the `shower-power` device from the HA integrations list and add it again by name.

amartinezflor, 2024-02-10:
Hi!
Same problem here. I can access form esphome logs but reporting key error. 
In my case, is a esphome with just two temperature dallas sensor.
I have other esp boards, also with few dallas sensor and did not report the issue.

Board "esp8265sradiantep1" not working
Board "esp8266caldera" working

I load config again on "esp8265sradiantep1"

Server:
Core
2024.1.5
Supervisor
2024.01.1
Operating System
11.4
Frontend
20240104.0

Logs:
esp8265sradiantep1 @ 192.168.100.35: Connection error occurred: esp8265sradiantep1 @ 192.168.100.35: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (Home Assistant 2024.1.5).
1:03:02 PM – (WARNING) runner.py - message first occurred at February 9, 2024 at 4:55:33 PM and shows up 883 times
Can't connect to ESPHome API for esp32ethernet @ 192.168.100.43: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.100.43', port=6053))]: [Errno 113] Connect call failed ('192.168.100.43', 6053) (SocketAPIError)
12:57:21 PM – (WARNING) runner.py - message first occurred at February 9, 2024 at 8:25:13 PM and shows up 3 times
esp8266caldera @ 192.168.100.39: disconnect request failed: Timeout waiting for DisconnectResponse after 10.0s
4:02:12 AM – (ERROR) components/esphome/manager.py
192.168.100.70: Pong not received, device is likely unresponsive; disconnecting
4:01:38 AM – (ERROR) runner.py - message first occurred at February 9, 2024 at 7:47:39 PM and shows up 3 times

Problem started arround 4 AM:
[image]

HOW I RECOVERED:
I tried with key of the other board, surprisingly, system connected. Anyway, temperature was not displayed and from board los I notice authentication error from server.
Then, I modify the key on the yaml file, save and install.
After that, esphome requested again the key and new key was admited and since then temperature is properly read.

Strange behaviour...