esphome/issues #5417: on_multi_click initialisation

plevrier, 2024-01-22:
### The problem

Hi,

When using on_multi_click the event is not firing the first time after boot.

Example code:
```
  - platform: template
    name: "button 1 cube"
    id: button1_cube
    internal: True
    on_multi_click:
      - timing:
          - on for at most 350ms
        then:
          - binary_sensor.template.publish:
              id: button1_press
              state: ON
          - delay: 500ms
          - binary_sensor.template.publish:
              id: button1_press
              state: OFF
      - timing:
          - on for at least 500ms
        then:
          - binary_sensor.template.publish:
              id: button1_presslong
              state: ON
          - delay: 500ms
          - binary_sensor.template.publish:
              id: button1_presslong
              state: OFF

```
This issue is already reported in issue https://github.com/esphome/issues/issues/2825

It would be nice that is solved. In the meanwhile I fire one event in an interval event:

```
interval:
  - interval: 10sec
    then:
      - if: 
          condition:
            lambda: 'return id(initialised) == false;'
          then:
            - binary_sensor.template.publish:
                id: button1_cube
                state: OFF
            - lambda: 'id(initialised) = true;'

```

### Which version of ESPHome has the issue?

2023.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.5

### What platform are you using?

ESP8266

### Board

Sonoff M5 ESP32

### Component causing the issue

binary sensor

ssieb, 2024-01-22:
Try adding `publish_initial_state: true`.  If that doesn't work, then show the logs with level verbose.

plevrier, 2024-01-23:
Thank you for that tip. This solves my issue!

closed 2024-01-23 by ssieb