esphome/issues #5420: xpt2046 issue

TotalPanther317, 2024-01-23:
### The problem

I am trying to use a xpt2046 touchscreen, it gives this error: Couldn't find any component that can be used for 'display::Display'. Are you missing a hub declaration?

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

touchscreen

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: GPIO15
  interrupt_pin: 16
  update_interval: 50ms
  threshold: 400
  calibration_x_min: 3860
  calibration_x_max: 280
  calibration_y_min: 340
  calibration_y_max: 3860
```

TotalPanther317, 2024-01-23:
Well, I added this. It will result is absolutely nothing.
It looks like this:
`spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

external_components:
  - source: "github://pr#5997"
    refresh: 1s
    components: [ display, touchscreen, xpt2046 ]

touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: GPIO15
  interrupt_pin: 16
  update_interval: 50ms
  threshold: 400
  calibration_x_min: 3860
  calibration_x_max: 280
  calibration_y_min: 340
  calibration_y_max: 3860`

nielsnl68, 2024-01-23:
Can you be more clear, what do you mean with "It will result is absolutely nothing."?

TotalPanther317, 2024-01-23:
Of course, i replaced the n with an s. I meant that just nothing changes.

nielsnl68, 2024-01-23:
Could you show the compile log and your full Yaml file? 
And maybe some screenshot of the display
So i can  beter understand your issue. 
At this moment it is hard to tell what is going one.

nielsnl68, 2024-01-23:
Thanks.

Now the error makes more sense to me.
Sadly, in esphome you can use the touchscreen without a display atm. That is why you get that error error.

For now, what you can do is setup a simple graphics `display` component where you can manual set the dimensions of your display. And everything should work, i think.

And please make a feature-requests with the idea that you want to use the touchscreen without a display.

let me know of this is working for you.

TotalPanther317, 2024-01-23:
So far it shows no error, sadly my unstable AP running on the homeassistant pi gave up. I will figure this out and come back when I can use the ap again.

nielsnl68, 2024-01-23:
okay suc6

TotalPanther317, 2024-01-25:
My esp wont connect anymore so I will close this, at least for now.

TotalPanther317, 2024-01-28:
I back, things are now working again and I installed this code:
```
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

display:
  - platform: ili9xxx
    model: ili9341
    dc_pin: GPIO15
    cs_pin: GPIO5
    reset_pin: GPIO16
    lambda: |-
      it.fill(COLOR_BLACK);
```
on my esp and connected it properly. However nothing happens. What is the problem here?

nielsnl68, 2024-01-28:
Without saying it negative: What did you expected to happen? 

I need more information then this to help you out sorry.

TotalPanther317, 2024-01-28:
I wanted it to do the thing it says in the lambda, fill the entire thing black.

nielsnl68, 2024-01-28:
Can show me the whole YAML and the install log?

TotalPanther317, 2024-01-28:
Of course

```
esphome:
  name: display
  friendly_name: display

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "..."

ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid2
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "..."
    password: "..."

captive_portal:

spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

display:
  - platform: ili9xxx
    model: ili9341
    dc_pin: GPIO15
    cs_pin: GPIO5
    reset_pin: GPIO16
    lambda: |-
      it.fill(COLOR_BLACK);
```

```
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/display.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing display (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- SPI @ 1.0
Compiling .pioenvs/display/src/main.cpp.o
<unicode string>: In lambda function:
<unicode string>:43:15: warning: 'esphome::COLOR_BLACK' is deprecated: Use Color::BLACK instead of COLOR_BLACK [-Wdeprecated-declarations]
In file included from src/esphome/components/display/display.h:8,
                 from src/esphome.h:17,
                 from src/main.cpp:3:
src/esphome/core/color.h:162:20: note: declared here
  162 | extern const Color COLOR_BLACK;
      |                    ^~~~~~~~~~~
Linking .pioenvs/display/firmware.elf
RAM:   [====      ]  41.8% (used 34256 bytes from 81920 bytes)
Flash: [=====     ]  48.6% (used 497721 bytes from 1023984 bytes)
Building .pioenvs/display/firmware.bin
esp8266_copy_factory_bin([".pioenvs/display/firmware.bin"], [".pioenvs/display/firmware.elf"])
========================= [SUCCESS] Took 20.30 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of display.local
INFO  -> 10.3.141.53
INFO Uploading /data/build/display/.pioenvs/display/firmware.bin (501872 bytes)
INFO Compressed to 352796 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 11.61 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from display.local using esphome API

```

I dont think the deprecated color thing affects this.

nielsnl68, 2024-01-28:
Now i need the startup log as well :D

TotalPanther317, 2024-01-28:
Here you go, it does some of this

`rlï¿½lÃ¯Â¿Â½r$Ã¯Â¿Â½nÃ¯Â¿Â½lÃ¯Â¿Â½b|Ã¯Â¿Â½Ã¯Â¿Â½Ã¯Â¿Â½rbÃ¯Â¿Â½bÃ¯Â¿Â½nnlnnbbpÃ¯Â¿Â½$blrlpÃ¯Â¿Â½nÃ¯Â¿Â½Ã¯Â¿Â½lÃ¯Â¿Â½bnÃ¯Â¿Â½nÃ¯Â¿Â½Ã¯Â¿Â½bÃ¯Â¿Â½Ã¯Â¿Â½nn'lÃ¯Â¿Â½l`Ã¯Â¿Â½nn$`nrÃ¯Â¿Â½Ã¯Â¿Â½Ã¯Â¿Â½nrrÃ¯Â¿Â½`pÃ¯Â¿Â½nÃ¯Â¿Â½rÃ¯Â¿Â½bbnÃ¯Â¿Â½nbÃ¯Â¿Â½Ã¯Â¿Â½nn'l`Ã¯Â¿Â½nn$`nrÃ¯Â¿Â½Ã¯Â¿Â½Ã¯Â¿Â½nrl`rÃ¯Â¿Â½Ã¯Â¿Â½nrl`Ã¯Â¿Â½Ã¯Â¿Â½$Ã¯Â¿Â½l`Ã¯Â¿Â½Ã¯Â¿Â½nÃ¯Â¿Â½`[I][logger:351]: Log initialized`

while starting up, when it finally did, it give this output:

```
[I][logger:351]: Log initialized
[C][ota:473]: There have been 10 suspected unsuccessful boot attempts.
[E][ota:480]: Boot loop detected. Proceeding to safe mode.
[I][app:029]: Running through setup()...
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   LocalÃ‚Â MAC: 34:94:54:81:6E:33
[D][wifi:455]: Starting scan...
[D][wifi:470]: Found networks:
[I][wifi:513]: - 'pi2' (B8:27:EB:80:94:1C) [redacted]Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[D][wifi:515]:     Channel: 1
[D][wifi:516]:     RSSI: -71 dB
[D][wifi:518]: - '...'[redacted] (52:E6:36:E3:C3:1A) [redacted]Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[D][wifi:518]: - '...'[redacted] (50:E6:36:E3:C3:1A) [redacted]Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[D][wifi:518]: - '...'[redacted] (0E:72:74:F4:3F:4E) [redacted]Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[D][wifi:518]: - '...'[redacted] (0C:72:74:F4:3F:4E) [redacted]Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[D][wifi:518]: - 'pi'[redacted] (DC:A6:32:86:0E:58) [redacted]Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[I][wifi:300]: WiFi Connecting to 'pi2'...
[I][wifi:587]: WiFi Connected!
[C][wifi:405]:   Local MAC: 34:94:54:81:6E:33
[C][wifi:410]:   SSID: 'pi2'[redacted]
[C][wifi:411]:   IP Address: 10.3.141.53
[C][wifi:412]:   BSSID: B8:27:EB:80:94:1C[redacted]
[C][wifi:414]:   Hostname: 'display'
[C][wifi:416]:   Signal strength: -77 dB Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[C][wifi:420]:   Channel: 1
[C][wifi:421]:   Subnet: 255.255.255.0
[C][wifi:422]:   Gateway: 10.3.141.1
[C][wifi:423]:   DNS1: 10.3.141.1
[C][wifi:424]:   DNS2: 0.0.0.0
[D][wifi:596]: Disabling AP...
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: display.local:8266
[C][ota:101]:   Using Password.
[W][ota:106]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[I][app:062]: setup() finished successfully!
[I][ota:493]: Waiting for OTA attempt.
[I][app:102]: ESPHome version 2023.12.9 compiled on Jan 28 2024, 12:32:01
[C][wifi:573]: WiFi:
[C][wifi:405]:   Local MAC: 34:94:54:81:6E:33
[C][wifi:410]:   SSID: 'pi2'[redacted]
[C][wifi:411]:   IP Address: 10.3.141.53
[C][wifi:412]:   BSSID: B8:27:EB:80:94:1C[redacted]
[C][wifi:414]:   Hostname: 'display'
[C][wifi:416]:   Signal strength: -77 dB Ã¢Â–Â‚Ã¢Â–Â„Ã¢Â–Â†Ã¢Â–Âˆ
[C][wifi:420]:   Channel: 1
[C][wifi:421]:   Subnet: 255.255.255.0
[C][wifi:422]:   Gateway: 10.3.141.1
[C][wifi:423]:   DNS1: 10.3.141.1
[C][wifi:424]:   DNS2: 0.0.0.0
[C][logger:439]: Logger:
[C][logger:440]:   Level: DEBUG
[C][logger:441]:   Log Baud Rate: 115200
[C][logger:443]:   Hardware UART: UART0
[C][captive_portal:088]: Captive Portal:
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: display
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: display.local:8266
[C][ota:101]:   Using Password.
[W][ota:106]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
```

nielsnl68, 2024-01-28:
Strange. i do not see the init of the display.

TotalPanther317, 2024-01-28:
I didn't know I have to use something like this, also I can't find it in the docs. What would I need to put in the code to init and use it?

nielsnl68, 2024-01-28:
Notting. That should be done automatically. 
I'm just not sure what is going on. Can you hook this device to a serial port and see of you see anything that is missing from the log you shown above?

TotalPanther317, 2024-01-28:
I already used a serial port

nielsnl68, 2024-01-28:
could you set logging to verbose and see what happens?. '

Btw what kind of device are you using now?

TotalPanther317, 2024-01-28:
I am using an esp8266 on a d1 mini.

this is the log:

```
fpm close 3 
pm open,type:0 0
rlï¿½lï¿½r$ï¿½nï¿½lï¿½b|ï¿½ï¿½ï¿½rbï¿½bï¿½nnlnn2bbpï¿½$blrlpï¿½nï¿½lï¿½bnï¿½nï¿½ï¿½bï¿½ï¿½nn'lï¿½l`ï¿½nn$`nrï¿½ï¿½ï¿½nrrï¿½`pï¿½nï¿½rï¿½bbnï¿½nbï¿½ï¿½nn'l`ï¿½nn$`nrï¿½ï¿½ï¿½nrl`rï¿½ï¿½nrl`ï¿½ï¿½l`ï¿½ï¿½nï¿½`
SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[I][logger:351]: Log initialized
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4021d3f1, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40266fbc, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021d3f1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40266fbc depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffd0 offset: 0190
3ffffeb0:  40100c10 00000000 000000ac 401012c4  
3ffffec0:  3fff164c 4021db27 000000a0 4022becb  
3ffffed0:  00000000 00000000 3ffefa64 4021db2d  
3ffffee0:  00000000 feefeffe feefeffe feefeffe  
3ffffef0:  9717f480 8ff121de a200c52f cb72634b  
3fffff00:  acd7de8a d52b06f8 d707f5cf bebc7523  
3fffff10:  3fffff18 0000000c 717a7167 4b51766b  
3fffff20:  52384d53 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 00000000 00000000 9717f480  
3fffff50:  8ff121de a200c52f cb72634b acd7de8a  
3fffff60:  d52b06f8 d707f5cf bebc7523 9717f480  
3fffff70:  8ff121de a200c52f cb72634b acd7de8a  
3fffff80:  d52b06f8 d707f5cf bebc7523 feefeffe  
3fffff90:  00000001 3fff16c4 00000000 feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3ffefe50  
3fffffb0:  3fffdad0 00000000 3ffefe3c 4022c3a4  
3fffffc0:  feefeffe feefeffe 3ffe8610 40100459  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0007ca90
~ld

SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[I][logger:351]: Log initialized
[C][ota:473]: There have been 1 suspected unsuccessful boot attempts.
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4021d3f1, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40266fbc, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021d3f1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40266fbc depc=0x00000000

>>>stack>>>

[... 543 lines cut ...]
[C][wifi:412]:   BSSID: B8:27:EB:80:94:1C[redacted]
[C][wifi:414]:   Hostname: 'display'
[C][wifi:416]:   Signal strength: -75 dB â–‚â–„â–†â–ˆ
[V][wifi:418]:   Priority: -1.0
[C][wifi:420]:   Channel: 1
[C][wifi:421]:   Subnet: 255.255.255.0
[C][wifi:422]:   Gateway: 10.3.141.1
[C][wifi:423]:   DNS1: 10.3.141.1
[C][wifi:424]:   DNS2: 0.0.0.0
[C][logger:439]: Logger:
[C][logger:440]:   Level: VERBOSE
[C][logger:441]:   Log Baud Rate: 115200
[C][logger:443]:   Hardware UART: UART0
[C][captive_portal:088]: Captive Portal:
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: display
[V][mdns:117]:   Services:
[V][mdns:119]:   - _http, _tcp, 80
[V][mdns:121]:     TXT: version = 2023.12.9
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: display.local:8266
[C][ota:101]:   Using Password.
[W][ota:106]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts

```

nielsnl68, 2024-01-28:
maybe it us better to switch to a esp32s2 or s3 they are also in D1 mini format.
And have build-in PSRAM

TotalPanther317, 2024-01-28:
The problem is that I don't have a spare esp32 at hand, I will get back when I do.

TotalPanther317, 2024-02-08:
Nevermind, openHASP now run on this touchscreen.

closed 2024-02-08 by TotalPanther317