esphome/issues #5421: ble_text_sensor reading characteristic it shouldn't

jklap, 2024-01-23:
### The problem

I have 5 ble_text_sensors set up, each reading from the same service UUID but different characteristics (using the handle/alias).

I'm seeing the following in my logs:
```
[08:49:51][W][ble_text_sensor:083]: Error reading char at handle 16, status=2
[08:49:51][W][ble_text_sensor:083]: Error reading char at handle 16, status=2
[08:49:51][W][ble_text_sensor:083]: Error reading char at handle 16, status=2
[08:49:51][W][ble_text_sensor:083]: Error reading char at handle 16, status=2
[08:49:51][W][ble_text_sensor:083]: Error reading char at handle 16, status=2
```

Note that the handle/alias my text sensors are reading from is NOT handle 0x10 as noted above. In fact, the service UUID they are reading from does not even have that handle:
```
Service UUID: 0x180A 
start_handle: 0x24 end_handle: 0xffff 
```

There is ANOTHER service UUID that does though AND that handle is write/notify only, no reads:
```
handle 0x10, properties 0x2c (write, notify)
```

And we know that status=2 is ESP_GATT_READ_NOT_PERMIT (which would be correct for that handle 0x10)

SO... first, why is it reading from that handle when the ble_text_sensor is not setup to read from it?

Second, whatever it causing this read likely needs additional logic so read attempts on handles that don't have read shouldn't execute?

Additionally, each of these ble_text_sensors is set to an update_interval of 5min, yet these errors show up exactly 60 seconds apart.

What can I do to do additional troubleshooting? 

### Which version of ESPHome has the issue?

2023.12.8

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

D1 Mini Esp32

### Component causing the issue

ble_text_sensor

ssieb, 2024-01-23:
config?

jklap, 2024-01-23:
> config?

Unable to post it in it's entirety, which parts would you like to see specifically? I can redact those parts and share

ssieb, 2024-01-23:
At least the ble client and sensor parts.

lufton, 2024-10-20:
I have the same issue. Still trying to figure out how to solve it. @jklap did you find a solution?