esphome/issues #5429: [2023.12.9] Compilations error - local component

ESP4Ever, 2024-01-24:
### The problem

I'm using a slightly modified tm1638 component as a local but since last ESPhome update it is not possible to compile the FW

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.5

### What platform are you using?

ESP8266

### Component causing the issue

tm1638

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/konlen-wifi-th-sensor.yaml...
INFO Detected timezone 'Europe/Moscow'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 407, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/config/esphome/local_components/tm1638/display.py", line 37, in to_code
    await display.register_display(var, config)
  File "/esphome/esphome/components/display/__init__.py", line 119, in register_display
    await cg.register_component(var, config)
  File "/esphome/esphome/cpp_helpers.py", line 56, in register_component
    raise ValueError(
ValueError: Component ID gn1638_display was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.
```
```

bkaufx, 2024-01-24:
If you have external components it is up to you to make sure you incorporate any changes that ESPHome makes back into your version.  I'm guessing if you remove this line from display.py it will fix your issue.  

[image]

https://github.com/esphome/esphome/commit/dbdcb39af903963efb6ba9e2106f42ad8495ac92

ssieb, 2024-01-24:
If you have any more questions, come ask on the esphome discord server.

closed 2024-01-24 by ssieb

ESP4Ever, 2024-01-25:
Thank you, it was a cause of the issue. I got the mentioned file out of sync with Git somehow!