esphome/issues #5432: BK72XX Compile error error in IRremoteESP8266

dianlight, 2024-01-25:
### The problem

I have a compiler error when I add climete/heatpumpir to my config.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1

### What platform are you using?

BK72XX

### Board

generic-bk7231t-qfn32-tuya

### Component causing the issue

climate/heaypimpir

### Example YAML snippet

```yaml
esphome:
  name: "smart-ir-display"

bk72xx:
  board: generic-bk7231t-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  password: !secret api_password

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

remote_receiver:
  id: rcvr
  pin:
    number: PWM2 #P8
    inverted: True
    mode: INPUT_PULLUP
  dump: all
  tolerance: 55%
  buffer_size: 10kB
[... 187 lines cut ...]
Compiling .pioenvs/smart-ir-display/lib84c/IRremoteESP8266/IRac.cpp.o
Compiling .pioenvs/smart-ir-display/lib84c/IRremoteESP8266/IRrecv.cpp.o
Archiving .pioenvs/smart-ir-display/libb29/libDNSServer.a
Archiving .pioenvs/smart-ir-display/lib2be/libESPAsyncWebServer-esphome.a
Compiling .pioenvs/smart-ir-display/lib84c/IRremoteESP8266/IRsend.cpp.o
Compiling .pioenvs/smart-ir-display/lib84c/IRremoteESP8266/IRtext.cpp.o
/piolibs/IRremoteESP8266/src/IRrecv.cpp:140:7: error: '_IRrecv' has not been declared
  140 | using _IRrecv::timer;
      |       ^~~~~~~
/piolibs/IRremoteESP8266/src/IRrecv.cpp:174:3: error: expected unqualified-id before 'if'
  174 |   if (params.rawlen) params.rcvstate = kStopState;
      |   ^~
/piolibs/IRremoteESP8266/src/IRrecv.cpp:181:1: error: expected declaration before '}' token
  181 | }
      | ^
/piolibs/IRremoteESP8266/src/IRrecv.cpp:184:27: error: expected initializer before 'gpio_intr'
  184 | static void USE_IRAM_ATTR gpio_intr() {
      |                           ^~~~~~~~~
/piolibs/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
/piolibs/IRremoteESP8266/src/IRrecv.cpp:389:35: error: 'gpio_intr' was not declared in this scope
  389 |   attachInterrupt(params.recvpin, gpio_intr, CHANGE);
      |                                   ^~~~~~~~~
*** [.pioenvs/smart-ir-display/lib84c/IRremoteESP8266/IRrecv.cpp.o] Error 1
========================= [FAILED] Took 217.86 seconds =========================
```

crankyoldgit, 2024-02-02:
FYI, I don't believe the bk7231t etc is part of the ESP chip family.
The IRremoteESP8266 library only supports the ESP8266 & ESP32 family of chips.

VivitechMT, 2024-02-07:
I've encountered the same issue when using 

climate:
  platform: heatpumpir

I understand the comments about the BK72xx not being ESP chips, but the curious thing is that I have a working BK72xx device using this config:

climate:
  platform: mitsubishi

So I wonder how does the mitsubishi configuration successfully send IR commands to the Beken chip ?!

crankyoldgit, 2024-02-07:
> So I wonder how does the mitsubishi configuration successfully send IR commands to the Beken chip ?!

That is something you will have to ask the ESPHome community. I have no idea.

If `ESP32` or `ESP8266` are not defined, you will get the error you orginally described.

rob-deutsch, 2024-06-25:
@crankyoldgit thanks for chiming in here.

I'm the codeowner for this part of `esphome`, and I agree with your perspective.

We'll resolve this internally and won't need to involve your `crankyoldgit/IRremoteESP8266` repo.

PS thanks for creating and maintaining `IRremoteESP8266`! I've used it a few times and its very helpful.

rob-deutsch, 2024-06-25:
> I've encountered the same issue when using
>
> climate: platform: heatpumpir
>
> I understand the comments about the BK72xx not being ESP chips, but the curious thing is that I have a working BK72xx [...]

That's a good quesiton, and its answered in the ESPHome documentation:

> The heatpumpir platform supports dozens of manufacturers and hundreds of AC units by utilising the [Arduino-HeatpumpIR library](https://github.com/ToniA/arduino-heatpumpir).
> 
> This platform should only be used if your AC unit is not supported by any of the other (native) platforms. No support can be provided for Arduino-HeatpumpIR, because it is a third party library.

The heatpumpir platform doesn't currently work on BK72xx chips. Fixes for this are being discussed in #6955 .

closed 2024-07-21 by jesserockz