esphome/issues #5442: Component ili9xxx display fails to allocate buffer with ili9341 and stops access to external sensors

TractorHound, 2024-01-28:
### The problem

Updated to home assistant after not revising any code since last March. Wished to install working code on another display, but it doesn't work anymore.
Using the previous platform = ili9341 and model TFT 2.4 everything worked ok. 
With the foced change to ili9xxx the display no longer allocates a buffer, the class passed to the lambda has changed from DisplayBuffer to Display and extrnal sensors can no longer be read.
Other minor issues are that the display rotation has changed (was 270, now 0) and the need to specify height and width as the defaults were different (width and height reversed.)

### Which version of ESPHome has the issue?

2023.19.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.1

### What platform are you using?

ESP32

### Board

ESP32-2432S028  this is a ESP32-WROOM-32 with 2.8" 240x320 resistive touch display

### Component causing the issue

ILI9xxx display

### Example YAML snippet

```yaml
The YAML that worked with my old installation is

display:
  - platform: ili9341
    model: TFT 2.4 #M5STACK #TFT 2.4 or TFT 2.4R
    cs_pin: 15 # GPIO5    #D8
    dc_pin: 2 # GPIO21 #D2
    reset_pin: 12 # GPIO22 #D1
    rotation: 270

    update_interval: 8s

    lambda: |-
      // backlight on GPIO21
      commonLCD(${device_name}_wifi_signal_sensor, &it, ${bar_height}, 21);

The version that will now compile and displays text but blocks external sensor reads is 

display:
  - platform: ili9xxx # Using an ILI9341 chip
    model: ILI9341
    cs_pin: 15 # GPIO5    #D8
    dc_pin: 2 # GPIO21 #D2
    reset_pin: 12 # GPIO22 #D1
    rotation: 0
    dimensions: 
      height: 240
      width: 320
    data_rate: 80MHz
    update_interval: 8s
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.9
[... 347 lines cut ...]
[10:11:47][C][psram:021]:   Available: NO
[10:11:47][C][mdns:115]: mDNS:
[10:11:47][C][mdns:116]:   Hostname: esp32-2x8lcd-usbc
[10:11:47][C][ota:097]: Over-The-Air Updates:
[10:11:47][C][ota:098]:   Address: 192.168.0.48:3232
[10:11:47][C][ota:101]:   Using Password.
[10:11:47][C][api:139]: API Server:
[10:11:47][C][api:140]:   Address: 192.168.0.48:6053
[10:11:47][C][api:142]:   Using noise encryption: YES
[10:11:47][C][debug:064]: Debug component:
[10:11:47][D][debug:076]: ESPHome version 2023.12.9
[10:11:47][D][debug:080]: Free Heap Size: 160300 bytes
[10:11:47][D][debug:110]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[10:11:47][D][debug:159]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=3
[10:11:47][D][debug:167]: ESP-IDF Version: v4.4.2
[10:11:47][D][debug:172]: EFuse MAC: FC:B4:67:F6:4E:84
[10:11:47][D][debug:272]: Reset Reason: Power On Reset
[10:11:47][D][debug:317]: Wakeup Reason: Unknown
[10:12:38][D][api:102]: Accepted 192.168.0.31
[10:12:38][W][api.connection:092]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
```

### Additional information

I've tried different data rates for the SPI and it works with same with 40 and 80MHz, just slightly quicker to update the display.

nielsnl68, 2024-01-28:
The data rate of 80MHz is in most experiences to much. Just try 40MHz or a bit slower.

TractorHound, 2024-01-28:
The display with this board works fine at 80MHz (possibly because everything is on one board or the later ILI9341 chips are faster), but I assume that either way it shold not lead to the "Could not allocate buffer for display!" message.

I just found a solution to the failure to access external sensors - delete the device in ESPHome and reinstall.

The main issue therefore is the unexplained lack of buffer.

nielsnl68, 2024-01-28:
This message happens because it could not create a 16bit memory space. And it falls back to 8 bit 332 mode.
As i said i expect the date rate to fail.

TractorHound, 2024-01-28:
Thanks for the quick responses. I'll keep one display running at 80MHz and one at a lower speed to see if there are any differences.

Does the "Could not allocate buffer for display!" message therefore mean that a 16 bit buffer couldn't be allocated but that an 8 bit one had been, or that a 16 bit buffer couldn't be allocated and there is no buffering? Possibly a slightly different message could reduce confusion?

Not that I need 16 bit memory space, with 320kB of RAM but I would have expected enough space as the compile  shows only used 41752 bytes from 327680 bytes. 

The documentation describes the colour palette as NONE, GRAYSCALE or IMAGE_ADAPTIVE but implies 8 bit only for each of these, so where does the 16 bit buffer come in and can it be disabled (assuming there was enough memory in the first case) should a use wish to reserve more space.  

I use a lambda like `commonLCD(${device_name}_wifi_signal_sensor, &it, ${bar_height}, 21);` and have noticed that while the previous code had the clas for 'it' as DisplayBuffer, the later code has changed to 'Display'.

nielsnl68, 2024-01-28:
> Does the "Could not allocate buffer for display!" message therefore mean that a 16 bit buffer couldn't be allocated but that an 8 bit one had been, or that a 16 bit buffer couldn't be allocated and there is no buffering?
It means indeed that 16 bit buffer did not fit but the 8 bit was allocated. Otherwise you would have seen two times that message.

I'm working on a new display_driver platform component that will tackle this issue. One new future will be that it direct sending content to the display.
  
> Not that I need 16 bit memory space, with 320kB of RAM but I would have expected enough space as the compile shows only used 41752 bytes from 327680 bytes.
The memory space shown is a total space available. But there can be parts used within the free space. I did not found a way to used memory blocks together.

> The documentation describes the colour palette as NONE, GRAYSCALE or IMAGE_ADAPTIVE but implies 8 bit only for each of these, so where does the 16 bit buffer come in and can it be disabled (assuming there was enough memory in the first case) should a use wish to reserve more space.

this will be converted in to 16 bit mode while sending the display content to the display.

> I use a lambda like commonLCD(${device_name}_wifi_signal_sensor, &it, ${bar_height}, 21); and have noticed that while the previous code had the clas for 'it' as DisplayBuffer, the later code has changed to 'Display'.

Yes, that is one of the changes that we made in the last couple of months,

TractorHound, 2024-01-28:
Thanks again for the extra information.

`It means indeed that 16 bit buffer did not fit but the 8 bit was allocated. Otherwise you would have seen two times that message.`

Just a suggestion - possibly change the messages to something like "Couldn't allocate a 16 bit buffer, using 8 bit instead" and "Couldn'tallocate any buffer" if there isn't even space for the smaller buffer.

`I'm working on a new display_driver platform component that will tackle this issue. One new future will be that it direct sending content to the display.`

This would be great if you can achieve that. I have a similar display to the one I'm using here. The other display uses a similar board with a ST7789 driving a 3.5" 480x320 display but no PSRAM so currently isn't usable. These are low cost boards via ALiExpress with capacitive touch so are otherwise very attractive.

`The memory space shown is a total space available. But there can be parts used within the free space. I did not found a way to used memory blocks together.`

Free memory space fragmentation can be a pain! One possible solution could be an array of pointers to row (or column) length buffers, although the additional overhead in time and code space may not be worth the effort.

Thanks for confirmaing the class change from DisplayBuffer to Display is permanent - I'll revise my code.

I think you are doing a great job here and your efforts are much appreciadted by those of us who benefit.

TractorHound, 2024-01-29:
Having slept on the problem overnight, I think there may be a much simpler solution to the memory allocation issue.

If I understand correctly, you are currently allocating the buffer on the heap at run time, but with a display the allocation needs to remain for the whole time the application is running. Therefore, statically allocate the buffer and the linker will ensure that it is in a single contoguous unmoving block of memory... and before the heap is even available. 

If there is not enough space then the compile will fail (i.e. problem trapped at build time) or there may not be enough subsequent free RAM for stack and heap (fail to run.) The compile process however will report the amount of free RAM which should give a clue to the potential problem. The use of PSRAM for the buffer, while helpful, may be something that the user may wish to have some control over, e.g. if they wish to reserve all the PSRAM for something else in their application.

I was thinking of additional options in YAML:

- Buffer size: None (if possible), 8 bit, 16 bit
- PSRAM: Use, never use, auto (this last option may not be practical)

If PSRAM is specified to use but not fitted then this would be an obvious run-time error.

The final aspect is how to debug the situation where the buffer is allocated but the application crashes with one suspecting lack of enough free RAM. Through YAML it is currently possible to specify the display width and height. If one had say a 320 x 480 physical display but specified 320 x 240, would the display still work but addressing only some of the pixels while using a smaller buffer? If so, one could shrink the active display area to free RAM and confirm if that is the problem or not.

nielsnl68, 2024-01-29:
That is not a bad idea. You sound knowledgeable, are you willing to create a PR for this? 

> The final aspect is how to debug the situation where the buffer is allocated but the application crashes with one suspecting lack of enough free RAM. Through YAML it is currently possible to specify [...]

Talking only about the ILI9XXX component here. Then yes because we set the address window to the display first before we send the content to the display.

Lets talk more at [Discord](https://discord.com/channels/429907082951524364/1119046373543780363).
We did have a discussion about this before but we never trough about this solution.

jeroenterheerdt, 2024-05-19:
is there any news on this? I am struggling with the same error and screen. The same hardware setup works fine with "regular" Arduino (using Adafruit libraries) but since I need to integrate with HA I would prefer to use esphome instead of going the mqtt route...

CodeMartn, 2024-06-07:
I am having the same issue. Working with Arduino library, but buffer error with ESPhome compilation. Is there any solution or workaround?
Thx

jeroenterheerdt, 2024-06-07:
workaround that works for me is to get a E32S and instruct ESPHOME to use the memory on that board:

```
esphome:
  name: kegeratordisplay
  friendly_name: kegeratordisplay
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
  flash_size: 16MB

```

CodeMartn, 2024-06-10:
> workaround that works for me is to get a E32S and instruct ESPHOME to use the memory on that board:
>   platformio_options:
>     build_flags: "-DBOARD_HAS_PSRAM"
>     board_build.arduino.memory_type: qio_opi

@jeroenterheerdt: Thank you for your reply. It helped me to understand that my first generation ESP32 is the problem, as it does not have any PSRAM.

For all others having the same issue: As a workaround you can install ESPHome version 2023.2.4 (last version before component ili9341 was substituted by ili9xxx ) and use ili9341 instead of ili9xxx.

jeroenterheerdt, 2024-06-10:
How did you install the older version? I never found out how.

clydebarrow, 2024-12-16:
The best solution to this is to use LVGL with a 25% buffer_size - that eliminates the need for the display driver to allocate any buffer.

```
display:
  - platform: ili9xxx
    model: ili9341
    data_rate: 20MHz
    dc_pin: GPIOXX
    auto_clear_enabled: false
    update_interval: never

lvgl:
  buffer_size: 25%
```

closed 2024-12-16 by clydebarrow

slongano, 2025-01-12:
A strange thing happened to me. The first time I installed the software I had no problems with the buffer allocation and the screen worked correctly (I'm using an AZ-TOUCH kit with an ESP32-DEVKITC from AZ-Delivery). I compiled with "arduino" as framework. Then I tried with esp-idf, which gave me a lot of errors with WiFi. I reverted to "arduino" and from then on I got the buffer allocation error.
This is strange enough, but it got weirder. To install the esp-idf version I connected the ESP32 via USB cable.
To figure out what changed from before I tried installing the same software wirelessly (I usually install a minimal version of ESPHome on every ESP32 with USB, then the specific version wirelessly after is connects to my WiFi network, since the Home Assistant server is not where I have my regular PC).
Well, after installing exactly the same software wirelessly the error disappeared.
So, I have even less clue as before regarding the possible causes of the failed allocation.

ozarchie, 2025-02-03:
Same problem here, but with Restouch/S3-devkitc-1(ili9488)

Working OK with Arduino - not using LVGL
Changed to espidf to use LVGL - no success.
Changed back to Arduino - display buffer problem.

slongano, 2025-02-17:
Have you tried installing the same code through OTA?