esphome/issues #5443: Can't compile for M5Stamp C3

b0c1, 2024-01-28:
### The problem

I can't compile the esphome for my m5stamp-c3 .

Here is my config:
```yaml
esphome:
  name: collector-1
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

wifi:
  ssid: "xxx"
  password: "yyyyyyyyy"
  manual_ip:
    static_ip: 192.168.0.228
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  ap:
    ssid: collector-1-fallback
    password: "yyyyyyyyyy"
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

captive_portal:

# Enable logging
logger:

time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 2
        days_of_week: MON-SUN
        then:
           - switch.toggle: restart_switch

switch:
  - platform: restart
    name: "Auto Restart"
    id: restart_switch
# Enable Home Assistant API
api:
  encryption:
    key: "yyyyyyyyyy"

ota:
  password: "yyyyyyyyyy"

mqtt:
  broker: 192.168.1.1
  username: "xxxxxxxxxx"
  password: "yyyyyyyyyy"
  discovery: True
  port: 1883
  topic_prefix: temperatures

web_server:
  port: 80

esp32_ble_tracker:
  scan_parameters:
    window: 300ms
    continuous: false
```

The error is when the application tries to link the objects:

```
Archiving .pioenvs\collector-1\libe85\libnoise-c.a
Archiving .pioenvs\collector-1\lib93c\libsodium.a
Linking .pioenvs\collector-1\firmware.elf
c:/users/janos/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pioenvs/collector-1/src/esphome/components/mdns/mdns_esp_idf.o: in function `esphome::mdns::MDNSComponent::setup()':
D:\esphome\.esphome\build\collector-1/src/esphome/components/mdns/mdns_esp_idf.cpp:13: multiple definition of `esphome::mdns::MDNSComponent::setup()'; .pioenvs/collector-1/src/esphome/components/mdns/mdns_esp32.o:D:\esphome\.esphome\build\collector-1/src/esphome/components/mdns/mdns_esp32.cpp:14: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\collector-1\firmware.elf] Error 1
``` 

I not understand why found multiple MDNSComponent setup. 

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

m5stamp-c3 

### Component causing the issue

mdns

ssieb, 2024-01-28:
Try a clean build.

b0c1, 2024-01-29:
> Try a clean build.

I deleted the .esphome folder, tried with esphome clean etc., any other idea?

closed 2024-06-20 by github-actions[bot]