esphome/issues #5446: Template Switch trigger on/off action on start-up

DJTerentjev, 2024-01-29:
### The problem

During start-up template switch change the state from unknown or undefined, so it shouldn't call for action in this case.
I use template Switch to control latching relays and every time while restarting ESP32 relays changing state.
After start-up all working fine.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.1

### What platform are you using?

ESP32

### Board

wt32-eth01

### Component causing the issue

Template Switch 

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    name: "light_status" # Light status
    id: light_status
    pin:
      pca9554: esp32_input
      number: 6 # Use pin number 6
      mode: input
      # inverted: true

switch:
  - platform: gpio
    name: "out_light" # Light control
    restore_mode: DISABLED
    pin:
      pcf8574: esp32_otput
      number: 5
      inverted: true
      mode:
        output: true
    id: out_light
    internal: true

  - platform: template
    name: room_light
    lambda: |- 
      if (id(light_status).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - switch.turn_on: out_light
      - delay: 10ms
      - switch.turn_off: out_light
    turn_off_action:
      - switch.turn_on: out_light
      - delay: 10ms
      - switch.turn_off: out_light
```

### Additional information

For now I created extra template switch (default state is off ) and checking it's state inside turn on/off action:
If new template is off -> turn it on
If new template is on -> trigger latching relay

ssieb, 2024-01-29:
Disable restoring the state.
But why do you even need that switch when you could use the binary sensor directly?

closed 2024-01-29 by ssieb

ssieb, 2024-01-29:
Use the binary sensor triggers.  If you need more help, come ask on discord.