esphome/issues #5449: Is __FILE__ broken or not supported?

michaelblight, 2024-01-30:
### The problem

I had this code a long time ago (as in a few years), and it worked to provide the source file in a sensor:
```
text_sensor:
  - platform: template
    name: "${name} Base"
    id: "${id}_base"
    lambda: |-
      return { __FILE__ };
    update_interval: 1h
```
However now it always produces "<unicode string>".

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.4

### What platform are you using?

ESP32

### Board

m5stick-c

### Component causing the issue

text_sensor

### Additional information

I am also using the following and this works fine:
```
  - platform: template
    name: "${name} Compile Date"
    id: "${id}_compile_date"
    lambda: |-
      return { __DATE__ " " __TIME__ };
    update_interval: 1h
```
[image]

marshn, 2024-02-01:
I use \_\_FILE\_\_  in exactly the same way, and it broke for me with ESPHome v2023.12. It has been resolved by PR https://github.com/esphome/esphome/pull/6139 which didn't make it into v2023.12.9, I'm hoping the fix will be included in v2023.12.10 or v2024.2. Also see https://github.com/esphome/issues/issues/5354 https://github.com/esphome/issues/issues/5377 https://github.com/esphome/issues/issues/5423.

marshn, 2024-02-23:
esphome 2024.2 has has resolved this issue, for me at least.