esphome/issues #5450: LED Strip flickering because of interrupts while filling the PIO queue

MrEbbinghaus, 2024-01-30:
### The problem

@Papa-DMan tagging you, because you are listed as the code owner.

I noticed flickering of longer LED strips, when using the `rp2040_pio_led_strip` component on my Pico W. 
[image]
(video is at half speed)

I tracked it down to the PIO statemachine stalling on a low bit, which is interpreted by the LEDs as the reset signal. 

This stall only occurs occasionally, when an interrupt is fired while filling the PIO queue in `write_state`. 

https://github.com/esphome/esphome/blob/92798751c2a0860dc5f85cf9f34395372cca72d6/esphome/components/rp2040_pio_led_strip/led_strip.cpp#L72-L80

This problem gets worse the longer the led strip is and might not be that noticeable on shorter strips. Especially if the entire data fits into the PIO queue. (≤10 RGB LEDs)

I improvised a fix by turning interrupts off during the loop and then reenable them. No more flickering. 🎉 But I don't know what side effects this may have. 😕

```cpp
#include <hardware/sync.h>

uint32_t irqs = save_and_disable_interrupts();
// loop
restore_interrupts(irqs);
```

Additionally this problem could be solved in different ways: 

1. Disable interrupts for the loop. (I don't know what side effects this may have.)
2. Run the fill loop on the second core. (<- could be the best solution)
3. Modify the PIO code so that it stalls while the output is high. But then it needs to be able to discern a stall from the end of the transmission. 

----

This is my first time with C++, so I decided against a pull request for now.

I have tested other improvements for `rp2040_pio_led_strip` as well, like shaving down the fill loop and the PIO code. 

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RP2040

### Board

Pico W

### Component causing the issue

rp2040_pio_led_strip

MrEbbinghaus, 2024-01-30:
Or an even better way may be to make use of DMA to pipe the bits to the PIO state machine.

Papa-DMan, 2024-01-30:
Yeah I've noticed this issue too, but was unsure of the source causing it. 
I pulled up the documentation for the DMA channels and I think I have a working solution here, but I don't have a physical test environment with me right now.

This branch here : [https://github.com/Papa-DMan/esphome/tree/pio_led_dma](url) removes the for loop in write_state and just uses a DMA channel. If you wouldn't mind giving it a try, I'd appreciate it.

MrEbbinghaus, 2024-02-01:
Nice, I checked your code and first, you need to use `dma_channel_transfer_from_buffer_now` instead of `dma_channel_transfer_to_buffer_now`. Since, you want to copy from `_buf` to the FIF0.

Second, when the DMA copies 8bit to the state machines FIF0 (which has 32bits words), it does so with byte lane replication: 

```
Size                 Content in FIFO
32bit AA BB CC DD -> AA BB CC DD
16bit AA BB CC DD -> AA BB AA BB | CC DD CC DD
 8bit AA BB CC DD -> AA AA AA AA | BB BB BB BB | CC CC CC CC | DD DD DD DD
```

So if you want to set the first led to red, the buffer contains:
```
 BUF:       (G)00       (R)FF       (B)00 ...
FIFO: 00 00 00 00 FF FF FF FF 00 00 00 00 
 OUT:    00 00 00    FF FF FF    00 00 00 (the current implementation pulls in only 24bit)
```

I think the best way to solve this is to have the PIO always pull and transmit byte after byte. This way you don't have to differentiate between 23bit and 32bit color values. 

Currently I'm using the code from https://github.com/raspberrypi/pico-examples/blob/master/pio/ws2812/ws2812.pio for that. It auto-pulls 8bit into the OSR.
But it doesn't work for custom timings. (Although everything related to the WS2812 should work.)
 
```asm
.program rp2040_pio_led_strip_{id}
.side_set 1
.define public T1 2
.define public T2 5
.define public T3 3

.wrap_target
bitloop:
    out x, 1       side 0 [T3 - 1] 
    jmp !x do_zero side 1 [T1 - 1] ; Branch on the bit we shifted out. Positive pulse
do_one:
    jmp bitloop    side 1 [T2 - 1] ; Continue driving high, for a long pulse
do_zero:
    nop            side 0 [T2 - 1] ; Or drive low, for a short pulse
.wrap
```
```cpp
#include "hardware/clocks.h"

static inline void rp2040_pio_led_strip_driver_{id}_init(PIO pio, uint sm, uint offset, uint pin, float freq) {
    pio_gpio_init(pio, pin);
    pio_sm_set_consecutive_pindirs(pio, sm, pin, 1, true);

    pio_sm_config c = rp2040_pio_led_strip_{id}_program_get_default_config(offset);
    sm_config_set_sideset_pins(&c, pin);
    sm_config_set_out_shift(&c, false, true, 8);
    sm_config_set_fifo_join(&c, PIO_FIFO_JOIN_TX);

    float f = 800000;
    int cycles_per_bit = rp2040_pio_led_strip_{id}_T1 + rp2040_pio_led_strip_{id}_T2 + rp2040_pio_led_strip_{id}_T3;
    float div = clock_get_hz(clk_sys) / (f * cycles_per_bit);
    sm_config_set_clkdiv(&c, div);

    pio_sm_init(pio, sm, offset, &c);
    pio_sm_set_enabled(pio, sm, true);
}
```

Papa-DMan, 2024-02-01:
I didn't realize that the DMA performs byte lane replication, but sure enough I pull up the docs and it's right there. I also agree that best way to solve this is to have the PIO always pull and transmit byte after byte. I don't recall why the current iteration is the way that it is.
The PIO program that is used is generated in light.py for each chipset / set of timings.
`.wrap_target
awaiting_data:
    ; Wait for data in FIFO queue
    out null, 24 ; discard the byte lane replication of the FIFO since we only need 8 bits
    pull block ; this will block until there is data in the FIFO queue and then it will pull it into the shift register
    set y, 7 ; set y to the number of bits to write counting 0, (always 7 because we are doing one word at a time)

mainloop:
    ; go through each bit in the shift register and jump to the appropriate label
    ; depending on the value of the bit

    out x, 1
    jmp !x, writezero
    jmp writeone

writezero:
    ; Write T0H and T0L bits to the output pin
    set pins, 1 [{t0h}]
{nops_t0h}
    set pins, 0 [{t0l}]
{nops_t0l}
    jmp y--, mainloop
    jmp awaiting_data

writeone:
    ; Write T1H and T1L bits to the output pin
    set pins, 1 [{t1h}]
{nops_t1h}
    set pins, 0 [{t1l}]
{nops_t1l}
    jmp y--, mainloop
    jmp awaiting_data

.wrap`

I'm pretty sure I had to do it this way because with some timings we cannot use the side-set feature.

Papa-DMan, 2024-02-01:
I've set up a PIO emulator from [rp2040pio](https://github.com/soundpaint/rp2040pio) and I've verified that the updated PIO code should run as expected and be able to deal with the replication by ensuring that the ISR and FIFO gets cleared before pulling fresh data from the fifo, which should trigger the DREQ pulling in the next 8 bits. 

Here are the commands I used to verify using the rp2040pio emulator:
```
load -f piocode.hex
sm -p 0 -s 0 +e
pinctrl -p 0 -s 0 --set-count=1 --set-base=0
fifo -p 0 -s 0 -e -v 0x0A0A0A0A -t

closed 2024-05-15 by jesserockz

rici44, 2024-06-29:
Platform RP2040 PIO LED Strip compiled in ESPHOME 2024.4.2 works with https://docs.m5stack.com/en/unit/rgb 
Platform RP2040 PIO LED Strip compiled in ESPHOME 2024.6.4 is not working (flickering, wrong colour, not all rgb)

This is my config (the same in both versions)

`light:
  - platform: rp2040_pio_led_strip
    name: led_strip
    id: led_strip
    pin: GPIO04
    num_leds: 3
    pio: 0
    rgb_order: GRB
    chipset: SK6812`