esphome/issues #5455: Wrong driver for waveshare e-ink displays

SO5HMI, 2024-01-31:
### The problem

esp32:
  board: esp32dev
  framework:
    type: arduino

pi:
  clk_pin: 13
  mosi_pin: 14

display:
  - platform: waveshare_epaper
    id: eink
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 4.2in
    update_interval: 10s
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");

I use original waveshare esp32 e-ink driver board with no other sensors connected

In this config the display doesnt change, but if i change the model to different i see Hello World but in the corner and the not used part of the display has w/b random noise.

working display models:
1.54in - display is changing but, screen is black
1.54inv2 - works part of display according to resolution
2.90in- display is changing but, screen is black
2.90inv2 - works part of display according to resolution
4.2in - not working
4.2in-bv2 - not working
5.83in - not working
5.83inv2 - not working
7.50in - not working
7.50in-bV2 - not working
7.50inV2 - not working

i used this display a half year a go and its working great.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core 2024.1.6

### What platform are you using?

ESP32

### Board

waveshare E-Paper ESP32 Driver Board

### Component causing the issue

SPI, 

### Example YAML snippet

```yaml
esphome:
  name: panel-salon
  friendly_name: Panel Salon

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:

ota:

wifi:
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
 
captive_portal:
    
#i2c:
#  sda: 21
# scl: 22
#  scan: true
#  id: bus_a

spi:
  clk_pin: 13
  mosi_pin: 14
  
font:
  - file: 'config/fonts/liberation/LiberationSansNarrow-Bold.ttf'
    id: font1
    size: 20 

#sensor:
#  - platform: bme280
#    temperature:
#      name: "Salon temperatura"
#      oversampling: 16x
#      accuracy_decimals: 1
#    pressure:
#      name: "Salon cisnienie"
#      accuracy_decimals: 1
#    humidity:
#      name: "Salon wilgotnosc"
#      accuracy_decimals: 1
#    address: 0x76
#    update_interval: 60s

#pn532_i2c:
 # update_interval: 1s

display:
  - platform: waveshare_epaper
    id: eink
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 1.54inv2
    update_interval: 10s
    lambda: |-
      it.print(0, 0, id(font1), "1.54inv2");
```

MrMDavidson, 2024-02-16:
See also #5369. I've got a [PR](https://github.com/esphome/esphome/pull/6209) that seems to fix the issue for my device.

SO5HMI, 2024-02-16:
Hi @MrMDavidson, thank you. I have tried this solution and ... didn't work for me. Screen is living — the refreshing (fill rectangle is working), but i don't get any next lambda drawings. The screen is black.

MrMDavidson, 2024-02-16:
Can you paste your config?

SO5HMI, 2024-02-16:
Yes, this is my config. I try to run display, there is no other code except beginning (board, wifi etc.)

captive_portal:

external_components:
  - source: github://pr#6209
    components: [ waveshare_epaper ]
    
font:
  - file: 'config/fonts/liberation/LiberationSansNarrow-Bold.ttf'
    id: font1
    size: 10

spi:
  clk_pin: 13
  mosi_pin: 14

display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 4.20in-v2
    full_update_every: 10
    reset_duration: 2ms
     pages:
      - id: paint_black
        lambda: |-
          const auto height = it.get_height();
          const auto width = it.get_width();

          it.filled_rectangle(0, 0, width, height, COLOR_ON);
      - id: paint_white
        lambda: |-
          const auto height = it.get_height();
          const auto width = it.get_width();

          it.filled_rectangle(0, 0, width, height, COLOR_OFF);
      - id: clear
        lambda: |-
          it.clear();
      - id: real_page_im_drawing
          it.print(10, 50, id(font1), "Hello World!");

MrMDavidson, 2024-02-16:
Ah, I'm not seeing any page change code. In my setup I'm navigating through the pages in response to a button press with `display.page.show_next`. if you don't have a physical button you could use a timer or something. Alternatively try just getting rid of the first 3 pages; they're just there to force some cleanup of the display.

GraffJosh, 2024-06-24:
I'm just going to poke this issue to unStale it, because I'm definitely having trouble with the esp32 Driver Board and the 4.2 display (I think I have the version 1: WFT0420CZ15)

I'm not getting any update on the display with the current waveshare epaper platform:

`
spi:
  id: epaper_spi
  clk_pin: 13
  mosi_pin: 14
  miso_pin: 19

display:
  - platform: waveshare_epaper
    spi_id: epaper_spi
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 4.20in
    update_interval: 30s
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");`