esphome/issues #5459: Atom S3 Lite Bluetooth Proxy adds no BT-devices

ViezeVisNL, 2024-02-02:
### The problem

I setup the Atom S3 Lite manually as Bluetooth Proxy, but no devices are added. With a BT Inspector App I can see BLE devices are discovered, but the proxy is not adding them. With the verbose-logging I also see devices popping up. I've added a sensor for 1 device and I can see the -dB value. In pairing mode the device will not be added somehow. 

Conf:
```
esphome:
  name: atoms3bluetooth-proxy-2
  friendly_name: AtomS3Bluetooth-proxy-2

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "****"

ota:
  password: "****"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Atoms3Bluetooth-Proxy-2"
    password: "****"

captive_portal:
    
# Setup BLE config (tried with interval window too)
esp32_ble_tracker:
  scan_parameters:
#    interval: 1100ms
#    window: 1100ms
    active: true

# Enable BT proxy
bluetooth_proxy:
  active: true
```

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.12.3

### What platform are you using?

ESP32

### Board

M5Stack Atom S3 Lite

FrancescoC87, 2024-02-13:
I confirm the same problem with an ESP32 S3 N16R8

closed 2024-06-27 by github-actions[bot]

anthonykeane, 2024-08-31:
Same issue