esphome/issues #5460: DHT fail after wakes up from deep sleep

kjudom, 2024-02-03:
### The problem

DHT will never work after it wakes up from deep sleep.

### Which version of ESPHome has the issue?

2023.12.19

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP8266

### Board

ESP01

### Component causing the issue

DHT22

### Example YAML snippet

```yaml
sensor:
  - platform: dht
    pin: GPIO02
    model: DHT22
    temperature:
      name: "Bed Room Temperature"
    humidity:
      name: "Bed Room Humidity"
    update_interval: 30s

deep_sleep:
  run_duration: 180s   
  sleep_duration: 1min
```

### Anything in the logs that might be useful for us?

```txt
[09:10:29][D][dht:048]: Got Temperature=32.6°C Humidity=68.9%
[09:10:29][D][sensor:093]: 'Bed Room Temperature': Sending state 32.60000 °C with 1 decimals of accuracy
[09:10:29][D][sensor:093]: 'Bed Room Humidity': Sending state 68.90000 % with 0 decimals of accuracy
[09:10:59][D][dht:048]: Got Temperature=32.6°C Humidity=68.9%
[09:10:59][D][sensor:093]: 'Bed Room Temperature': Sending state 32.60000 °C with 1 decimals of accuracy
[09:10:59][D][sensor:093]: 'Bed Room Humidity': Sending state 68.90000 % with 0 decimals of accuracy
[09:11:16][I][deep_sleep:117]: Beginning Deep Sleep
[09:11:16][I][deep_sleep:119]: Sleeping for 60000000us
INFO Processing expected disconnect from ESPHome API for dht22-esp01 @ 192.168.0.100
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for dht22-esp01 @ 192.168.0.100: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.0.100', port=6053))]: [Errno 113] Connect call failed ('192.168.0.100', 6053) (SocketAPIError)
INFO Trying to connect to dht22-esp01 @ 192.168.0.100 in the background
INFO Successfully connected to dht22-esp01 @ 192.168.0.100 in 0.006s
INFO Successful handshake with dht22-esp01 @ 192.168.0.100 in 2.883s
[09:12:35][D][api:102]: Accepted 192.168.0.103
[09:12:35][D][api:102]: Accepted 192.168.0.103
[09:12:35][W][api.connection:091]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[09:12:35][W][api.connection:091]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[09:12:36][W][dht:169]: Requesting data from DHT failed!
[09:12:36][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[09:12:36][D][sensor:093]: 'Bed Room Temperature': Sending state nan °C with 1 decimals of accuracy
[09:12:36][D][sensor:093]: 'Bed Room Humidity': Sending state nan % with 0 decimals of accuracy
[09:12:57][D][api:102]: Accepted 192.168.0.103
```

kjudom, 2024-02-03:
I tried to use a static IP, but the result was the same.

fran6120, 2024-02-04:
I'm using the deep_sleep and dht11 components on an esp32s2 running esphome 2023.12.9 and no problems

kjudom, 2024-02-04:
I changed to dht11 and it works. So the problem should be the dht22 or the library.

closed 2024-06-18 by github-actions[bot]