esphome/issues #5461: I2S Audio Speaker gets stopped before TTS Steam End of Audio, spontaneously

michaelberg79, 2024-02-03:
### The problem

Helly
I try to get running a Raspiaudio muse proto for voice assistant. I installed the voice assistant module with the esp-idf framework. Its so far its running... sometimes. The voice commands are recognized and executed. But I get the response output only by a chance of 50%. I tried several times and got some a differences in the log files between good and bad runs. But I cant get any details of this behavior.

anyone who can help with this?

The debug level is VERBOSE...

best wishes and thx for help 
Michael

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32-IDF

### Board

raspiaudio muse proto

### Component causing the issue

i2c_speaker

### Example YAML snippet

```yaml
---
esphome:
  name: muse-proto
  friendly_name: RaspiAudio Muse Proto
  name_add_mac_suffix: false
  project:
    name: raspiaudio.muse-proto-voice-assistant
    version: "1.0"
  min_version: 2023.11.1

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

logger:
  level: VERBOSE
ota:

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/raspiaudio-muse-proto.yaml@main

wifi:
  on_connect:
    - delay: 5s # Gives time for improv results to be transmitted
    - ble.disable:
  on_disconnect:
    - ble.enable:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

[... 285 lines cut ...]
[11:12:53][D][esp-idf:000]: I (322468) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=8

[11:12:53][D][i2s_audio.speaker:164]: Started I2S Audio Speaker
[11:12:55][D][voice_assistant:519]: Event Type: 99
[11:12:55][D][voice_assistant:665]: TTS stream end
[11:12:55][D][voice_assistant:283]: End of audio stream received
[11:12:55][D][voice_assistant:412]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[11:12:55][D][voice_assistant:418]: Desired state set to RESPONSE_FINISHED
[11:12:56][D][i2s_audio.speaker:167]: Stopping I2S Audio Speaker
[11:12:56][D][i2s_audio.speaker:178]: Stopped I2S Audio Speaker
[11:12:56][D][light:036]: 'RaspiAudio Muse Proto' Setting:
[11:12:56][D][light:051]:   Brightness: 60%
[11:12:56][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[11:12:56][D][light:085]:   Transition length: 1.0s
[11:12:56][D][voice_assistant:315]: Speaker has finished outputting all audio
[11:12:56][D][voice_assistant:412]: State changed from RESPONSE_FINISHED to IDLE
[11:12:56][D][voice_assistant:418]: Desired state set to IDLE
[11:12:56][D][voice_assistant:412]: State changed from IDLE to START_MICROPHONE
[11:12:56][D][voice_assistant:418]: Desired state set to WAIT_FOR_VAD
[11:12:56][D][voice_assistant:153]: Starting Microphone
[11:12:56][D][voice_assistant:412]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[11:12:56][D][esp-idf:000]: I (325949) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[11:12:56][D][voice_assistant:412]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
```

michaelberg79, 2024-02-04:
Hallo,
I give it more tries. The line 

> [i2s_audio.speaker:167]: Stopping I2S Audio Speaker

appears also in successful runs. So thats not the reason for none responses.

closed 2024-06-18 by github-actions[bot]

LumenSoftNL, 2024-06-18:
try pr#6718 and see if that will help you.