esphome/issues #5462: need to add mac_suffix also for hostname internal use

michaelberg79, 2024-02-03:
### The problem

Hallo,
example domain: device

I cant access the logfile if I enable name_add_mac_suffix. The mac suffix is added for extrernal use. For exampe if getting the dhcp ip. So the dhcp is creating the bind entry with device-34bc23.local. 
But if i try to connect the device log files wireless in esp-home he try to use device.local and cant resolve the ip address.

Currently i disabled the name_add_mac_suffix. But in my mention this is a wrong behavior.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32-IDF

### Board

raspiaudio muse proto

### Component causing the issue

wifi

fpovoski, 2024-05-23:
Same issue here. Want to obtain mac_suffix name for internal use like for mqtt topics. Also using mac_suffix, esphome still uses device name for wifi flashing.

tmjpugh, 2024-06-05:
having same issue

ESPHome 2024.5.5 and several version prior

```
Wrote 0x37b830 bytes to file /config/.esphome/build/box3/.pioenvs/box3/firmware-factory.bin, ready to flash to offset 0x0
======================== [SUCCESS] Took 603.27 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of box3.localdomain
ERROR Error resolving IP address of box3.localdomain. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: [Errno -2] Name or service not known
```

relevant parts from config
```

substitutions:
  name: box3
  friendly_name: Box3

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
```

router shows device hostname as box3-5ac0dc
routher shows device mac as 30:30:f9:5a:c0:dc  

dig box3.localdomain FAIL
dig box3-5ac0dc.localdomain OK

homeassistant connecting and discovery without issue
ping box3-5ac0dc.localdomain from ESPhome OK