esphome/issues #5464: Logs Flooding with RATGDO Security 1.0 Firmware

Steven4x4, 2024-02-03:
### The problem

I'm new to ESP Home. RATGDO is my first ESP Home device.
I'm not entirely sure what ESP Platform the v2.53i  is.

When I go to the ESP Home Devices Webpage Directly...
With the RATDGO v2.53i board with the Security 1.0 firmware it is flooding the logs every other second with:
14:46:32 | [D] | [ratgdo:210] | Lock state=UNLOCKED
14:46:33 | [D] | [ratgdo:204] | Light state=OFF
14:46:33 | [D] | [ratgdo:210] | Lock state=UNLOCKED
14:46:35 | [D] | [ratgdo:204] | Light state=OFF
14:46:35 | [D] | [ratgdo:210] | Lock state=UNLOCKED
14:46:36 | [D] | [ratgdo:204] | Light state=OFF
Video - https://www.youtube.com/watch?v=F8vxAVPcjjo

For reference on the Security 2.0 firmware, it only does this every 5 minutes. (Which is about what I would expect.

Secondly, inside of home assistant, on the integrations > esp home > garage door device.
When you try to use the feature where you can open to a specific percentage. It does no work on the 1.0 firmware, but does work on the 2.0 firmware. I reach out to RATDGO directly, and he directed me to post here.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Board

RATGDO v2.53i

### Example YAML snippet

```yaml
ESP HOME YAML:
substitutions:
  name: ratgdov25i-ca08b2
  friendly_name: Garage Door 1
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <REDACTED>

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

ssieb, 2024-02-06:
The logs issue is to do with their component, not esphome.
If you can reproduce the cover issue using non-external components, then create a new issue with just that.

closed 2024-02-06 by ssieb

Steven4x4, 2024-02-06:
> The logs issue is to do with their component, not esphome. If you can reproduce the cover issue using non-external components, then create a new issue with just that.

Thank you for your response.
Are you saying the log issue could be fixed by RATGDO?
What do you mean by "using a non-external component"?

ssieb, 2024-02-06:
The log issue can only be fixed by them because that's where the messages are coming from.
And the cover is also an external component, not part of esphome.  So if you can reproduce it with one of the built-in esphome covers, then it's an issue for here.  Otherwise, it's an issue for the creator of the component.