esphome/issues #5465: MQTT client from ESP-IDF framework used on ESP32 uses too much RAM

peterhalicky, 2024-02-04:
### The problem

Pull request [5157](https://github.com/esphome/esphome/pull/5157) introduced a different MQTT client than what was used before on ESP32. This new client, however, uses nearly 21kB more memory than the one that was in use before (Arduino). 

With commit preceding 9980b9972fcac080a1eb1987107484644e7a8d1d, I get this:
```
[08:00:29][D][debug:072]: Free Heap Size: 53036 bytes
```

With commit 9980b9972fcac080a1eb1987107484644e7a8d1d, I get this:
```
[08:03:02][D][debug:072]: Free Heap Size: 32540 bytes
```

This difference can mean whether a configuration will work just fine, or will be crashing horribly. The PR doesn't say why is the ESP-IDF mqtt client better - so unless there's a very good reason to use ESP-IDF mqtt client, I'd say esphome should switch back to using the Arduino library.

### Which version of ESPHome has the issue?

commit 9980b997

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
esphome:
  name: something
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

debug:
  update_interval: 5s

logger:

ota:

mqtt:
  broker: host-name

web_server:
  port: 80

esp32_ble_tracker:

xiaomi_ble:

text_sensor:
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"

sensor:
  - platform: debug
    free:
      name: "Heap Free"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"

  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:99:20:13"
    temperature:
      name: "AAA Temperature"
    humidity:
      name: "AAA Humidity"
    battery_level:
      name: "AAA Battery level"
    battery_voltage:
      name: "AAA Battery voltage"
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:E1:74:A9"
    temperature:
      name: "BBB Temperature"
    humidity:
      name: "BBB Humidity"
    battery_level:
      name: "BBB Battery level"
    battery_voltage:
      name: "BBB Battery voltage"
```

closed 2024-06-18 by github-actions[bot]

spuder, 2024-10-13:
Correct me if I'm wrong, but can't you switch back to the arduino framework with something like this?

```
esphome
  framework: 
    name: arduino
```

peterhalicky, 2024-10-13:
Arduino framework is used by default anyway; that config won't change anything: https://esphome.io/components/esp32.html#esp32-arduino-framework