esphome/issues #5473: Please fix board type in Shelly Plus 1 Mini

foltymat, 2024-02-07:
### The problem

Hi,

the default esphome sketch for [Shelly Plus 1 Mini](https://github.com/esphome/esphome-devices/blob/main/src/docs/devices/Shelly-Plus-1-Mini/index.md) has a wrong board type in the configuration.

The board mentioned is esp32-c3-devkitm-v1 while the correct one is esp32-c3-devkitm-1.

The "v" at the end is not correct.

Thanks

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

ssieb, 2024-02-07:
Make a PR on the devices site.

foltymat, 2024-02-08:
Issue closed. Created a PR instead.

closed 2024-02-08 by foltymat