esphome/issues #5474: Waveshare 7.5v2 flickering while updating

Barzille, 2024-02-07:
### The problem

Hi,
I am unsure if this is the expected behavior.
I am using a wave share 7.5 v2 display directly connected to an ESP32 from Waveshare.
It works fine so far, but every time it refreshes it does it completely and looks like it is doing it more than once, please see the attached video. Is that a normal behavior or should it update a little less annoyingly ;)
And if it should work differently, can you guide me what I have to change?

https://github.com/esphome/issues/assets/1096322/f2a0ea63-be43-4e80-aef2-396c99aeb3d7

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Board

ESP32 Driver Board

### Example YAML snippet

```yaml
font:
  - file: "gfonts://Roboto"
    id: status_font
    size: 18
    
  - file: 'gfonts://Roboto'
    id: watch_font
    size: 90
    glyphs:
      [':', ".", '0', '1', '2', '3', '4', '5', '6', '7', '8', '9']
  
  - file: 'gfonts://Material+Symbols+Outlined'
    id: icon_font_25
    size: 25
    glyphs: [
      "\U0000e425", # mdi-timer
      "\U0000f0a4" # battery
    ] 
  
    
sensor:
  - platform: homeassistant
    id: battery_charge
    entity_id: sensor.battery_state_of_capacity
    unit_of_measurement: "Wh"
    device_class: "ENERGY_STORAGE"

text_sensor:
  - platform: homeassistant
    id: pv_power
    entity_id: sensor.input_power_with_efficiency_loss
    icon: "mdi:solar-power"
  - platform: homeassistant
    id: mytime 
    entity_id: sensor.time
  

image:
  - file: mdi:solar-power
    id: solar
    resize: 80x80

spi:
  clk_pin: 13
  mosi_pin: 14
  
display:
  - platform: waveshare_epaper
    id: epaper
    cs_pin: 15
    busy_pin:
      number: 25
      inverted: true
    reset_pin: 26
    dc_pin: 27
    model: 7.50inV2
    #model: 7.50inV2alt
    update_interval: 30s
    reset_duration: 2ms
    
    lambda: |
      ESP_LOGI("display", "Updating..");
      it.printf(1, 60, id(status_font), TextAlign::BOTTOM_LEFT, "PV Erzeugung akt.: %s W", id(pv_power).state.c_str());
      
      it.printf(1, 80, id(icon_font_25),  TextAlign::BASELINE_LEFT, "\U0000f0a4");
      it.printf(30, 80, id(status_font), TextAlign::BOTTOM_LEFT, "Batterie Ladung: %.2f%%", id(battery_charge).state);
      
      //Time
      it.printf(380, 100, id(watch_font),TextAlign::BOTTOM_LEFT, "%s", id(mytime).state.c_str());
      it.line(380, 108, 600, 108);
      it.printf(1, 220, id(icon_font_25),  TextAlign::BASELINE_LEFT, "\U0000e425");
```

stephanvierkant, 2024-04-25:
Already found a solution? I've got the same problem.

team-bfw, 2024-04-25:
No, it seems to be a normal behaviour :(

Barzille, 2024-04-25:
No, this seems to be a normal behaviour :(