esphome/issues #5475: esphome web server v2 does not update a second number template value when using number.set in an automation

philpownall, 2024-02-08:
### The problem

I am using 2 number templates and some button templates to control servos.  One the v2 web server, the second number template does not update when a button is pressed on the web server v2 web page (which calls number.set on the number template, which calls servo.write).  The first number template updates, and reflects the value of the number.set

If I am using this incorrectly, please let me know.

### Which version of ESPHome has the issue?

ESPHome 2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

not using HA in this case, I am using esphome dashboard standalone.

### What platform are you using?

ESP32

### Board

Chip: Nodemcu v1.1 Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=3

### Component causing the issue

web server: version: 2

### Example YAML snippet

```yaml
# local LAN web server
web_server:
  version: 2
  include_internal: true
  local: true

# Define a number template to control servo 1
number:
  - platform: template
    name: Servo 1
    id: servo1
    min_value: 0
    restore_value: False
    max_value: 100
    optimistic: true
    step: 1
    set_action:
      then:
        - servo.write:
            id: my_servo1
            level: !lambda 'return x / 100.0;'

# Define a number template to control servo 2
  - platform: template
    name: Servo 2
    id: servo2
    min_value: 0
    restore_value: False
    max_value: 100
    step: 1
    set_action:
      then:
        - servo.write:
            id: my_servo2
            level: !lambda 'return x / 100.0;'

# buttons for servo control presets
# note that we use number.set rather than servo.write
# so that the number slider updates on the web page and in HA
button:
  - platform: template
    name: "servos_stop"
    icon: "mdi:stop"
    on_press:
      then:
        - number.set:
            id: servo1
            value: 35
        - number.set:
            id: servo2
            value: 35
```

### Anything in the logs that might be useful for us?

```txt
no error message is generated.
```

### Additional information

I am running esphome dashboard, and opening the web page for this node (espnode.local).

ssieb, 2024-02-08:
Your second number is missing `optimistic`.

closed 2024-02-08 by ssieb

philpownall, 2024-02-08:
Thank you!  I must have wasted 4 hours on that and never saw it...
Regards, and sorry for the inconvenience
Phil