esphome/issues #5476: New daly K-type bms not working with esphome

Kimbo01, 2024-02-08:
### The problem

The nodemcu does not receiver any info from the bms

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.0

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Example YAML snippet

```yaml
esphome:
  name: nodemcuv3-1
#  includes:
#    - custom/digipot1.h

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:

web_server:

ota:
  password: "a35e5982e34c323e347f4e7fae976cdf"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nodemcuv3-1 Fallback Hotspot"
    password: "jUOA4ifP8Wqp"

captive_portal:

uart:
  tx_pin: TX
  rx_pin: RX
  baud_rate: 9600
  rx_buffer_size: 512

daly_bms:
  update_interval: 10s

binary_sensor:
  - platform: daly_bms
    charging_mos_enabled:
      name: "Daly Charging MOS"
      id: bin_daly_chg_mos # binary MOS sensor must have ID to use with switch
      internal: True # but you can make it internal to avoid duplication
    discharging_mos_enabled:
      name: "Daly Discharging MOS"
      id: bin_daly_dischg_mos # binary MOS sensor must have ID to use with switch
      internal: True # but you can make it internal to avoid duplication
sensor:
  - platform: daly_bms
    voltage:
      name: "Battery Voltage"
    current:
      name: "Battery Current"
    battery_level:
      name: "Battery Level"
    max_cell_voltage:
      name: "Max Cell Voltage"
    max_cell_voltage_number:
      name: "Max Cell Voltage Number"
    min_cell_voltage:
      name: "Min Cell Voltage"
    min_cell_voltage_number:
      name: "Min Cell Voltage Number"
    max_temperature:
      name: "Max Temperature"
    max_temperature_probe_number:
      name: "Max Temperature Probe Number"
    min_temperature:
      name: "Min Temperature"
    min_temperature_probe_number:
      name: "Min Temperature Probe Number"
    remaining_capacity:
      name: "Remaining Capacity"
    cells_number:
      name: "Cells Number"
    temperature_1:
      name: "Temperature 1"
    temperature_2:
      name: "Temperature 2"
    cell_1_voltage:
      name: "Cell 1 Voltage"
    cell_2_voltage:
      name: "Cell 2 Voltage"
    cell_3_voltage:
      name: "Cell 3 Voltage"
    cell_4_voltage:
      name: "Cell 4 Voltage"
    cell_5_voltage:
      name: "Cell 5 Voltage"
    cell_6_voltage:
      name: "Cell 6 Voltage"
text_sensor:
  - platform: daly_bms
    status:
      name: "BMS Status"
switch:
  - platform: template
    name: "Daly Charging MOS"
    lambda: |-
      if (id(bin_daly_chg_mos).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - uart.write:
          data: [0xA5, 0x40, 0xDA, 0x08, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC8]
      - logger.log:
          format: "Send cmd to Daly: Set charge MOS on"
    turn_off_action:
      - uart.write:
          data: [0xA5, 0x40, 0xDA, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC7]
      - logger.log:
          format: "Send cmd to Daly: Set charge MOS off"

  - platform: template
    name: "Daly Discharging MOS"
    lambda: |-
      if (id(bin_daly_dischg_mos).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - uart.write:
          data: [0xA5, 0x40, 0xD9, 0x08, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC7]
      - logger.log:
          format: "Send cmd to Daly: Set discharge MOS on"
    turn_off_action:
      - uart.write:
          data: [0xA5, 0x40, 0xD9, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC6]
      - logger.log:
          format: "Send cmd to Daly: Set discharge MOS off"
```

kamachips, 2024-04-05:
same question

Koxx3, 2024-07-30:
I also have the issue.

Here is the protocol I have :
[KVMS intranet communication UART protocol (customer version)..xlsx](https://github.com/user-attachments/files/16429626/KVMS.intranet.communication.UART.protocol.customer.version.xlsx)

patagonaa, 2024-09-28:
I have the same issue and have done some digging in the PC software to try and understand how the new protocol works.

DALY calls this protocol "Modbus" (both in the PC software and the documentation @Koxx3 posted), which would be great because ESPHome already has Modbus support, however as it turns out, it's _not quite Modbus_, which really sucks.

The Modbus address of the BMS is the BMS id (1 by default) + 128, however in the response the BMS doesn't respond with its Modbus address, but with its Modbus address - 48, which means that no standard Modbus software will work with that (because it receives a response from a device address it didn't request).

To read the cell voltages 1-16 (register 0 to 15), the request looks like this (valid modbus)
```
0x81 // device address: offset 128 + 1 = 129
0x03 // function code: 3 = "read holding"
0x00 0x00 // start from register 1
0x00 0x10 // read 16 registers
0x5B 0xC6 // CRC
```

and the BMS responds with the wrong device address

```
0x51 // device address: offset 128 + 1 - 48 = 81, should be 129!
0x03 // function code 3 = "read holding"
0x20 // response length (32 bytes)
0x0c 0xfd // first register (3325 = 3.325V)
[...]
0xe4 0xd5 // CRC
```

Maybe I'll try and implement this once I have the time.

patagonaa, 2024-10-10:
@Kimbo01 @kamachips @Koxx3
I'm somewhat done writing and testing the integration. 
The documentation preview is also available under https://deploy-preview-4296--esphome.netlify.app/components/sensor/daly_hkms_bms

You can try my current development version using something like

```yaml
external_components:
  - source: github://patagonaa/esphome@daly_hkms
    components: [daly_hkms_bms]
    refresh: 10s

uart:
  id: bms_uart
  rx_pin: GPIO04
  tx_pin: GPIO05
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

modbus:
  uart_id: bms_uart

daly_hkms_bms:
  - id: bms_id
    address: 1
    update_interval: 10s

sensor:
  - platform: daly_hkms_bms
    daly_hkms_bms_id: bms_id
    voltage:
      name: "Battery Voltage"
    current:
      name: "Battery Current"
    battery_level:
      name: "Battery Level"
    max_cell_voltage:
      name: "Cell Voltage Max"
    max_cell_voltage_number:
      name: "Cell Voltage Max Number"
    min_cell_voltage:
      name: "Cell Voltage Min"
    min_cell_voltage_number:
      name: "Cell Voltage Min Number"
    max_temperature:
      name: "Temperature Max"
    max_temperature_probe_number:
      name: "Temperature Max Probe Number"
    min_temperature:
      name: "Temperature Min"
    min_temperature_probe_number:
      name: "Temperature Min Probe Number"
    remaining_capacity:
      name: "Remaining Capacity"
    cells_number:
      name: "Cells Number"
    temperature_1:
      name: "Temperature 1"
    temperature_2:
      name: "Temperature 2"
    temperature_mos:
      name: "Temperature MOS"
    temperature_board:
      name: "Temperature Board"
    cell_1_voltage:
      name: "Cell Voltage  1"

text_sensor:
  - platform: daly_hkms_bms
    daly_hkms_bms_id: bms_id
    status:
      name: "Status"

binary_sensor:
  - platform: daly_hkms_bms
    daly_hkms_bms_id: bms_id
    charging_mos_enabled:
      name: "Charging MOS enabled"
    discharging_mos_enabled:
      name: "Discharging MOS enabled"
    precharging_mos_enabled:
      name: "Precharging MOS enabled"
    balancing_active:
      name: "Balancing active"
```