esphome/issues #5483: New devices are not sorted alphabetically.

tomlut, 2024-02-12:
### The problem

After adding a new device it is appears at the end of the add-on devices dashboard. 

Refreshing the page or clearing the web browser cache does not change this. 

Only restarting the add-on sorts the devices correctly (alphabetically). 

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 	core-2024.2.1

### What platform are you using?

ESP32

### Board

QuinLED ESP32

### Component causing the issue

Nil

### Example YAML snippet

```yaml
Irrelevant.
```

### Anything in the logs that might be useful for us?

```txt
No.
```

Masterz69, 2024-02-12:
Confirm.
Add-on restart resort devices in right order.
But after restart - new devices again appears at the end/bottom of list.

nielsnl68, 2024-02-12:
okay, it is not only me. seeing this as an issue.

randybb, 2024-02-12:
Yea, I noticed that yesterday when I have added a new device.
Just tested that on my dev machine - was working fine in 2023.11.6, it started in 2023.12.0.

Masterz69, 2024-02-12:
One more just noticed.
Previously sorting in dashboard was case-insensitive.
Now see first names with first captial letter, then all with small first letter, then added after add-on restart.

Masterz69, 2024-03-30:
As well devices sorting differs between listing in ESPHome dashboard and listing in HA integration.

ESPHome dashboard - case-sensitive, capital letters first;
HA integration - case-INsensitive.

Would be nice to have consistent approach, as well update ESPHome dashboard listing on adding devices, not just putting at the end of list.

My preferred sorting in case-INsensitive, if someone care :)

closed 2024-04-03 by jesserockz