esphome/issues #5486: Incompatibility with python-voluptuous 0.14.2

rubdos, 2024-02-12:
### The problem

I received an update on Arch Linux to python-voluptuous 0.14.2, which yields this error:

```
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome==2023.12.5', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/esphome", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 13, in <module>
    from esphome import const, writer, yaml_util
  File "/usr/lib/python3.11/site-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components, iter_component_configs
  File "/usr/lib/python3.11/site-packages/esphome/config.py", line 14, in <module>
    from esphome import core, yaml_util, loader, pins
  File "/usr/lib/python3.11/site-packages/esphome/loader.py", line 13, in <module>
    import esphome.core.config
  File "/usr/lib/python3.11/site-packages/esphome/core/config.py", line 7, in <module>
    import esphome.config_validation as cv
  File "/usr/lib/python3.11/site-packages/esphome/config_validation.py", line 671, in <module>
    Schema(
  File "/usr/lib/python3.11/site-packages/esphome/voluptuous_schema.py", line 27, in __init__
    super().__init__(schema, required=required, extra=extra)
  File "/usr/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 214, in __init__
    self._compiled = self._compile(schema)
                     ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 298, in _compile
    return self._compile_dict(schema)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 570, in _compile_dict
    base_validate = self._compile_mapping(schema, invalid_msg='dictionary value')
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/voluptuous_schema.py", line 67, in _compile_mapping
    for skey, svalue in vol.iteritems(schema):
                        ^^^^^^^^^
AttributeError: module 'voluptuous' has no attribute 'iteritems'
```

The error goes away after downgrading said package to 0.14.1.

### Which version of ESPHome has the issue?

2023.12.5

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

nodemcu, but n/a

### Component causing the issue

core

### Example YAML snippet

```yaml
esphome:
  name: flowerpot3

esp32:
  board: nodemcu-32s
```

### Anything in the logs that might be useful for us?

```txt
See the issue itself :-)
```

### Additional information

voluptuous changelog: https://github.com/alecthomas/voluptuous/releases/tag/0.14.2

These caught my attention:

- https://github.com/alecthomas/voluptuous/pull/499: support: drop support for python 3.7
- https://github.com/alecthomas/voluptuous/pull/501: support: run tests on python 3.11
- https://github.com/alecthomas/voluptuous/pull/502: support: run tests on python 3.12

Since `iteritems()` might be something from a backwards compatible age, that's why it caught my eye. But I'm furthermore clueless here.

Baldhor, 2024-02-18:
@rubdos got the same problem, so I reinstalled previous version of voluptuous:
`
python -m pip install --force-reinstall -v "voluptuous==0.13.1"
`

Seems to work out just fine

rubdos, 2024-02-18:
@Baldhor downgrading with `pacman -U /var/cache/pacman/pkg/python-voluptuous-0.14.1-1-any.pkg.tar.zst` worked fine for me too, but partial upgrades are notably unsupported on Arch. I'd rather not have my OS eat my cat! :-)

zeirix, 2024-03-03:
Same here with Manjaro Release 23.1.3.
The latest voluptuous version from the Arch repo is 0.14.2-1.
Sadly, I've flushed my pkg chache for the quick repair...

leander091, 2024-03-03:
This should be fixed on the next release. 
[Commit](https://github.com/esphome/esphome/pull/6181/commits/846d738a6928f0468366f61a8c952def48a894b7)  will fix it

rubdos, 2024-03-04:
> This should be fixed on the next release.
> This [commit](https://github.com/esphome/esphome/pull/6181/commits/846d738a6928f0468366f61a8c952def48a894b7) will fix it

@zeirix That would mean that the next workaround for Arch is to use `esphome-dev` from the AUR :-)

zeirix, 2024-03-04:
Thank you so much!
I fixed it in the `/usr/lib/python3.11/site-packages/esphome/voluptuous_schema.py` directly and it works!
Trying to install `esphome-dev` ended with some dependency trouble.

rubdos, 2024-03-04:
> Trying to install `esphome-dev` ended with some dependency trouble.

Removing `esphome` and `platformio-core` first should handle that. There's currently a comment on the AUR to ask set `Provides = platformio-core` on `platformio-git`, but that has not been pushed yet.

```sh
pacman -R esphome platformio-core
paru eshome-dev platformio-git
```

or whatever favorite aur-helper you have around! :-)

closed 2024-04-21 by jesserockz