esphome/issues #5489: Failed to resolve component 'freertos'

TotalPanther317, 2024-02-13:
### The problem

This error 

```
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'freertos'.
```

will occur when I try to compile anyting with esp-idf.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.1

### What platform are you using?

ESP32

### Board

nodeMCU

### Component causing the issue

esp-idf framework

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/bridge.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing hub (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40400.0
INFO Installing platformio/tool-esptoolpy @ ~1.40400.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
INFO tool-esptoolpy@1.40400.0 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.16.0
INFO Installing platformio/tool-cmake @ ~3.16.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: tool-cmake@3.16.9 has been installed!
INFO tool-cmake@3.16.9 has been installed!
Tool Manager: Installing platformio/tool-ninja @ ^1.7.0
INFO Installing platformio/tool-ninja @ ^1.7.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-ninja@1.10.2 has been installed!
INFO tool-ninja@1.10.2 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40502.0
INFO Installing platformio/tool-scons @ ~4.40502.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-scons@4.40502.0 has been installed!
INFO tool-scons@4.40502.0 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
[... 99 lines cut ...]
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/build/hub/.pioenvs/hub/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'freertos'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:237 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:237 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 91.20 seconds =========================
```

ssieb, 2024-02-14:
Try re-installing the addon.

TotalPanther317, 2024-02-14:
It fixed itself, uninstalled all esphome installs this time.

closed 2024-02-14 by TotalPanther317