esphome/issues #5491: [Bug] Atom Echo does not connect to wifi with hidden SSID

o-su, 2024-02-14:
### The problem

Steps to reproduce
Case1: Try to connect to wifi with hidden SSID --> Connection fails
Case2:
Make SSID visible, connect Atom Echo to wifi --> Connected, works as expected
Hide SSID again --> Echo immediately disconnects and wont connect back again

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  name: m5stack-atom-echo-10b448
  friendly_name: M5Stack Atom Echo 10b448
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: secret
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
[W][wifi:524]: No matching network found!
```

Rubeast, 2024-02-14:
Just came to github to see if this was a known problem. I'm also experiencing this with a esp8266.

randybb, 2024-02-14:
```yaml
wifi:
  fast_connect: true
```
is needed if you want to connect to a hidden network as is explained in the [documentation](https://esphome.io/components/wifi.html)

closed 2024-02-14 by randybb