esphome/issues #5492: internal_temperature throws "Ignoring invalid temperature" on az-delivery-devkit-v4

Qhilm, 2024-02-15:
### The problem

Hello,

I have the Espressif ESP32_DevKitc_V4 and somehow it is not reporting the internal_temperature, YAML and error are below.

Other sensors, such as wifi_signal, work fine.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.1

### What platform are you using?

ESP32

### Board

ESP32_DevKitc_V4

### Component causing the issue

internal_temperature

### Example YAML snippet

```yaml
esphome:
  name: gasmeter
  friendly_name: Gasmeter

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable Home Assistant API
api:  # Add encryption key as desired
  encryption:
    key: !secret home_assistant_encryption_key
  services:
    - service: set_total
      variables:
        new_total: int
      then:
        - pulse_meter.set_total_pulses:
            id: sensor_pulse_meter_gas
            value: !lambda 'return new_total;'

ota:  # Add password as desired
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  reboot_timeout: 15min
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret fallback_ssid_password

web_server:

sensor:
  - platform: pulse_meter
    name: "gas consumption"
    id: sensor_pulse_meter_gas
    pin:
      number: 16
      inverted: true
      # If PULLUP, connect second cable to GND on ESP board. If PULLDOWN, take the 3.3V pin.
      mode: INPUT_PULLUP
    unit_of_measurement: "m³"
    device_class: gas
    state_class: measurement
    accuracy_decimals: 2 # no point for higher accuracy since 1 impulse = 0,01 m³

# Wi-Fi
# https://github.com/MallocArray/airgradient_esphome/blob/main/packages/sensor_wifi.yaml

  - platform: wifi_signal
    name: "WiFi Signal"
    id: wifi_dbm
    update_interval: 60s

# diagnostics
# https://github.com/MallocArray/airgradient_esphome/blob/main/packages/diagnostic.yaml

  - platform: internal_temperature
    name: "ESP Temperature"
    id: sys_esp_temperature
```

### Anything in the logs that might be useful for us?

```txt
`[14:14:41][D][internal_temperature:052]: Ignoring invalid temperature (success=0, value=53.3)`
```

Qhilm, 2024-02-15:
Somehow I had it working once. It was reporting something like 45°C for while. Then I changed the config and installed it (but the changes where in the pulse_meter sensor, GPIO16) and now it does not work anymore.

I never changed theses lines:

```yaml
  - platform: internal_temperature
    name: "ESP Temperature"
    id: sys_esp_temperature
 ```

very strange.

randybb, 2024-02-15:
**Note**
_Some ESP32 variants return a large amount of invalid temperature values, including 53.3°C which equates to a raw value of 128. Invalid measurements are ignored by this component._
https://esphome.io/components/sensor/internal_temperature.html
So this is normal.

Qhilm, 2024-02-15:
ok, thanks, that issue is then moot.

closed 2024-02-15 by Qhilm