esphome/issues #5494: ESP8266 (sonoff RE5V1C) not connecting to network (or sometimes)

AsacreHA, 2024-02-15:
### The problem

Hello everyone ! (I'm new in the game) 

I have executed the standard setup for an ESP8266 device via the ESPHome integration in Home Assistant. Despite a successful configuration process, the ESP8266 device is experiencing intermittent Wi-Fi connectivity, managing to establish a connection only about 1 in 10 attempts.

**setup :**  
- sonoff relay module RE5V1C : 
[image]
(chip esp8285)

- hub usb c --> usb A
- usb to serial : I tried the two I have
"esp-01 programmer" (CH340)
[image]

"HW-598"  (CP2102)
[image]

**Steps I Performed: (also in the pdf) :** 
[sonoff RE5V1C not connecting to network.pdf](https://github.com/esphome/issues/files/14297655/sonoff.RE5V1C.not.connecting.to.network.pdf)

1: Clicked 'New Device' on ESPHome in Home Assistant.
2: Clicked 'Continue.'
3: Chose a name for the device and clicked 'Next.'
4: Selected device type 'ESP8266.'
5: Copied the encryption key provided.
6: Clicked 'Install.'
7: Chose 'Plug into this computer' as the install mode.
8: Downloaded the compiled project after compilation.
9: Clicked 'Open ESPHome web.'
10: Selected the appropriate port for the device.
11: Clicked 'Connect.'
12: Clicked 'Install.'
13: Selected the compiled binary firmware and clicked 'Install.'
14: Closed the window after the configuration was installed.

remark, when I click on the 3dots, then "connect to wifi" that says : "an error occured. Improv wifi serial not detected" 
[image]

When it does connect to the network, and I modify the .yaml (only things like switch led_status,...) from the ESPHome integration in Home Assistant, I can upload it to the ESP via OTA. The log indicates that it's been uploaded, but then it can no longer resolve the IP. 
[image]
[logs.txt](https://github.com/esphome/issues/files/14299506/logs.txt)

Could it be a network configuration problem ? (I've heard that having different frequencies may not help, so I disabled the 5Ghz on my modem)

I hope someone have a solution for me, 

Thank you

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.1 

### What platform are you using?

ESP8266

### Board

Sonoff RE5V1C (basic) 

[... 313 lines cut ...]
Compiling .pioenvs/r2/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/sqrt32.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/stdlib_noniso.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/time.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/uart.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pioenvs/r2/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pioenvs/r2/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pioenvs/r2/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pioenvs/r2/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving .pioenvs/r2/libFrameworkArduino.a
Linking .pioenvs/r2/firmware.elf
RAM:   [====      ]  42.0% (used 34372 bytes from 81920 bytes)
Flash: [=====     ]  48.3% (used 494449 bytes from 1023984 bytes)
Building .pioenvs/r2/firmware.bin
esp8266_copy_factory_bin([".pioenvs/r2/firmware.bin"], [".pioenvs/r2/firmware.elf"])
========================= [SUCCESS] Took 62.20 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of r2.local
ERROR Error resolving IP address of r2.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
```

AsacreHA, 2024-02-16:
edit: tasmota works

othorp, 2024-02-21:
If you have WiFi connection issues, watch the serial console while the device is booting to see what is going on.

You might want to look at a known issue with ESP8266 and some 802.11n / 802.11ax WiFi access points:

https://github.com/esp8266/Arduino/issues/7965

If you want a shot in the dark, you could try adding the following to the top of your configuration to force the ESP8266 into 802.11g mode and see if it fixes you:

esphome:
  on_boot:
    - priority: 300 # AFTER_BLUETOOTH (Before WiFi)
      then:
        lambda: |-
          WiFi.setPhyMode(WIFI_PHY_MODE_11G);

asperger514, 2024-03-27:
After months of struggling with daily connectivity issues with my sonoff-sv & sonoff basic devices flashed with esphome, and countless variations on settings on my asus RT-AX86U and its RT-AX58U mesh node othorp's 'shot in the dark' solution solved everything. thank you @othorp

othorp, 2025-03-07:
ESP8266 802.11n issue was fixed in ESPHome 2024.4.0 with bump of Arduino version so WiFi mode workaround above should no longer be required.

532910, 2025-03-07:
2025.2.2 still doesn't work